m-labs/artiq #1375: SU-servo feedback

hartytp, 2019-10-15:
Now we've got some experience using SU-servo in anger, it would be useful to collate user feedback with the aim of thinking about improvements/extensions we might make in the future.

cc @cjbe @dnadlinger

dtcallcock, 2019-10-15:
Did you guys write any extensions?  I vaguely recall talk of a gui applet that allows one to monitor how much headroom is available on each laser.

Also, anecdotally, it seems not many other groups have picked up on SU-servo yet.  Hopefully [now it's part of M-Labs build system](https://github.com/m-labs/artiq/issues/1343) and with more documentation and development that will change.

hartytp, 2019-10-15:
>  I vaguely recall talk of a gui applet that allows one to monitor how much headroom is available on each laser.

Discussed but not implemented yet

dtcallcock, 2019-10-15:
Btw, my student wrote some example code that might be helpful if it was polished up and included in the documentation: https://forum.m-labs.hk/d/19-suservo-sample-code/20. As you can see, without much control theory background, he found the learning curve a bit steep from just looking at terse function docs.

hartytp, 2020-03-09:
@jordens the overall feedback from here is that the current implementation is very tight, making it somewhat hard to modify (e.g. to do phase feedback or to implement phase coherence). Partially, this is a matter of documentation/coding style -- the high-level docs are great, but it's somewhat hard to follow the detailed implementation of the sequencing -- and some of it is a trade-off between modularity (e.g. using a separate memory per channel) v resource usage. 

Not sure if @cjbe @dnadlinger have anything else they want to add to that?

hartytp, 2020-03-09:
The example code could also be bulked up a little to showcase basic configuration/usage of the servo.

hartytp, 2020-03-09:
https://github.com/m-labs/artiq/issues/1400 would also be nice

closed 2020-03-09 by hartytp