m-labs/artiq #1377: Coreanalyzer can't handle RTIO underflow exceptions raised by spi2 devices

jonaskeller, 2019-10-17:
# Bug Report

## One-Line Summary

The `coreanalyzer` tool exits with an error when run after an RTIO underflow was caused by an `SPIMaster` device (module `artiq.coredevice.spi2`).

## Issue Details

### Steps to Reproduce

1.  Run this experiment:

```
from artiq.experiment import *

class cause_underflow(EnvExperiment):
    '''cause an RTIO underflow'''
    def build(self):
        self.setattr_device("core")    
        self.setattr_device("spi0")

    @kernel
    def run(self):
        self.core.reset()
        at_mu(self.core.get_rtio_counter_mu())
        self.spi0.set_config_mu(0, 32, 24, 1)

```

Relevant `device_db.py` entry:
```
    "spi0": {
        "type": "local",
        "module": "artiq.coredevice.spi2",
        "class": "SPIMaster",
       "arguments": {"channel": 46}    
    }
```

2. Run `artiq_coreanalyzer -w output.vcd`

### Actual (undesired) Behavior
```

Traceback (most recent call last):
  File "C:\Anaconda3\envs\artiq4.0\Scripts\artiq_coreanalyzer-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Anaconda3\envs\artiq4.0\lib\site-packages\artiq\frontend\artiq_coreanalyzer.py", line 57, in main
    decoded_dump)
  File "C:\Anaconda3\envs\artiq4.0\lib\site-packages\artiq\coredevice\comm_analyzer.py", line 543, in decoded_dump_to_vcd
    channel_handlers[message.channel].process_message(message)
  File "C:\Anaconda3\envs\artiq4.0\lib\site-packages\artiq\coredevice\comm_analyzer.py", line 363, in process_message
    data = message.data
AttributeError: 'ExceptionMessage' object has no attribute 'data'
```

### Your System (omit irrelevant parts)

* Operating System: Windows 7 64bit
* ARTIQ version: 4.0 0+gitbf50dcf7

sbourdeauducq, 2019-10-17:
Is that only spi2 devices, or all devices?

jonaskeller, 2019-10-17:
If I cause the underflow using a `TTLOut` device, it doesn't occur.

jordens, 2019-10-28:
Yes. Thanks for submitting and analyzing!

commit 1f15e550 (2019-10-28): comm_analyzer: don't assume every message has data

close #1377
closed 2019-10-28 by jordens