m-labs/artiq #1383: PermissionError resulted from uploading files to root's /tmp at remote while other users are using the directory

HarryHo90sHK, 2019-11-04:
# Bug Report

## Summary

When the `artiq.remoting` module utility is used to upload files to the `/tmp` directory at a remote server through `paramiko.sftp`'s library, if there is another user who is using the `/tmp` folder, an PermissionError (Permission denied) will be thrown.

## Issue Details

### Steps to Reproduce

1. Prepare a remote computer and have someone else using the root's `/tmp` folder.
1. Prepare an ARTIQ system with its JTAG connected to the remote computer.
2. Start and enter an `artiq-fast` Nix shell (e.g. by running `nix-shell "<artiq-fast/shell-dev.nix>"`).
3. Use `artiq_mkfs` and `artiq_flash` to flash the IP address of the remote to the ARTIQ.
4. Prepare a storage image for flashing. For example, if Kasli is used as the core device, prepare the system description JSON file and compile gateware using `python -m artiq.gateware.targets.kasli_generic`.
5. Use `artiq_flash --srcbuild -d <path-to-board-binary-directory> -H <remote-hostname>` to attempt flashing the storage image to the ARTIQ at the remote.
6. The program will exit at a "permission denied" error.

### Expected Behavior

* After step 5, `artiq_flash` should run and exit without error.

### Actual (undesired) Behavior

* When flashing the storage image to the ARTIQ at the remote, the program exited. Traceback as follows:
```
  File "/nix/store/ar9x9y6hvsi91yc23vahdmamd49hfyni-python3.7-artiq-5.7028.9a35a2ed.beta/bin/.artiq_flash-wrapped", line 11, in <module>
    sys.exit(main())
  File "/nix/store/dn1mywnrir3xvjmgx9gzl0xpfmy8ar8p-python3-3.7.4-env/lib/python3.7/site-packages/artiq/frontend/artiq_flash.py", line 416, in main
    programmer.write_binary(*config["storage"], storage_img)
  File "/nix/store/dn1mywnrir3xvjmgx9gzl0xpfmy8ar8p-python3-3.7.4-env/lib/python3.7/site-packages/artiq/frontend/artiq_flash.py", line 161, in write_binary
    filename = self._client.upload(filename)
... (traceback messages for paramiko.sftp)
PermissionError: [Errno 13] Permission denied
```

### System

* Operating System: NixOS 19.09.976.c75de8bc12c (Loris)
* ARTIQ version: v5.7028.9a35a2ed.beta
* Software ident 5.7028.9a35a2ed.beta;test
* Gateware ident 5.7028.9a35a2ed.beta;test
* Hardware involved:
    * Kasli
    * SMA-DIO
    * Urukul

sbourdeauducq, 2019-11-04:
>     1. have someone else using the root's `/tmp` folder.

This typically only occurs if that other user used ``artiq_flash`` to write to ``/tmp``; because ``artiq.remoting.SSHClient`` always attempts to use the same ``/tmp/artiq`` folder for all users, and does not delete it afterwards.
I suggest using ``/tmp/artiq-$USER`` (or similar), plus deleting it at the end.

sbourdeauducq, 2019-11-04:
> I suggest using `/tmp/artiq-$USER` (or similar)

That would still cause problems if the user runs several ``artiq_flash`` commands simultaneously. A better solution is to run ``mktemp -d`` on the remote machine and get the output.

sbourdeauducq, 2019-11-05:
This would break the current caching mechanism. ``artiq-$USER`` is probably the best option then, with USER determined using ``getpass`` like paramiko does.

commit 307f39e9 (2019-11-05): remoting: fix multiuser access. Closes #1383
closed 2019-11-05 by sbourdeauducq