m-labs/artiq #1384: review IOB flip-flop usage

sbourdeauducq, 2019-11-05:
as per discussion in https://github.com/m-labs/artiq/pull/1382

jordens, 2020-07-29:
Vivado complains about the `IOB=TRUE` usage:
```
CRITICAL WARNING: [Place 30-722] Terminal 'urukul5_dds_reset_sync_in_n' has IOB constraint set to TRUE, but it is either not connected to a FLOP element or the connected FLOP element could not be brought into the I/O
```

occheung, 2025-02-19:
Vivado may not place the buffer in I/O logic when `IOB=True` is absent. This is consistent with the observation in https://github.com/sinara-hw/Urukul/issues/16

As for the application of `IOB=TRUE`, UG903 stated this for the `IOB`  constraint:
> Tells the placer to try to place FFs in I/O Logic instead of the fabric slice. This property must be assigned to the register and not to the port.

The buffer in `DifferentialOutput` should get the constraint instead of the `p` and `n` ports (as OFD), or replace the same buffer with an ODDR.

I can't replicate the critical warning above relating to the n port in recent vivado versions, but I would assume it has to do with both ports attempting to bring the same buffer into the I/O logic twice.