m-labs/artiq #1386: AD9910.set_att() overwrites attenuator settings of other channels

schmidf, 2019-11-12:
# Bug Report

## One-Line Summary

Calling `set_att()` for one AD9910 channel overwrites the attenuator settings of all other channels that were made in previous experiments.

## Issue Details

### Steps to Reproduce
Run `AttTest1` , then `AttTest2` from the following snippet:

```python
from artiq.experiment import *

class AttTest1(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch0")

    @kernel
    def run(self):
        self.core.reset()
        self.urukul0_cpld.init()
        self.urukul0_ch0.init()
        self.urukul0_ch0.set_att(10.*dB)
        self.urukul0_ch0.set(10*MHz)
        self.urukul0_ch0.sw.on()
        
class AttTest2(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("urukul0_ch1")

    @kernel
    def run(self):
        self.core.reset()
        self.urukul0_cpld.init()
        self.urukul0_ch1.init()
        self.urukul0_ch1.set_att(10.*dB)
        self.urukul0_ch1.set(10*MHz)
        self.urukul0_ch1.sw.on()
```

### Expected Behavior

The first experiment turns on ch0 with the attenuator set to 10 dB. The second experiment does the same for ch1, leaving ch0 unaffected.

### Actual (undesired) Behavior

The second experiment resets the attenuator of ch0 to its maximum setting (31.5 dB).

This seems to be caused by the following line:
https://github.com/m-labs/artiq/blob/e65303cfe9d94be0d1ab932160ace76bebdcbcee/artiq/coredevice/urukul.py#L297

`self.att_reg` is set to `0x00000000` in the constructor of the CPLD object and is not overwritten with the actual hardware settings by default.

A workaround seems to be calling `self.urukul0_cpld.get_att_mu()` before changing the attenuator settings in the experiments. If this is intended behaviour, it has to be documented in the docstrings of the `set_att` family of functions. Currently, nothing in the documentation suggests that these function may affect other channels.

### Your System (omit irrelevant parts)

* Operating System: Windows 7
* ARTIQ version: v5.7044.e65303cf.beta
* Version of the gateware and runtime loaded in the core device: 5.7044.e65303cf.beta
* Hardware involved: Kasli v1.1, Urukul v1.3 (AD9910)

jordens, 2019-11-12:
No. `att_reg` is not set to 0 but to the value in the device database.

The behavior is extensively documented:

https://github.com/m-labs/artiq/blob/e65303cfe9d94be0d1ab932160ace76bebdcbcee/artiq/coredevice/urukul.py#L149-L152

https://github.com/m-labs/artiq/blob/e65303cfe9d94be0d1ab932160ace76bebdcbcee/artiq/coredevice/urukul.py#L288-L291

But feel free to propose a PR to add that sentence to `set_att()` as well.

Future versions of Urukul can access the attenuators individually. There is no gateware and coredevice driver support funding yet.

https://github.com/sinara-hw/Urukul/issues/30
https://github.com/quartiq/urukul/issues/7

schmidf, 2019-11-12:
I have to disagree. This behavior being mentioned in the docstring of a different function in a different class is not proper documentation.

Also note that the docstring of `CPLD.set_att_mu()` does not explain _which_ settings will be written to the other channels, and the one of `CPLD.get_att_mu()` does not mention that the result is buffered in a member variable and will be used in future calls of `set_att_mu()`. Sure, one can somehow guess this behavior from careful reading of the class docstring, but reading documentation should not be guesswork.

I will propose a PR to address these issues, but I think that the underlying issue is that there is no theory of operation section for any of the EEMs. Auto-generated documentation may by convenient, but it cannot replace a proper high-level manual.

jordens, 2019-11-12:
Sure. It would be great if someone could invest time/money into improved EEM documentation.

closed 2019-11-18 by jordens

jbqubit, 2019-11-18:
Thanks for updating the docs @schmidf !

PR #1390 (2019-11-18): Improve documentation of urukul attenuator behavior
 

# ARTIQ Pull Request

## Description of Changes

Add documentation to all urukul-related functions that overwrite the attenuator settings of other channels. Document that the result of `CPLD.get_att_mu()` is stored and used in future calls of `CPLD.set_att_mu()`.

### Related Issue

Closes #1386 

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :scroll: Docs |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [ ] Close/update issues.
 [... cut]
commit f73e2a3d (2019-11-18): doc: clarify urukul attenuator behavior

Closes #1386