m-labs/artiq #1388: Artiq 3 documentation offline

jonaskeller, 2019-11-16:
# Question

## Category: Documentation

## Description

We have multiple experiments still on Artiq 3.7, and it seems like the HTML version of the documentation for that version has vanished (moved?) with the release of 5.

sbourdeauducq, 2019-11-17:
I've put it back, but why not upgrade?
https://m-labs.hk/artiq/manual-release-3/

closed 2019-11-17 by sbourdeauducq

jonaskeller, 2019-11-17:
Thanks.

Upgrading the control system is always disrupting at some level (and bears some risk of unexpected consequences), so people are reluctant to do it during ongoing experiments.

dhslichter, 2019-11-18:
As a general principle it's nice to have the old documentation versions still available for archival purposes in a "useful" (e.g. compiled HTML) format.  The magtrap is still running ARTIQ 2, for example, because of the disruption of changing versions with the various breaking changes that went into ARTIQ 3.  We will be switching to ARTIQ 4 for the next round of experiments, but in the meantime we need the documentation.  It's also important because sometimes you want to look up how something used to be done and how it has changed over time.

sbourdeauducq, 2019-11-19:
The git history never gets deleted (e.g. you can still find docs for pre-1.0) and the .rst files can be read manually, anyway I've also restored the release-2 links.
Starting with ARTIQ-5, there is also a PDF manual as a single file that is easier for users to save. Hopefully, that can become the preferred way of archiving the compiled documentation. One issue with the old manuals is they get indexed by Google and show up first in the search results.

dnadlinger, 2019-11-19:
While I agree that having the documentation available online might be convenient for users, it should also be pointed out that it is very easy to build locally – `cd doc/manual; make html`.

sbourdeauducq, 2019-11-19:
That needs Sphinx and various modules (wavedrom etc.) set up; sadly, most people are using Windows where installing those isn't straightforward.

dhslichter, 2019-11-19:
@dnadlinger this is the issue - when you're not running on Linux, it makes some of these things much more of a pain.  

@sbourdeauducq a PDF manual with embedded hotlinks, compiled fresh for each release, would certainly be good and would be best for archival purposes for people running older versions.  I have noticed the Google issue with the old manual versions too, and I am sympathetic.  I see you have a warning bar across the top for older manual versions, which is helpful in avoiding confusion.  Could you just put an appropriate `robots.txt` file in such that the older versions of the HTML manual are not crawled?  Then you have only the latest manual version appearing in Google, but you have prominent links or drop-down menus in the nav bar to previous releases, for example the way that the Python docs allow you to pull down to different versions of Python for any given page.  Thoughts?