m-labs/artiq #1389: Driver repository or listing

dtcallcock, 2019-11-18:
# ARTIQ Feature Request

## Problem this request addresses

Python Drivers and ARTIQ controllers for 3rd party hardware are scattered across the M-Labs and Quartiq repositories.  The list of available drivers in the documentation is 'non-exhaustive'.  It's unclear how groups can contribute or advertise drivers they've written to the wider ARTIQ community.

## Describe the solution you'd like

A central repository to host community contributed drivers or a listing where drivers can be advertised.

## Additional context

QCoDeS (the Microsoft version of ARTIQ!) has two driver repositories - the [centrally maintained one](https://github.com/QCoDeS/Qcodes/tree/master/qcodes/instrument_drivers) and the [user-submitted one](https://github.com/QCoDeS/Qcodes_contrib_drivers) (with guidelines on how to contribute).  Perhaps we can mirror this approach, with the maintained ones being the ones people have paid M-Labs to develop? The QCoDeS drivers themselves might actually be useful as they are nominally Python (though I haven't really checked them for suitability or license restrictions).

sbourdeauducq, 2019-11-18:
> a listing where drivers can be advertised

https://m-labs.hk/artiq/manual/list_of_ndsps.html

It is possible to add more information or user-submitted controllers to this list.

dtcallcock, 2019-11-18:
> It is possible to add more information or user-submitted controllers to this list.

Sure.  But if we just leave it at that, it's very likely the number of contributed drivers will remain at zero.  

I'm thinking of something that's much lower barrier to entry than actual ARTIQ development and doesn't require changing the main docs.  It seems that setting up a repository with some guidelines and basic documentation is very low effort and could become a big selling point of Artiq if it catches on (especially in the communities that QCoDeS is targeting that rely on less custom real-time hardware).  For example, one of the few things LabView has going for it is that you can find a driver for pretty much anything (http://www.ni.com/downloads/instrument-drivers/).

sbourdeauducq, 2019-11-18:
> it's very likely the number of contributed drivers will remain at zero.
> I'm thinking of something that's much lower barrier to entry than actual ARTIQ development and
> doesn't require changing the main docs.

Are you seriously suggesting that sending us a quick message or pull request to add the driver to the list is the main barrier to entry for writing drivers?
We can add a few lines encouraging the submission of contributed drivers for listing in the documentation, if that is the problem.

Having one big repository creates issues with versioning, bloat, CI, and packaging/dependencies, and it actually makes things *more* inconsistent since there would still be some third-party controllers in their own repositories.

Letting people host their controllers where they want and then having a controller registry makes much more sense IMO. The controller registry can be as simple as the current documentation with manual maintenance, or something more polished and automated once that makes sense.

> For example, one of the few things LabView has going for it is that you can find a driver for pretty much anything (http://www.ni.com/downloads/instrument-drivers/).

We can develop a nice search engine or web tool once there are enough drivers, but right now it just sounds overkill to me.

dtcallcock, 2019-11-18:
> Are you seriously suggesting that sending us a quick message or pull request to add the driver to the list is the main barrier to entry for writing drivers?

It wouldn't be if it was stated that was the right way of going about it.

> We can add a few lines encouraging the submission of contributed drivers for listing in the documentation, if that is the problem.

Yes, I think articulating a vision is an obvious first step.  We can contribute a couple to get the ball rolling.  Perhaps they should be in a separate table or you could add another table column to allow the maintainer to be identified.

 > Having one big repository creates issues

I was thinking of something like organizing them under an organization like sinara-hw.  

> Letting people host their controllers where they want and then having a controller registry makes much more sense IMO. The controller registry can be as simple as the current documentation with manual maintenance, or something more polished and automated once that makes sense.

Sure, it makes sense to start small and seeing if there is any uptake.  

> but right now it just sounds overkill to me.

Of course.  I just wanted to make the point that NI have managed to provide drivers for almost everything out there at almost no cost to themselves.

sbourdeauducq, 2019-11-18:
> at almost no cost to themselves

I'm not so sure about this.

closed 2019-11-18 by sbourdeauducq

dhslichter, 2019-11-18:
@dtcallcock if you want to set up a Github organization for hosting ARTIQ NDSPs, and have each one (or coherent sub-cluster) be an individual repository, and grant permissions to interested parties to create/populate/maintain those repositories, it'd be a good start.  I bet that @jbqubit would have some to contribute.

dtcallcock, 2019-11-18:
>if you want to set up a Github organization ... it'd be a good start.

If people like @jbqubit @cjbe @hartytp have things they want to commit and think this is the best way of doing it then I'm happy to take the lead.  Otherwise we can just start with people hosting them in their own organizations as @sbourdeauducq suggests.

commit 9f459f37 (2019-11-18): doc: NDSP URLs, mention contributions can be added to list

Closes #1389