m-labs/artiq #1391: Misoc package missing from conda install

drewrisinger, 2019-11-20:
# Bug Report

## One-Line Summary

On a vanilla conda install of ARTIQ-beta, the ``misoc`` python package is missing.

## Issue Details

### Steps to Reproduce

From a command-line prompt with conda:
```bash
conda create -y -n artiq-5-beta
conda install -y -n artiq-5-beta -c https://conda.m-labs.hk/artiq-beta artiq
conda activate artiq-5-beta
python -c "import misoc"
```

Similarly, the following produces no information:
```bash
pip search misoc
conda search -c https://conda.m-labs.hk/artiq-beta
```

### Expected Behavior

Last line should produce no output.

### Actual (undesired) Behavior

Python command produces ``ImportError``

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: 6.7069.56d4b70e.beta
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
```
artiq                     6.7069.56d4b70e.beta               0    https://conda.m-labs.hk/artiq-beta
binutils-or1k-linux       2.27                 h93a10e1_6    m-labs
blas                      1.0                         mkl
bscan-spi-bitstreams      0.10.0                        2    m-labs
ca-certificates           2019.10.16                    0
certifi                   2018.8.24                py35_1
cffi                      1.11.5           py35h74b6da3_1
curl                      7.60.0                   vc14_0  [vc14]  conda-forge/label/main
h5py                      2.8.0            py35h3bdd7fb_2
hdf5                      1.10.2                   vc14_0  [vc14]  conda-forge/label/main
icc_rt                    2019.0.0             h0cc432a_1
icu                       58.2                     vc14_0  [vc14]  conda-forge/label/main
intel-openmp              2019.4                      245
jpeg                      9b                       vc14_2  [vc14]  conda-forge/label/main
krb5                      1.14.6                   vc14_0  [vc14]  conda-forge/label/main
libgit2                   0.27.0                   vc14_0  [vc14]  conda-forge/label/main
libiconv                  1.15                 h1df5818_7
libpng                    1.6.34                   vc14_0  [vc14]  conda-forge/label/main
libssh2                   1.8.0                hd619d38_4
libusb                    1.0.20                        0    m-labs
llvmlite-artiq            0.23.0.dev               py35_5    m-labs
mkl                       2018.0.3                      1
mkl_fft                   1.0.6            py35hdbbee80_0
mkl_random                1.0.1            py35h77b88f5_1
numpy                     1.14.2           py35h5c71026_1
openocd                   0.10.0                        6    m-labs
openssl                   1.0.2o                   vc14_0  [vc14]  conda-forge/label/main
pip                       10.0.1                   py35_0
prettytable               0.7.2                      py_3    conda-forge/label/main
pycparser                 2.19                     py35_0
pygit2                    0.27.1              py35_vc14_0  [vc14]  conda-forge/label/main
pyqt                      5.9.2            py35h1aa27d4_0
pyqtgraph                 0.10.0           py35h28b3542_3
python                    3.5.6                he025d50_0
python-dateutil           2.8.1                      py_0
python-levenshtein        0.12.0                   py35_0    conda-forge/label/main
pythonparser              1.1                        py_8    m-labs
qt                        5.9.4            vc14hd9fb3db_0
quamash                   0.5.5                      py_4    m-labs
regex                     2018.08.29       py35hfa6e2cd_0
scipy                     1.1.0            py35hc28095f_0
setuptools                40.2.0                   py35_0
sip                       4.19.8           py35h6538335_0
sipyco                    1.0                      pypi_0    pypi
six                       1.11.0                   py35_1
sqlite                    3.22.0                   vc14_0  [vc14]  conda-forge/label/main
vc                        14                  h21ff451_11
vs2015_runtime            14.16.27012          hf0eaf9b_0
wheel                     0.31.1                   py35_0
wincertstore              0.2              py35hfebbdb8_0
zlib                      1.2.11                   vc14_0  [vc14]  conda-forge/label/main
```

sbourdeauducq, 2019-11-21:
MiSoC is not required for running ARTIQ, only for building firmware and gateware.
Building those with conda is no longer supported, and never was on Windows anyway.
Supporting conda flows would be expensive (here is a data point: the [PSF is offering $220k](https://pyfound.blogspot.com/2019/11/seeking-developers-for-paid-contract.html) for some minor improvements to pip) and the results would be poor.
Use Nix, build from source, or offer $220k to build the firmware with conda (which would likely fund a complete conda rewrite, since it's so shitty).

closed 2019-11-21 by sbourdeauducq

drewrisinger, 2019-11-21:
I was just trying to download the package to use some of the modules in it for migen simulations that I'm doing. I was building a new conda environment on my laptop because it's on Windows, so no Nix. 

The old system had misoc as a conda package, and it was not clear that it had been removed. This means that MiSoC is no longer available anywhere as an easy-download package (excepting installing from source via GitHub). I wasn't aware that installing one package would require an entire conda rewrite.

sbourdeauducq, 2019-11-21:
Then that issue should have been filed in misoc and not artiq.

For migen there are third-party packages on anaconda.org.

> I wasn't aware that installing one package would require an entire conda rewrite.

Try replicating the current build and test system but with a conda-based flow, and we'll talk about that.
And misoc has binary dependencies (LLVM-or1k, Clang, GCC-or1k, ...) to be useful so it's obviously not even "one" package.