m-labs/artiq #1392: Conda ARTIQ 5 packages not available.

drewrisinger, 2019-11-21:
# Bug Report

## One-Line Summary

ARTIQ 5 packages are not available anywhere via Conda

## Issue Details

### Expected Behavior

Conda packages for ARTIQ 5, not just ARTIQ-6 (``artiq-beta``) should be available now that ARTIQ 5 has been released.

### Actual (undesired) Behavior

Presumably related to #1326 . #1326 never stated that stable packages for a specific release wouldn't be available.

Further, instructions from https://raw.githubusercontent.com/m-labs/artiq/master/install-with-conda.py only point to the https://conda.m-labs.hk/artiq-beta channel, which means if it is not there then it's not available/unsupported/undocumented.

```bash
$ conda search artiq -c https://conda.m-labs.hk/artiq-beta
Loading channels: done
# Name                       Version           Build  Channel
... ARTIQ 2 & 3 packages excluded ...
artiq                            4.0   0+gitbf50dcf7  m-labs
artiq                            4.0   0+gitbf50dcf7  m-labs/label/main
artiq                6.7074.c536f6c4.beta               0  artiq-beta
```

drewrisinger, 2019-11-21:
Also, the conda.m-labs.hk/artiq-beta implies that any conda packages installed from there are only BETA, if there is a stable channel it's not documented.

sbourdeauducq, 2019-11-21:
> Further, instructions from https://raw.githubusercontent.com/m-labs/artiq/master/install-with-conda.py only point to the https://conda.m-labs.hk/artiq-beta channel,

But commands from https://raw.githubusercontent.com/m-labs/artiq/release-5/install-with-conda.py, which is correctly liked in the release-5 docs contrary to what you say, point to the other channel.

closed 2019-11-21 by sbourdeauducq