m-labs/artiq #1408: error_specialized_annot.py test broken

sbourdeauducq, 2019-12-18:
https://nixbld.m-labs.hk/build/28099/nixlog/1
after #1407

dnadlinger, 2019-12-18:
Darn, that was an accidental inclusion of an unrelated change. Will fix asap. What's the best way of running Hydra tests ahead of time? (Like previously in BuildBot maybe; or at least the same nix build.)

sbourdeauducq, 2019-12-18:
You can do something like:
```
nix-scripts/artiq-fast$ nix-build -A artiq -I artiqSrc=<path to ARTIQ sources>
```
Hydra is not required, you only need Nix (The other tests can be similarly run as well, but need things like Vivado and the Windows VM image set up).

dnadlinger, 2019-12-18:
Fixed in 594ff457, although I didn't get to run the other tests locally yet.

closed 2019-12-18 by dnadlinger