m-labs/artiq #1410: Network devices support in artiq_ddb_template

airwoodix, 2019-12-19:
# ARTIQ Feature Request

## Problem this request addresses

When generating a ``device_db.py`` from a ``sinara-systems`` JSON description files using ``artiq_ddb_template``, aliases and network devices are not supported.

Are there known problems associated with describing aliases and network devices in the JSON files?

## Describe the solution you'd like

Support aliases and network devices in JSON descriptions and ``artiq_ddb_template``.

## Additional context

Also requires modifying minimally other Kasli generic tooling.

sbourdeauducq, 2019-12-20:
It's a device database *template*.
I would like to keep the JSON descriptions simple (i.e. stick to Sinara hardware, which is where the difficulty is for writing device databases with the correct RTIO channels) and not include support for whatever can be done in the device database.

Note that device databases are Python, and you can rename the template to ``device_db_template.py`` and then have a ``device_db.py`` file like this:

```python
import copy
import device_db_template

device_db = copy.copy(device_db_template.device_db)
device_db["my_new_device"] = ...
```

airwoodix, 2019-12-20:
Thanks, makes sense. From the end-user point of view, it's just very attractive to have one single file describing all the hardware. But I also see the potential escalation with adding more and more extra features to the JSON file / ddb_template generator.

closed 2019-12-20 by airwoodix

sbourdeauducq, 2019-12-20:
Those template files are generated on our server (nixbld.m-labs.hk) and it doesn't seem convenient to have users somehow send their whole lab hardware description to the build server.