m-labs/artiq #1417: Mod operatoin fails in Kernel

connorgoham, 2020-01-09:
# Bug Report

## One-Line Summary

Is the mod operation (%) meant to be usable in kernels? If so, it fails.

## Issue Details
Using the mod operation (%) in a kernel method in such a way that it can't be optimized away by the compiler fails yielding an error message.

### Steps to Reproduce

run code:
```
class ModTest(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    def run(self):
        self.a = 0.0
        self.krun(5.0)
        print(self.a)

    @kernel
    def krun(self, num):
        for k in range(1,2):
            self.a=num%k
```

### Expected Behavior

Print out 

> 1.0

### Actual (undesired) Behavior

* There is an error:

> root:Terminating with exception (LoadError: cannot load kernel: symbol lookup error: fmod)
> Traceback (most recent call last):
>   File "/home/rabi/artiq-dev/artiq/artiq/master/worker_impl.py", line 287, in main
>     exp_inst.run()
>   File "/home/rabi/gitlab/brittonlab/artiq-lab/hucul-west/repository/test_programs/mod_test.py", line 13, in run
>     self.krun(5.0)
>   File "/home/rabi/artiq-dev/artiq/artiq/language/core.py", line 54, in run_on_core
>     return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
>   File "/home/rabi/artiq-dev/artiq/artiq/coredevice/core.py", line 129, in run
>     self.comm.load(kernel_library)
>   File "/home/rabi/artiq-dev/artiq/artiq/coredevice/comm_kernel.py", line 240, in load
>     raise LoadError(self._read_string())
> artiq.coredevice.comm_kernel.LoadError: cannot load kernel: symbol lookup error: fmod

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 18.04.3 LTS
* ARTIQ version: 4.0.dev0+1298.ge4d48a78
* [conda_list.txt](https://github.com/m-labs/artiq/files/4042132/conda_list.txt)

* Hardware involved: Kasli v1.1

sbourdeauducq, 2020-01-10:
I suppose this is because LLVM emits a ``fmod`` symbol instead of the expected ``__moddi3`` for the ``frem`` IR. I don't know why though, if you can read the LLVM sources (https://github.com/m-labs/llvm-or1k) and find out what is going on, that would be helpful.

sbourdeauducq, 2020-01-10:
Not the issue, but you seem to have made a mess of your conda environment:
```
artiq                     4.0.dev0+1298.ge4d48a78.dirty           <pip>
artiq-dev                 5.0.dev         228+git4e142dfb    m-labs/label/dev
```

I recommend switching to ARTIQ-5, reflash the board to get the fix (if this is for a M-Labs or QUARTIQ purchased system, you should have -5 and -6 firmware builds for it), and use the Nix packages instead of conda since you are running Linux.

commit 583a18dd (2020-01-10): firmware: expose fmod to kernels. Closes #1417
commit 1d678b7d (2020-01-10): firmware: expose fmod to kernels. Closes #1417
commit abd3c2a0 (2020-01-10): firmware: expose fmod to kernels. Closes #1417
closed 2020-01-10 by sbourdeauducq