m-labs/artiq #1418: Access datasets across experiments

ajrazander, 2020-01-10:
# Question

## Category: Support

## Description

I'd like to access datasets across experiments as described in on this [page](https://m-labs.hk/artiq/manual/environment.htm), but have not been able to find any information on how to do that and my guess work hasn't been successful either. In the first experiment, a dataset is generated. In the second experiment, I use the get_dataset method to retrieve the first experiment's dataset, but get_dataset is returning an attribute error.

I am both asking how to access data from a previous experiment and that such a useful feature be explained on the tutorial page https://m-labs.hk/artiq/manual/getting_started_mgmt.html#datasets

Thank you!

dnadlinger, 2020-01-10:
The experiment that sets the dataset needs to specify `broadcast=True` for it to be accessible on the master.

ajrazander, 2020-01-13:
The experiment generating the data does have `Broadcast=True` . Still getting the Attriburre Error

jordens, 2020-01-13:
1. Show us your code.
2. Show us the error message.

ajrazander, 2020-01-13:
Thank you for taking time to look into this.

In the first file, we run an experiment and save data to the dataset "pmt_counts". In the analyze function, we (successfully) use `get_dataset` to recover "pmt_counts" and compute a new dataset called "mean_counts". The problem arises in the second file. I try to access the "mean_counts" dataset with `get_dataset` in the 3rd line of the analyze function, but get the error:
`AttributeError: 'RabiAnalyze' object has no attribute 'get_datasets'`

**First file**

```
from artiq.experiment import *
import matplotlib.pyplot as plt
import numpy as np
import time

DELAY = 1000*ms

class RabiOscillation(EnvExperiment):

    def build(self):
        # Hardware setup
        self.setattr_device("core")
        self.setattr_device("urukul0_cpld")
        self.doppler_cooling = self.get_device("urukul0_ch3")  # Doppler cooling AOM
        self.protection = self.get_device("urukul0_ch1")       # Protection doppler cooling AOM
        self.optical_pumping = self.get_device("urukul0_ch0")  # Optical Pumping AOM
        self.setattr_device("ttl4")                            # TTL connected to horn
        self.detection = self.get_device("urukul0_ch2")        # Detection AOM
        self.setattr_device("ttl0")                            # PMT channel

        # Hyperparameters
        self.dc_time = 5000      # time spent doppler cooling in us
        self.op_time = 3         # time spent optical pumping in us
        self.det_time = 1000     # time to count photons in us

        # User input hyperparameters
        self.setattr_argument("longest_pulse_time", NumberValue(ndecimals=2, step=1))  # longest pulse time in us
        self.setattr_argument("pause_time", NumberValue(ndecimals=2, step=1))  # time window shift
        self.setattr_argument("n_steps", NumberValue(ndecimals=0, step=1))  # number of pulse steps longest_pulse_time
        self.setattr_argument("n_reps", NumberValue(ndecimals=0, step=10))  # number of iterations with SAME pulse time

        try:
            # Calculate pulse times
            self.pulse_times = []
            for i in range(self.n_steps):
                self.pulse_times.append(self.longest_pulse_time * i / (self.n_steps - 1.0) + self.pause_time)
        except:
            #  Input values may not defined yet on start up. Set input hyperparameters to default values
            self.longest_pulse_time = 34.0
            self.pause_time = 0.0
            self.n_steps = np.int32(10.0)
            self.n_reps = np.int32(100.0)

            # Calculate pulse times
            self.pulse_times = []
            for i in range(self.n_steps):
                self.pulse_times.append(self.longest_pulse_time * i / (self.n_steps - 1.0) + self.pause_time)

    @kernel
    def run(self):
        # Initialize core timeline
        self.core.break_realtime()
        delay(DELAY)
        self.core.reset()
        delay(DELAY)

        # cpld and channel initializations
        self.urukul0_cpld.init()
        self.doppler_cooling.init()
        self.protection.init()
        self.optical_pumping.init()
        self.detection.init()
[... 139 lines cut ...]
        print('Fit parameters:', popt)
        print("Estimated pi pulse time (us):", pi_pulse_time)

        # Save pi pulse time to h5 file
        # if self.work_with_current_dataset:
        #     self.set_dataset("pi_pulse_time", pi_pulse_time, broadcast=True)
        # else:
        #     with h5py.File(self.filename, 'r+') as f:
        #         f['datasets']['pi_pulse_time'][...] = pi_pulse_time
        #         f.close()
        #         print("Wrote pi pulse time to filename")

        # Compute x and y data for fit plot
        x = np.linspace(pulse_times[0], pulse_times[-1], len(pulse_times) * 20)  # make x values 20x finer than data
        fit_cos = cosine(x, *popt)  # compute y values from fit

        # Static HTML file where plot can be interacted with
        output_file("plot.html")

        # Plot data
        p = figure(title="Rabi Flopping", x_axis_label='µs', y_axis_label='photon counts', plot_width=900, plot_height=500)
        p.line(pulse_times, mean_counts, legend="Averaged Counts", line_width=2, color='blue')
        p.line(x, fit_cos, legend="fit", line_width=2, color='green')
        show(p)
```

closed 2020-01-13 by ajrazander