m-labs/artiq #1421: ADF5355 register level interface and PLL setup API

jordens, 2020-01-20:
Add register level interface for the ADF5355 family on mirny to allow PLL setup and tuning.
Algorithms to choose PLL parameters.

Similar to the register abstraction in the older PLL driver in `ptb-drivers`: https://github.com/quartiq/ptb-drivers/blob/master/ptb/adf4350.py

Context:

* original mirny issue: #1130 
* mirny infrastructure PR: #1405

airwoodix, 2020-03-10:
@jordens Would regdesc be an acceptable dependency for artiq?

jordens, 2020-03-10:
There is also https://github.com/m-labs/artiq-chip-reg
I don't see a problem with adding dependencies per se. Just needs packaging, publishing and maintenance.
But I don't know what the best dependency is.
I wonder how that integrates when these registers are exercised from kernels. Does that go smoothly or do we need something more like the register map generator in artiq-chip-reg?

jordens, 2020-03-10:
Or do you see this more as a non-kernel tool to generate register values?

airwoodix, 2020-03-16:
IIRC (December) there were some typing issues with regdesc in kernels. This should be fixable though. I'd be happy to use regdesc or similar in the main code, makes it more readable.

sbourdeauducq, 2020-03-19:
For the sake of consistency and simplicity, I would rather not use regdesc but have a list of registers like it's done in the other drivers. Is the ADF5356 programming interface particularly different from e.g. the AD9910, AD9912, AD53** chips that we use on the other cards?

airwoodix, 2020-11-10:
@jordens  Is this considered closed by #1530 ?

jordens, 2020-11-10:
Certainly. Thanks for the contribution!

closed 2020-11-10 by jordens