m-labs/artiq #1422: conda version with broken git commit hash

jbqubit, 2020-01-22:
## One-Line Summary

Name of conda repository typically includes build number and snippet of git commit hash. This is no longer the case. 

## Issue Details

Installed ARTIQ using conda. conda list tells me the version of ARTIQ that is installed is 
```
artiq                     5.7101.8ca1fea1               0    https://conda.m-labs.hk/artiq
```

There is no git commit for ARTIQ including this hash snippet. 

https://github.com/m-labs/artiq/search?q=8ca1fea1&unscoped_q=8ca1fea1

sbourdeauducq, 2020-01-23:
Github search not working or not used correctly?
https://github.com/m-labs/artiq/commit/8ca1fea161f50bf9ffc683127a7410b806af572e

closed 2020-01-23 by sbourdeauducq

jbqubit, 2020-01-23:
Indeed I can checkout 8ca1fea using my local git client. So not an ARTIQ problem. 

Looks like this is a github.com search bug. Annoyingly, github.com search interface also fails to find the full hash  `8ca1fea161f50bf9ffc683127a7410b806af572e`. 

I submitted a support request to github.com for whatever that's worth.

jbqubit, 2020-01-29:
Word from github....

>It looks like that commit is only on the release-5 branch but not on master. This is why the commit isn't showing up in the search results. So if 8ca1fea1 is ever merged to master, then it would show up in commit search.

"When you search for commits, only the default branch of a repository is searched. In most cases, this will be the master branch." ([link](https://help.github.com/en/github/searching-for-information-on-github/searching-commits))

dnadlinger, 2020-01-29:
For reference, just typing the URL manually woks, even with shortened hashes (`https://github.com/m-labs/artiq/commit/8ca1fea`).