m-labs/artiq #1424: intermittent GTH Ethernet breakage on Metlino (Sayma?)

sbourdeauducq, 2020-02-04:
With some gateware builds, GTH Ethernet is not working on Metlino. This is entirely determined by the gateware binary (a working bitstream with work all the time, a broken one will fail all the time).

Making unrelated changes to the gateware can trigger the problem. For example ARTIQ bf9f4e380a31143f762cd2b184d89e5b76168958 has working Ethernet, but adding this to the SoC top-level makes Ethernet fail:
```
 self.comb += platform.request("filtered_clk_sel").eq(0)
```

I have not done enough testing on Sayma to know is this is a problem there as well or not.

hartytp, 2020-02-04:
uuurgh. I had various problems like that with Sayma. I make a trivial change to one part of the gateware and something unrelated would break.

sbourdeauducq, 2020-02-04:
On v1 with the broken VCCINT power supply this sort of behavior can be explained. But v2 doesn't have this issue AFAICT.

jbqubit, 2020-03-02:
@sbourdeauducq Does this problem repeat on your hardware in HK? Is this a problem with P3V3?

sbourdeauducq, 2020-03-03:
I don't have a Metlino in HK. I have not seen it on Sayma, but I didn't do a lot of tests either, so it's probably not significant. Why would this be related to P3V3?

sbourdeauducq, 2020-05-29:
Still seeing the problem.

sbourdeauducq, 2020-06-11:
Still an issue with Vivado 2020.1.

jbqubit, 2020-06-12:
@marmeladapk Do you see this on any of the Metlinos at Creotech?

marmeladapk, 2020-06-17:
@jbqubit I don't recall seeing this in CTI, but I didn't run ARTIQ on our Metlinos in a long time.

jbqubit, 2020-06-17:
@marmeladapk a CTI is preparing to deliver Demo System with Metlino, seems like now would be a good time to get ARTIQ up and running on Metlino again. 

https://github.com/sinara-hw/Sayma_AMC/wiki/Sayma_v2_contract_creotech#c335-ht5-demonstration-system

HarryMakes, 2020-08-20:
Hi, by @sbourdeauducq's request I've been testing with Kintex's GTH on Metlino. Using **ARTIQ-5** to build an SoC that uses one of the SFP as 1000BASE-X transceiver, I was able to reproduce a design that consistently breaks (i.e. never gets link-up) the GTH, done by adding some code on another design that consistently function properly. This confirms that GTH breaks on ARTIQ-5 with unknown (or very obscure) reasons.

However, using **ARTIQ-6** (v6.7345.5f6aa02b.beta), I have so far always been able to reproduce a consistently functioning design, with no link-up issues or Ethernet breakage at all - pinging, or even running ARTIQ experiments. One of the difference between ARTIQ-5 and ARTIQ-6 is the PCS logic: I recently submitted and got approval for [a SGMII-related PR on misoc](https://github.com/m-labs/misoc/pull/101), which also changed up the auto-negotiation FSM a bit to avoid non-detection of link-up. Perhaps this was crucial to fixing this apparent GTH breakage problem. And I believe the current ARTIQ-6 should guarantee consistency of the GTH's operation.

closed 2021-02-17 by sbourdeauducq