m-labs/artiq #1425: Metlino crashes when attempting to use many DRTIO links

sbourdeauducq, 2020-02-04:
When attempting to connect 7 DRTIO links, UART output on Metlino stops or outputs binary garbage (similar behavior was observed on Sayma with SAWG, and was due to the VCCINT issues).

Example output:
```
[     1.555740s]  WARN(board_artiq::drtio_routing): could not read routing table from configuration, using default
[     1.564444s]  INFO(board_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; 5: 5 0; 6: 6 0; 7: 7 0; }
��t(]Hi�▒���:x����:&���R
                        RL4����易��Α*��J��)F�(��)N�����F�J����
B��▒��)�Z�▒��O���kK��������k�)���:����)�k.ưk�J�9��骨J��*���B�2��
```

The links appear established on the satellite side, and the routing table is transferred. They don't work in a startup kernel. Ethernet could not be tested due to #1424.

marmeladapk, 2020-02-05:
I have a build that recreates that problem, so I'll be able to probe power supplies.

gkasprow, 2020-02-05:
Does the buck converter measure the feedback voltage under the FPGA?

sbourdeauducq, 2020-02-07:
It seems Sayma is not affected.
```
[     0.000005s]  INFO(runtime): ARTIQ runtime starting...
[     0.003865s]  INFO(runtime): software ident 5.0.dev+847.g6d26def3.dirty;master
[     0.011083s]  INFO(runtime): gateware ident 5.0.dev+847.g6d26def3.dirty;master
[     0.018321s]  INFO(runtime): log level set to INFO by default
[     0.024029s]  INFO(runtime): UART log level set to INFO by default
[     0.030395s]  INFO(runtime::rtio_clocking): using internal RTIO clock (by default)
[     0.296976s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     4.197332s]  INFO(board_artiq::si5324):   ...locked
[     4.201116s]  INFO(runtime): network addresses: MAC=02-00-00-00-00-11 IPv4=192.168.1.60 IPv6-LL=fe80::ff:fe00:11 IPv6=no configured address
[     4.215971s]  WARN(board_artiq::drtio_routing): could not read routing table from configuration, using default
[     4.224677s]  INFO(board_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; 5: 5 0; 6: 6 0; 7: 7 0; }
[     4.241342s]  INFO(runtime::mgmt): management interface active
[     4.254611s]  INFO(runtime::session): accepting network sessions
[     4.260077s]  INFO(runtime::session): running startup kernel
[     4.265055s]  INFO(runtime::session): no startup kernel found
[     4.270773s]  INFO(runtime::session): no connection, starting idle kernel
[     4.277633s]  INFO(runtime::session): no idle kernel found
[     4.282996s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[    10.119329s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 30 packets
[    10.196170s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[    10.202624s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[    10.209503s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[    10.215988s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128

```

```
diff --git a/artiq/gateware/targets/sayma_amc.py b/artiq/gateware/targets/sayma_amc.py
index 18e6f70e..7161da8b 100755
--- a/artiq/gateware/targets/sayma_amc.py
+++ b/artiq/gateware/targets/sayma_amc.py
@@ -393,7 +393,7 @@ class Master(MiniSoC, AMPSoC):
         self.submodules.drtio_transceiver = gth_ultrascale.GTH(
             clock_pads=platform.request("cdr_clk_clean", 0),
             data_pads=[platform.request("sfp", 0)] +
-                      [platform.request("rtm_gth", i) for i in range(8)],
+                      [platform.request("rtm_gth", i) for i in range(6)],
             sys_clk_freq=self.clk_freq,
             rtio_clk_freq=rtio_clk_freq)
         self.csr_devices.append("drtio_transceiver")
```

The only major difference is there was only 1 satellite connected (but still 7 DRTIO master links). @marmeladapk Can you confirm if Metlino with 7 links and 1 satellite connected crashes as well?

marmeladapk, 2020-02-12:
@sbourdeauducq Yes, it does.

sbourdeauducq, 2020-02-12:
Okay, so that really starts to look like a VCCINT-type problem...

gkasprow, 2020-02-12:
Is the R393 installed?

marmeladapk, 2020-02-14:
Yes.

gkasprow, 2020-02-14:
So it is not th VCCINT level issue because it is stabilized at the FPGA.
I'd check its stability as a step response.

marmeladapk, 2020-02-19:
I didn't see anything different in power supplies when probing VCCINT MGTAVCC and MGTVCCAUX between 3 channels and 7 channels. I'll post screenshots from oscilloscope on friday.

marmeladapk, 2020-02-21:
@sbourdeauducq I compiled Metlino with 12 drtio channels, Si5324 and 150MHz DRTIO. It doesn't crash. The onlly oddity is that link 11 is always recognized as up, regardless of any AMC card being connected.

marmeladapk, 2020-02-21:
Yellow VCCINT, blue MGTVCCAUX, pink MGTAVCC: 7 links
[image]

Noise in measurments comes mostly from crate fans (probes are inside the crate):

With crate turned off:
[image]

Only MGTVCCAUX seems noisy on its own.

12 links with Si5324, RTIO 125 MHz
[image]

This build doesn't crash and noise levels are the same (crate turned on).

marmeladapk, 2020-02-21:
So I tried both 125 MHz and 150 MHz RTIO with more than 6 channels. It seems to me that it could be a problem in a quick WRPLL hack that was done to get Metlino working when @sbourdeauducq was in Warsaw.

marmeladapk, 2020-02-21:
```
[     4.279491s]  INFO(runtime::rtio_mgt::drtio): [LINK#3] link RX became up, pinging
[    10.317353s]  INFO(runtime::rtio_mgt::drtio): [LINK#3] remote replied after 31 packets
[    10.402165s]  INFO(runtime::rtio_mgt::drtio): [LINK#3] link initialization completed
[    10.408603s]  INFO(runtime::rtio_mgt::drtio): [LINK#4] link RX became up, pinging
[    10.416289s]  INFO(runtime::rtio_mgt::drtio): [LINK#4] remote replied after 1 packets
[    10.502302s]  INFO(runtime::rtio_mgt::drtio): [LINK#4] link initialization completed
[    10.508744s]  INFO(runtime::rtio_mgt::drtio): [LINK#11] link RX became up, pinging
[    30.616093s] ERROR(runtime::rtio_mgt::drtio): [LINK#11] ping failed
[    30.621047s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[    30.627954s]  INFO(runtime::rtio_mgt::drtio): [DEST#4] destination is up
[    30.634419s]  INFO(runtime::rtio_mgt::drtio): [DEST#4] buffer space is 128
[    30.641496s]  INFO(runtime::rtio_mgt::drtio): [DEST#5] destination is up
[    30.647964s]  INFO(runtime::rtio_mgt::drtio): [DEST#5] buffer space is 128
[    30.854023s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] link RX became up, pinging
[    30.860414s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] remote replied after 1 packets
[    30.945863s]  INFO(runtime::rtio_mgt::drtio): [LINK#1] link initialization completed
[    30.952486s]  INFO(runtime::rtio_mgt::drtio): [LINK#11] link RX became up, pinging
[    51.059183s] ERROR(runtime::rtio_mgt::drtio): [LINK#11] ping failed
[    51.064447s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] destination is up
[    51.070834s]  INFO(runtime::rtio_mgt::drtio): [DEST#2] buffer space is 128
```

sbourdeauducq, 2020-02-21:
>  It seems to me that it could be a problem in a quick WRPLL hack that was done to get Metlino working when @sbourdeauducq was in Warsaw.

Hmm. I can't see how this code could produce this sort of memory corruption - it's almost entirely safe Rust code, and the unsafe bits only access those WRPLL gateware CSRs. I've also experienced no memory corruption whatsoever while developing WRPLL on Sayma.

jbqubit, 2020-03-02:
@sbourdeauducq Can you reproduce this problem in HK? 

@marmeladapk Are you still relying on the WRPLL hack from several weeks ago? Are you waiting on @hartytp for the full-on WRPLL solution?

sbourdeauducq, 2020-03-02:
I don't have a Metlino in HK. @marmeladapk do you know what the state of the front panels is?

The wrpll hack was a workaround for a Si5324 problem on Metlino that is now resolved. I just wanted to get the transceivers clocked and the Si5324 wasn't working then. I don't think whatever the Si549 or this hack was doing could explain this issue.

marmeladapk, 2020-03-02:
@jbqubit, @sbourdeauducq doesn't have Metlino yet.

> @marmeladapk Are you still relying on the WRPLL hack from several weeks ago? Are you waiting on @hartytp for the full-on WRPLL solution?

We're testing Metlino without ARTIQ, so I just used this hack to recreate this problem that we had during @sbourdeauducq's visit and check power supplies.

sbourdeauducq, 2020-03-02:
Are you sure that it's the wrpll hack-patch that triggers the problem? There's really nothing offensive in there. Could it be the clock switch line (which selects between Si549 and Si5324 on the hardware)?

sbourdeauducq, 2020-03-02:
Or it's a stack overflow maybe, then all sorts of weird things can happen.

sbourdeauducq, 2020-03-02:
Try this:
```
diff --git a/artiq/firmware/runtime/runtime.ld b/artiq/firmware/runtime/runtime.ld
index 67841f91..f92097a5 100644
--- a/artiq/firmware/runtime/runtime.ld
+++ b/artiq/firmware/runtime/runtime.ld
@@ -67,7 +67,7 @@ SECTIONS
 
     .stack ALIGN(0x1000) :
     {
-        . += 0x4000;
+        . += 0x10000;
         _fstack = . - 4;
     } > runtime
 
```

I'm actually surprised that the runtime stack was only 16k.

sbourdeauducq, 2020-03-09:
@marmeladapk Do you know when you can ship a Metlino to HK, preferably with front panels and everything?

marmeladapk, 2020-03-09:
> Try this:

It did not help.

> Do you know when you can ship a Metlino to HK, preferably with front panels and everything?

Probably in two weeks. We want to finish stress test and other tests first to rule out problem with hardware.

sbourdeauducq, 2020-05-29:
Not seeing the problem here but there is a "phantom" link:

```
[     0.000004s]  INFO(runtime): ARTIQ runtime starting...
[     0.003845s]  INFO(runtime): software ident 5.0.dev+892.g5db2afc7.dirty;master
[     0.011064s]  INFO(runtime): gateware ident 5.0.dev+892.g5db2afc7.dirty;master
[     0.018281s]  INFO(runtime): log level set to INFO by default
[     0.024001s]  INFO(runtime): UART log level set to INFO by default
[     0.030367s]  INFO(runtime::rtio_clocking): using internal RTIO clock (by default)
[     0.296961s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     4.407511s]  INFO(board_artiq::si5324):   ...locked
[     4.412718s]  INFO(runtime): network addresses: MAC=02-00-00-00-00-19 IPv4=10.100.0.99 IPv6-LL=fe80::ff:fe00:19 IPv6=no configured address
[     4.426409s]  WARN(board_artiq::drtio_routing): could not read routing table from configuration, using default
[     4.435116s]  INFO(board_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; 5: 5 0; 6: 6 0; 7: 7 0; 8: 8 0; 9: 9 0; 10: 10 0; 11: 11 0; 12: 12 0; }
[     4.455776s]  INFO(runtime::mgmt): management interface active
[     4.469019s]  INFO(runtime::session): accepting network sessions
[     4.474438s]  INFO(runtime::session): running startup kernel
[     4.479426s]  INFO(runtime::session): no startup kernel found
[     4.485157s]  INFO(runtime::session): no connection, starting idle kernel
[     4.492002s]  INFO(runtime::session): no idle kernel found
[     4.497395s]  INFO(runtime::rtio_mgt::drtio): [LINK#11] link RX became up, pinging
[    24.605096s] ERROR(runtime::rtio_mgt::drtio): [LINK#11] ping failed
[    24.610050s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[    24.816063s]  INFO(runtime::rtio_mgt::drtio): [LINK#11] link RX became up, pinging
```

Could it be some µTCA backplane shenanigans where e.g. one of the links is looped back? This is on the small NATIVE-R5 crate which does not have that many links.

sbourdeauducq, 2020-05-29:
With one fewer link compiled in, it crashes after ``RoutingTable``.

sbourdeauducq, 2020-05-29:
Okay, that's a simple problem, we are exhausting the CSR address space.

commit cb76f9da (2020-05-29): metlino: fix CSR collisions

Closes #1425
closed 2020-05-29 by sbourdeauducq