m-labs/artiq #1431: Ethernet connection trouble

alexist2623, 2020-02-26:
Hello, I am first to ARTIQ! 

So I have some trouble with connecting FPGA(KC705) to computer(OS : Ubuntu 18.04 LTS ).

FPGA is connected through JTAG cable, and Ethernet cable directly (without any hub).

Input commands were

"
$ artiq_flash -t kc705 -V nist_clock

$ artiq_mkfs flash_storage.img -s mac 00:0A:35:02:A1:DF -s ip 169.254.8.19

$ artiq_flash -t kc705 -V nist_clock -f flash_storage.img storage start

$ artiq_run led.py
"
And execution result is below.

[image]
<Figure.1>

[image]
<Figure.2>

[image]
<Figure.3>

Device connection is set as below!

[image]
<Figure.4>

As you can see in figures, 'artiq_flash' command works well!

So, I think JTAG works without any problem.

However, when I implement 'artiq_run led.py', it displays 'No route to host'.

Also, 'ping 169.254.8.19' responds well, but 'ping 169.254.8.10' does not.

My question is below!

1. Do I need additional connection, like UART and SFP, or device, for example hub ?

2. Is 'core_addr' in 'device_db.py' the IP of enp3s0?

Thank you!

P.S. If I set 'core_addr' to IP of PC(192.168.0.24) or enp3s0(169.254.8.19), 'conncetion refused' occurs.

sbourdeauducq, 2020-03-20:
> * Do I need additional connection, like UART and SFP, or device, for example hub ?

No. As long as the board is flashed and the RJ45 is connected it should ping.

> * Is 'core_addr' in 'device_db.py' the IP of enp3s0?

No. It should be the IP of the device.

From your screenshot, you flashed a link-local IPv4 address into the device, which can cause problems. You should flash a proper LAN IP address into the device - perhaps ask your network administrator for one. If you are unfamiliar with IP networking, try setting up a network connection manually between two PCs, and once you understand what is going on, move to the device.

I don't see any issue coming from ARTIQ itself here, so closing.

sbourdeauducq, 2020-03-20:
> P.S. If I set 'core_addr' to IP of PC(192.168.0.24) or enp3s0(169.254.8.19), 'conncetion refused' occurs.

This is expected since ARTIQ will try to connect to your PC instead of the device.

closed 2020-03-20 by sbourdeauducq