m-labs/artiq #1433: zombie runs

jordens, 2020-03-09:
# Bug Report

## One-Line Summary

Sometimes a run remains stuck in the "pending" stage and becomes invisible to other runs and is un-deletable.

## Issue Details

### Steps to Reproduce

No idea how to reproduce reliably.

In opticlock we have a couple tweak-up scripts hooked up to the F-keys. These scripts only exercise software functions (via controllers) and don't touch the coredevice. They also run in their own pipeline. (There are other runs running in other pipelines, among them runs that do interact with the coredevice). But activity or state changes in other pipelines are not needed.

Depending on conditions they might be submitted at reasonably high rates (~Hz) and will back up somewhat (maybe ~10 runs deep). But they should always be executed in sequence and removed when done.

### Expected Behavior

The runs execute according to the pipeline rules and be removed then.

### Actual (undesired) Behavior

* Sometimes runs get stuck in the pending stage and never run and can't be deleted.

```
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
|  RID  | Pipeline |     Status     | Prio | Due date | Revision |         File         |  Class name |
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
| 88501 |  motor   |    pending     |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 94501 |  motor   |    pending     |  0   |          |   N/A    |   tweak_motors.py    |   Tweak0Up  |
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
```

* Graceful or force deletion have no effect.
* There are no corresponding artiq worker processes (anymore).
* The run did not fail but also didn't leave a h5 file around. It looks like it never got out of pending.
* Nothing in the logs (at INFO level).
* Interestingly, other runs (same pipeline, same priority) have no problem bypassing and skipping the zombies. This should not be the case AFAIK.
* There does not seem to be any effect (side or not) of the zombies.
* This is rare. I've seen it maybe four times over a year and 50k rids including the two runs above.
* I have no idea what to make of the exact 6000 rid difference between these two occurrences. Would be a weird coincidence even given confirmation bias (any multiple of a reasonable power of ten or two would be surprising).
* An `artiq_master` restart obviously clears them.

* I haven't looked into debugging the running artiq_master process and I don't know whether it's easy to do (with `pdb`) and what to look for.

### Your System (omit irrelevant parts)

* Operating System: Linux, Debian buster
* ARTIQ version: ARTIQ v5.7073.d5f92a20: d5f92a20
* Version of the gateware and runtime loaded in the core device: n/a, software only
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
* Hardware involved: none

jordens, 2020-03-16:
This has evolved a bit.

* All the `deleting` runs are also zombies and can't be deleted.
* Some of them still have worker processes around.
* It looks like in the current state the following reproduces it (as in "adds another zombie"):
  * Submit run to `motor` pipeline
  * Run enters `prepare_done`
  * Attempt graceful termination
  * Run becomes `deleting`
  * Run becomes zombie, can't be deleted forcefully. Worker process hangs around.

```
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
|  RID  | Pipeline |     Status     | Prio | Due date | Revision |         File         |  Class name |
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
| 88501 |  motor   |    pending     |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 94501 |  motor   |    pending     |  0   |          |   N/A    |   tweak_motors.py    |   Tweak0Up  |
| 95049 |   log    |    running     |  0   |          |    0     | logging_collector.py | Controllers |
| 99335 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99336 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99337 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99338 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99339 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99340 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99341 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99342 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99343 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99344 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99345 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99346 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99347 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99348 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99349 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99350 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99351 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99352 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99353 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99354 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99355 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99356 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99357 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99358 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak1Down |
| 99359 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99360 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99361 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99362 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99363 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99364 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99365 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 99366 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak0Down |
| 99367 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak0Down |
| 99368 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak0Down |
| 99369 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak0Down |
| 99370 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |  Tweak0Down |
| 99371 |  motor   |    deleting    |  0   |          |   N/A    |   tweak_motors.py    |   Tweak1Up  |
| 92188 |   main   |    running     | -10  |          |   N/A    |       idle.py        |  IdleOnline |
+-------+----------+----------------+------+----------+----------+----------------------+-------------+
```

```
(opticlock5) opticlock@vettel:~/src/opticlock$ ps ax | grep worker_impl
 6877 ?        Ss    15:27 /home/opticlock/src/conda/envs/opticlock5/bin/python3.5 -m artiq.master.worker_impl 26,38 30
11640 ?        Ss     0:00 /home/opticlock/src/conda/envs/opticlock5/bin/python3.5 -m artiq.master.worker_impl 43,46 30
[... cut ...] : None, 'priority': 0}, 94501: {'status': 'pending', 'pipeline': 'motor', 'flush': False, 'due_date': None, 'expid': {'arguments': {}, 'repo_rev': 'N/A', 'file': 'tweak_motors.py', 'class_name': 'Tweak0Up', 'log_level': 30}, 'repo_msg': None, 'priority': 0}, 99356: {'status': 'deleting', 'pipeline': 'motor', 'flush': False, 'due_date': None, 'expid': {'arguments': {}, 'repo_rev': 'N/A', 'file': 'tweak_motors.py', 'class_name': 'Tweak1Down', 'log_level': 30}, 'repo_msg': None, 'priority': 0}}
```

sbourdeauducq, 2020-03-16:
Immediately logging termination of asyncio tasks due to exceptions might perhaps help.

jordens, 2020-03-16:
Help for fixing or help for debugging?

sbourdeauducq, 2020-03-16:
Debugging