m-labs/artiq #1435: Dashboard moninj connects to the coredevice directly

airwoodix, 2020-03-17:
# ARTIQ Feature Request

## Problem this request addresses

In the dashboard, the hostname to the coredevice for DDS/TTL moninj is read from the device DB and not overridable. In particular, it is not proxied through the master. This is problematic when tunnelling to the master from a network that doesn't see the coredevice itself.

Example: coredevice and computer running the master are on a private network with only the master computer visible from outside. One can forward all the master / broadcast ports to the client via SSH. The dashboard however does the coredevice lookup on it's own, from the client, which obviously doesn't resolve the coredevice.

## Describe the solution you'd like

Working solution: static lookup (coredevice -> localhost) on the client with port 1283 forwarded. But this requires extra hardcoding for the client.

IMHO better solution: proxy moninj through the master. Then the client really only needs to connect to the master.

## Additional context

Maybe linked to #676 
VPN should not be considered a solution here.

jordens, 2020-03-17:
Agreed that a transparent broker for moninj (potentially same protocol on both sides, and with a backing store for the state) looks like a reasonable step here.
Would also address/involve #838

sbourdeauducq, 2020-03-19:
The issue will resurface if you e.g. try to connect to controllers from an applet or other program on your local machine. ARTIQ is designed to work in a distributed environment, and I am afraid the proper solution is a VPN indeed (or IPv6, e.g. IPsec AH doesn't seem too crazy to implement on the device and elsewhere). Having the master proxy all sorts of network traffic is equivalent to a VPN; if you really want we can perhaps bundle a SOCKS server with ARTIQ (SOCKS is very nice to traverse firewalls as the server doesn't require any special system permissions) and add SOCKS support in the clients.

Nevertheless, if you dislike the idea of the dashboard connecting to the core device directly, it remains possible to add a controller that does what @jordens describes.

airwoodix, 2020-03-19:
Thanks for the feedback.

- Direct connection from the dashboard to the coredevice: it was just surprising, I never looked in the details. I expected everything to be proxied by the master but I see the point;
- VPN: in principle the right solution indeed. But requires upgrades to the lab infrastructure, which is difficult in these Covid-19 times... Hence the "not a solution here";
- SOCKS: IMHO too much complexity for too little gain. Full remote access still sounds like a niche. With time, infrastructure, and support from the universities' IT departments, VPN is relatively easy.

As an intermediate solution, we'll got for NAT on the client (easy for Linux clients, more complicated for Windows...).

cjbe, 2020-03-19:
I also agree that the core device is special, and that only the master should be talking to it. This also makes it easier to have the core device (and other embedded devices) on a segregated and secured network.

hartytp, 2020-03-19:
It seems odd having each gui connect to the core device directly (particularly given slow network speeds and generally limited resources). It also makes it more effort for applets to get moninj info. It would be nice to have moninj be just like all the other data the master broadcasts, with a unified api. I def feel that the future for moninj could be via applets rather than doing everything in the dashboard (much easier to customise).

sbourdeauducq, 2020-03-19:
> I def feel that the future for moninj could be via applets rather than doing everything in the dashboard (much easier to customise).

Yes.

sbourdeauducq, 2021-09-12:
Funded by Duke

sbourdeauducq, 2022-03-19:
done

closed 2022-03-19 by sbourdeauducq