m-labs/artiq #1436: Artiq_dashboard -s cannot connect to the artiq_master's PC

cyqian97, 2020-03-20:
# Question
Artiq_dashboard -s cannot connect to the artiq_master's PC

## Category: network

## Description
I am very new to the artiq and I have a basic question about using artiq_dashboard remotely.
I was trying to use artiq_dashboard in different computer from where the artiq_master is running. 
I got the correct IP address of the PC running artiq_master and I can ping it but my artiq_dashboard still cannot access that PC. Below is the error information I got when I run `artiq_dashboard -s [ip address of the artiq_master PC]`:
`WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 121] The semaphore timeout period has expired
Traceback (most recent call last):
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\Scripts\artiq_dashboard-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\site-packages\artiq\frontend\artiq_dashboard.py", line 115, in main
    args.server, args.port_control, "master_" + target))
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\site-packages\Quamash-0.5.5-py3.6.egg\quamash\__init__.py", line 273, in run_until_complete
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\futures.py", line 294, in result
    raise self._exception
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\tasks.py", line 242, in _step
    result = coro.throw(exc)
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\site-packages\sipyco\pc_rpc.py", line 203, in connect_rpc
    await asyncio.open_connection(host, port, limit=100*1024*1024)
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\streams.py", line 75, in open_connection
    lambda: protocol, host, port, **kwds)
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\base_events.py", line 776, in create_connection
    raise exceptions[0]
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\base_events.py", line 763, in create_connection
    yield from self.sock_connect(sock, address)
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\futures.py", line 381, in __iter__
    yield self  # This tells Task to wait for completion.
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\tasks.py", line 310, in _wakeup
    future.result()
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\futures.py", line 294, in result
    raise self._exception
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\site-packages\Quamash-0.5.5-py3.6.egg\quamash\_windows.py", line 41, in _process_events
  File "C:\Users\selen\Anaconda3\envs\artiq_rice\lib\asyncio\windows_events.py", line 509, in finish_connect
    ov.getresult()
OSError: [WinError 121] The semaphore timeout period has expired`

I opened the default TCP port (3250, 3251, 1067) for artiq_dashboard on both computer but the error still remain. I will much appreciate it if you can look into this case and give me some advice!

dnadlinger, 2020-03-20:
Did you launch `artiq_master` with `--bind=<desired IP address>"? It might only listen on the loopback interfacec by default.

[Added block code tags to the original post.]

cyqian97, 2020-03-22:
> Did you launch `artiq_master` with `--bind=<desired IP address>`? It might only listen on the loopback interfacec by default.
> 
> [Added block code tags to the original post.]

That works! Thanks a lot!!!

closed 2020-03-22 by cyqian97