m-labs/artiq #1446: Coredevice Input Validation

pathfinder49, 2020-05-07:
# ARTIQ Feature Request

## Problem this request addresses
Many coredevice drivers provide and/or use a conversion from SI units to machine-units. Many of these are vulnerable to machine-unit code over/underflow. This can result in silent undesired hardware responses (see #1443 & #1444).

## Describe the solution you'd like

Before returning a code, it should be checked for validity. Invalid codes should raise a `ValueError`. For many devices, codes are shorter than the default internal 32-bit integer representation. The majority of invalid codes can therefore be detected by checking the code bounds. For a 16-bit code I imagine somehting like:
```Python
code = do_conversion(*args)
if code < 0x0 or code > (1<<16)-1:
    raise ValueError("Invalid input!")
return code
```
This is both straight-forwad to implement and has minimal performance overheads (see #1445).

## Relevant coredevice methods/functions
It seems useful to compile a list of relevant coredevice methods/functions here. Feel free to add any that I've missed.

I've annotated the code size.

- [ ] urukul.CPLD.set_att (8-bit)

- [ ] AD9910.amplitude_to_asf (14-bit)

- [ ] AD9910.frequency_to_ftw (32-bit -> would need to work with int64 or do floating point math)

- [ ] AD9910.turns_to_pow (16-bit)

- [ ] AD9912.frequency_to_ftw (48-bit -> already using int64)

- [ ] AD9912.turns_to_pow (16-bit)

- [ ] AD9914.amplitude_to_asf (12-bit)

- [ ] AD9914.frequency_to_ftw (32-bit)

- [ ] AD9914.frequency_to_xftw (63-bit -> an int64 check might be a bit weak here...)

- [ ] AD9914.turns_to_pow (16-bit)

- [ ] suservo.Channel.set_y (17-bit)

jordens, 2020-05-07:
For DDS frequency and phase there should be no input validation. Frequencies outside the first Nyquist zone alias correctly according to Nyquist-Shannon. Preventing them also prevents convenient access to the other Nyquist zones and makes computation needlessly complicated. Phase wraps around modulo 2*pi radian (or 1 turn). Both are always valid. Please don't touch those and please don't restrict the input range. Adding a sentence to the docstring ist certainly worth though. The one thing that needs to be ensured is that overflow and sign bits don't collide with other data.

pathfinder49, 2020-05-07:
> The one thing that needs to be ensured is that overflow and sign bits don't collide with other data.

In these cases, it might be nice to mask the code? Example: `code = do_conversion(*args) & 0xffff`

jordens, 2020-05-07:
Yes. Absolutely.

commit 25388407 (2020-05-17): Coredevice Input Validation (#1447)

* Input validation and masking of SI -> mu conversions (close #1446)

* Update RELEASE_NOTES
closed 2020-05-17 by jordens