m-labs/artiq #1449: Error importing pandas in more than one Experiment

charlesbaynham, 2020-05-14:
# Bug Report

## One-Line Summary

If pandas is imported in more than one module in the experiments repo, one of two errors occurs. 

## Issue Details

If `pandas` is imported in at least two modules in the experiments repo (they don't have to be Experiments), one of two errors occur. Either:

```
root:Terminating with exception (TypeError: Parameterized generics cannot be used with class or instance checks)
Traceback (most recent call last):
...
  File "C:\Users\cb6\.conda\envs\artiq-5\lib\typing.py", line 1169, in __subclasscheck__
    raise TypeError("Parameterized generics cannot be used with class "
TypeError: Parameterized generics cannot be used with class or instance checks
```

or 

```
root:Terminating with exception (RecursionError: maximum recursion depth exceeded)
Traceback (most recent call last):
...
  File "C:\Users\cb6\.conda\envs\artiq-5\lib\typing.py", line 907, in __extrahook__
    if issubclass(subclass, scls):
RecursionError: maximum recursion depth exceeded
```

### Steps to Reproduce

1. Create a new repository of experiments
2. Create `file1.py` with the following contents:
```
from artiq.experiment import *

import pandas as pd

class TestPandas(EnvExperiment):
    
    def build(self):
        pass
```
3. Create `file2.py` with the following contents:
```
import pandas as pd
```

4. Scan the repository in artiq. 

### Expected Behavior

No error: experiment is listed. 

### Actual (undesired) Behavior

RecursionError as shown above, and `file2.py` is skipped (so any Experiments that it might contain are ignored). 

[image]

### Your System (omit irrelevant parts)

* Operating System: Window 10
* ARTIQ version: ARTIQ v5.7111.5168b831
* Version of the gateware and runtime loaded in the core device: N/A - no core device required
* If using Conda, output of `conda list` : [environment.txt](https://github.com/m-labs/artiq/files/4629648/environment.txt)
* Hardware involved: None

charlesbaynham, 2020-05-14:
I suspect that this might be related to #1364

sbourdeauducq, 2020-06-22:
Try again with the new Python 3.7 packages.

charlesbaynham, 2020-06-22:
Will do, I've got #652 on "watch"

charlesbaynham, 2020-07-24:
Closed by #652

closed 2020-07-24 by charlesbaynham