m-labs/artiq #1454: influxdb_schedule: Bug with storing string parameters

charlesbaynham, 2020-05-24:
# Bug Report

## One-Line Summary

Storing string parameters with double quotes and spaces breaks the influxdb_schedule parser.

## Issue Details

### Steps to Reproduce

1. Create an experiment with the following:

```
from artiq.experiment import EnvExperiment, StringValue

class InfluxSchedulerLogger(EnvExperiment):

    def build(self):
        self.setattr_argument('arg', StringValue(default='"hello world"'))
                              
        
    def run(self):
        pass
```

2. Run it with a `device_db.py` containing an `influxdb_schedule` controller. 

### Expected Behavior

Experimental parameters logged as normal. 

### Actual (undesired) Behavior

```
artiq_comtools.artiq_influxdb_schedule:got HTTP status 400 trying to update schedule: {"error":"unable to parse 'schedule,class_name=InfluxSchedulerLogger,file=influx_scheduler_issue.py,pipeline=main,status=add priority=0i,rid=620i,arguments=\"{\\\"arg\\\": \\\"\\\\\"hello world\\\\\"\\\"}\",repo_rev=\"N/A\",due_date=-1.0,log_level=30i,flush=False 1590322590826': bad timestamp"}
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v5.0.dev+889.g25388407
* Version of the gateware and runtime loaded in the core device: N/A
* Hardware involved: None

charlesbaynham, 2020-05-24:
Maybe this should be in `artiq_comtools`, depending on the source of the error.

sbourdeauducq, 2022-05-24:
https://github.com/m-labs/artiq-comtools/pull/12

closed 2022-05-25 by sbourdeauducq