m-labs/artiq #1457: Sayma RTM Si5324 not ACKing

hartytp, 2020-05-29:
# Bug Report

## One-Line Summary

Short summary.

## Issue Details

### Steps to Reproduce

1. Step 1.
2. Step 2.
3. Step 3.

### Expected Behavior

Behavior

### Actual (undesired) Behavior

* Text description
* Log message, tracebacks, screen shots where relevant

### Your System (omit irrelevant parts)

* Operating System:
* ARTIQ version: (with recent versions of ARTIQ, run ``artiq_client --version``)
* Version of the gateware and runtime loaded in the core device: (in the output of ``artiq_coremgmt -D .... log``)
* If using Conda, output of `conda list` (please submit as a file attachment, as this tends to be long)
* Hardware involved:

Booting Sayma and looking at the UARTs. On the primary I see

```
 __  __ _ ____         ____
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |
| |  | | |___) | (_) | |___
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2020 M-Labs Limited

Bootloader CRC passed
Gateware ident 6.7231.25388407.beta;satellite.sawg
Initializing SDRAM...
DQS initial delay: 111 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011100000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 111 taps
Write leveling: 89 110 128 123 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 236+-83 227+-94 203+-89 193+-91 done
SDRAM initialized
Memory test passed

Loading slave FPGA gateware...
  magic: 0x5352544d, length: 0x001a99c8
  ...done
Booting from flash...
Starting firmware.
[     0.000006s]  INFO(satman): ARTIQ satellite manager starting...
[     0.004753s]  INFO(satman): software ident 6.7231.25388407.beta;satellite.sawg
[     0.011972s]  INFO(satman): gateware ident 6.7231.25388407.beta;satellite.sawg
[     0.278541s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.184774s]  INFO(board_artiq::si5324):   ...locked
```

while on the secondary
```
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[     0.000005s]  INFO(satman): ARTIQ satellite manager starting...
[     0.004585s]  INFO(satman): software ident 6.7231.25388407.beta;satellite
[     0.011435s]  INFO(satman): gateware ident 6.7231.25388407.beta;satellite
panic at src/libcore/result.rs:945:5: cannot initialize Si5324: "Si5324 failed to ack write address"
```

any ideas @sbourdeauducq ?

hartytp, 2020-06-01:
@marmeladapk what's the plan for debugging this? AFAICT @sbourdeauducq hasn't seen it on any of his boards so can't debug it, CTI have seen it on a few boards, but haven't looked into it yet. When I can get my hands on a new Sayma I'll see if the problem goes away. However, it does seem like a real issue that will have to be solved properly before Sayma can be considered ready for deployment.

marmeladapk, 2020-06-01:
@hartytp I'm not sure how to proceed either right now. During our tests communication with Si5324 on RTM is checked hundreds of times and we didn't have problems with it on any board. Your boards were also tested before shipping.

You could check with a scope at Si5324 inputs if all timings and voltage levels are correct. Unfortunately I can't devote any time this week to this issue. :/

hartytp, 2020-06-01:
Without access to the lab, probing signals is pretty time consuming, so I won't attempt to debug this on the HW. If you send me the binary you used for testing then I'll happily try that. Otherwise I'll test out on another board when I can get access to it.

sbourdeauducq, 2020-10-15:
AFAIK this is fixed by 901be75ba4c08f3392ecc9398b0ebcead2bfc78c

closed 2020-10-15 by sbourdeauducq