m-labs/artiq #1463: Compilation failure for AD9912 .set introduced

charlesbaynham, 2020-06-12:
# Bug Report

## One-Line Summary

After commit 2538840756ee76ee49a72dc714361f25be2170ff, running `self.urukul0_ch0.set(100e6)` results in 

```
<artiq>\coredevice\ad9912.py:152:25-152:28: error: the type of this expression cannot be fully inferred
        self.bus.write((pow << 16) | (int32(ftw >> 32) & 0xffff))
                        ^^^
<artiq>\coredevice\ad9912.py:152:25-152:28: note: the expression has type numpy.int?
        self.bus.write((pow << 16) | (int32(ftw >> 32) & 0xffff))
```

## Issue Details

### Steps to Reproduce

1. Write the following to a file `test_DDS.py`:
```
from artiq.experiment import EnvExperiment, kernel

class TestDDS(EnvExperiment):

    def build(self, **kwargs):
        self.setattr_device('core')
        self.setattr_device('urukul0_ch0')

    @kernel
    def run(self):
        self.core.reset()
        self.urukul0_ch0.set(100e6)
```

2. Execute `artiq_run test_dds.py` in a directory with your `device_db` for a system with an AD9912 Urukul. 
3. `git bisect` the artiq code to find that before commit 2538840756ee76ee49a72dc714361f25be2170ff this executes as expected, but from this commit onwards you get the error shown above. 

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: 2538840756ee76ee49a72dc714361f25be2170ff onwards
* Version of the gateware and runtime loaded in the core device: `5.0.dev+567.g99e490f9;npl2`
* If using Conda, output of `conda list`: [env.txt](https://github.com/m-labs/artiq/files/4771405/env.txt)
* Hardware involved: Kasli rack with AD9912 Urukul

pathfinder49, 2020-06-12:
Ack, the breaking change should be the `& <mask>` here: https://github.com/m-labs/artiq/blob/1a17d0c869f31163d42b79f9f206c2f5cbcadddd/artiq/coredevice/ad9912.py#L177

According to the [docs](https://m-labs.hk/artiq/manual/compiler.html) this should be inferred as TInt32. I expect wraping the expression in another `int32()` to fix this.

charlesbaynham, 2020-06-12:
Hmm changing that to 
```
@portable(flags={"fast-math"})
    def turns_to_pow(self, phase):
        """Returns the 16-bit phase offset word corresponding to the given
        phase.
        """
        return int32(int32(round((1 << 14)*phase)) & 0xffff)
```

changes the error to 

```
WARNING:npl_drivers.device_db_config._non_core_devices_1:Initiating devices in simulation mode
<artiq>\coredevice\ad9912.py:177:28-177:50: error: the type of this expression cannot be fully inferred
        return int32(int32(round((1 << 14)*phase)) & 0xffff)
                           ^^^^^^^^^^^^^^^^^^^^^^
<artiq>\coredevice\ad9912.py:177:28-177:50: note: the expression has type numpy.int?
        return int32(int32(round((1 << 14)*phase)) & 0xffff)
                           ^^^^^^^^^^^^^^^^^^^^^^
```

pathfinder49, 2020-06-12:
Strange 🤔 Just to confirm, this works fine without the & 0xffff? Could you email me your simulation device_db?

Random thought: Do type hints on turns_to_pow make a difference? (They shouldn't make a difference on the core? I'm unfamiliar with the simulation mode. Hopefully it behaves the same.)

charlesbaynham, 2020-06-12:
Yes, it's the mask on l.177 that's the issue: removing that, the kernel compiles. 

Type hints do something! Adding a  `-> TInt32` seems to make it compile. However, the DDS now doesn't output anything... Same with removing the mask. 

You can ignore the simulation mode in the above, that only refers to the peripherals we have in our lab: the core itself is connected. 

Here's the `device_db` FYI: [_device_db_2.txt](https://github.com/m-labs/artiq/files/4772212/_device_db_2.txt)

pathfinder49, 2020-06-12:
The only other ad9912 change in that commit is the mask on line 163. Does commenting both changes resore normal operation?

charlesbaynham, 2020-06-15:
Yes, using artiq as in https://github.com/charlesbaynham/artiq/tree/hack_AD9912 fixes things. I assume you put them there for a reason though!

pathfinder49, 2020-06-15:
The masks were added so that only valid codes are returned from by the SI unit to mu conversion. Previously, entering an out of range Si number would return an invalid code.

pathfinder49, 2020-06-16:
@charlesbaynham Unfortunatly, I don't have an ad9912 to test this. Could you run a few tests to see if any of these fix the issue:
- TInt64 type annotation on frequency_to_ftw
- AD9912.bus.write expects int32 inuput, so wrapping line 152 in `int32()` might help? (The way I read the docs, I expect this to already be an int32)

charlesbaynham, 2020-06-16:
It was a combination of frequency_to_ftw and turns_to_pow that did the trick: see #1466

closed 2020-06-16 by jordens