m-labs/artiq #1468: Compiler TInt64 type inferrence breaks silently

pathfinder49, 2020-06-16:
# Bug Report

## One-Line Summary

Compiler TInt64 type inferrence breaks silently. This results in silent conversions to incorrect type-casting/corruption.  Type hints help in some, but not all cases.

## Issue Details
There are serveral related behaviours that are quite syntax sensitive. The two listed experiments demonstrate the bugs. #1466 seems to narrowly avoid hitting this (see example 3).

The bug seems to be todo with type inferrence of in64 vs int32 under a range of conditions. It seems like a combination of factors is required to produce the bug, but I have been unable to nail thios down.

All of the listed experiments are broken and erroneously print: 
```Python3
0  # this should print 1
0
```
Implementation details significantly change the behaviour around this bug and what actions resolve the bug.

### Example 1 - type casting in run
```Python3
from artiq.experiment import *
import numpy as np

class CompilerBugToError(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        c0 = self.case0()
        print(np.int32(c0>>32))
        print(np.int32(c0))  # commenting this line raises compiler error!

    @kernel()
    def case0(self):  # bug NOT fixed by type hint
        return np.int64(2**32) & ((np.int64(1)<<48) - 1)   # commenting & ... does NOT fix the bug
```
### Example 2 - type casting in a function
```Python3
from artiq.experiment import *
import numpy as np

class CompilerNoError(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        a = self.case0()
        self.kfun(a)

    @kernel
    def kfun(self, c0):
        print(np.int32(c0>>32))
        print(np.int32(c0))  # commenting does not raise error

    @kernel()
    def case0(self):  # adding TInt64 type-hint does NOT fix
        return np.int64(2**32) & ((np.int64(1)<<48) - 1)  # commenting &... does NOT fix
```
### Example 3 - adding a function argument
```Python3
from artiq.experiment import *
import numpy as np

class CompilerTypeHintFix(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        a = self.case0(2.**32)
        self.kfun(a)

    @kernel
    def kfun(self, c0):
        "Broken if cast to int32 twice"
        print(np.int32(c0>>32))
        print(np.int32(c0))  # commenting or adding some bit-shift avoids bug

    @kernel()
    def case0(self, float_arg):  # adding TInt64 type-hint fixes
        return np.int64(round(float_arg)) & ((np.int64(1)<<48) - 1)  # commenting &... fixes
```

pathfinder49, 2020-06-16:
This bug was encountered in #1463

pathfinder49, 2020-09-28:
We've just upgraded one of our experiments from an approximatly 10 month old version of artiq. (I'll follow with more details when known.) During this time the Int32/64 type inference must have changed as code that ran before now throws `cannot unify numpy.int32 with numpy.int64: 32 is incompatible with 64` errors. Looking over the [release notes](https://github.com/m-labs/artiq/blob/master/RELEASE_NOTES.rst) I can't see anything suggesting a change in this area. It seems like type inference may have been unwittingly changed. This may be realted to the above.

pathfinder49, 2020-10-02:
Digging through issues, I just came across https://github.com/m-labs/artiq/issues/1242. This looks like were hitting the same issue. The difference here is that the behaviour is very code structurede dependant.

sbourdeauducq, 2021-12-13:
NAC3 does not have this problem.

dnadlinger, 2021-12-13:
I wonder whether it's worth keeping these issues open as long as they occur on master to avoid duplicate reports/time wasted tracking them down again. (A label could be used to mark them for closure once NAC3 has replaced the old compiler.)

closed 2021-12-13 by sbourdeauducq

dnadlinger, 2021-12-13:
Fair enough – by default, closed issues are not shown, but I guess users would typically need to include those anyway, as they wouldn't be using the absolutely most recent master version.

dhslichter, 2021-12-13:
I imagine if people are searching they will find closed issues as well.  However, just for information, @sbourdeauducq do you have an estimate for when NAC3 will be merged into the master branch?  Is this 6 months away, more, less?  A rough estimate is fine.  Certainly we will have to study all of the consequences of the upgrade in terms of backwards incompatibility -- a lot of code rewriting will have to occur, so good to be able to plan.