m-labs/artiq #1469: Installing on Windows fails due to llvmlite_artiq

charlesbaynham, 2020-06-17:
# Bug Report

## One-Line Summary

It's another packaging problem on Windows / Conda! Installing artiq into a new environment using the `install-with-conda.py` script results in a broken installation: calling e.g. `artiq_run` results in 

```
(artiq) c:\artiq>artiq_run
Traceback (most recent call last):
  File "C:\Users\me\.conda\envs\artiq\lib\site-packages\llvmlite_artiq\binding\ffi.py", line 119, in <module>
    lib = ctypes.CDLL(os.path.join(_lib_dir, _lib_name))
  File "C:\Users\me\.conda\envs\artiq\lib\ctypes\__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\me\.conda\envs\artiq\Scripts\artiq_run-script.py", line 5, in <module>
    from artiq.frontend.artiq_run import main
  File "C:\Users\me\.conda\envs\artiq\lib\site-packages\artiq\frontend\artiq_run.py", line 13, in <module>
    from llvmlite_artiq import binding as llvm
  File "C:\Users\me\.conda\envs\artiq\lib\site-packages\llvmlite_artiq\binding\__init__.py", line 6, in <module>
    from .dylib import *
  File "C:\Users\me\.conda\envs\artiq\lib\site-packages\llvmlite_artiq\binding\dylib.py", line 4, in <module>
    from . import ffi
  File "C:\Users\me\.conda\envs\artiq\lib\site-packages\llvmlite_artiq\binding\ffi.py", line 124, in <module>
    lib = ctypes.CDLL(_lib_name)
  File "C:\Users\me\.conda\envs\artiq\lib\ctypes\__init__.py", line 351, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found
```
## Issue Details

### Steps to Reproduce

0. Use Windows 10
1. Run install script
2. Execute `artiq_run`

### Expected Behavior

Works. 

### Actual (undesired) Behavior

Error message shown above. 

Inspecting the contents of the new environment and comparing with an old one that I already had (which works) shows that `C:\Users\me\.conda\envs\artiq\Lib\site-packages\llvmlite_artiq\_version.py` differs: 

#### Broken version of llvmlite_artiq's _version.py

``` python
# [...]

version_version = '0+unknown'
version_full = 'unknown'
def get_versions(default={}, verbose=False):
    return {'version': version_version, 'full': version_full}
```

#### Working version of llvmlite_artiq's _version.py

``` python
# [...]

version_version = '0.6.0+600.g1d167be'
version_full = '1d167be4eec5d6c32498952be8b3ac17dd30df8d'
def get_versions(default={}, verbose=False):
    return {'version': version_version, 'full': version_full}

```

I think the source of the error is the different .dll in `llvmlite_artiq.binding`. The working env's `.dll` file is 20,763 KB whereas the broken env's is 29,221 KB, so there is a difference. 

### Your System (omit irrelevant parts)

* Operating System:
* ARTIQ version: New installation on 2020/06/17
* Version of the gateware and runtime loaded in the core device: n/a
* If using Conda, output of `conda list`: n/a
* Hardware involved: none

sbourdeauducq, 2020-06-18:
https://git.m-labs.hk/M-Labs/nix-scripts/issues/27

Install the old package from the m-labs channel on anaconda.org for now (I thought conda would look at build numbers but apparently not).

Could you kindly install MSYS2 (with the official installer on https://www.msys2.org/, not the conda packages), and see if MSYS2-PyQt5 works on your machine? I see no other reports of the freezes I am seeing. MSYS2 is technically way superior to Conda and would also allow us to remove Visual Studio from the process entirely.

Running a hello world application e.g. https://pythonbasics.org/pyqt-hello-world/ is sufficient to crash it (the program just freezes at the creation of the first widget).

sbourdeauducq, 2020-06-18:
https://git.m-labs.hk/M-Labs/nix-scripts/commit/9479f297fc57ad31c48fa04facdb4754b1786141

closed 2020-06-18 by sbourdeauducq

charlesbaynham, 2020-06-18:
Hi @sbourdeauducq, thanks for the help

I ran the following in a msys2 terminal on Windows 10:
```
$ pacman -S --disable-download-timeout mingw64/mingw-w64-x86_64-python3-pyqt5
```

(The flag is required due to https://github.com/msys2/MSYS2-packages/issues/1658, I think I might have an old version of MSYS2 installed.)

I then ran `python3 test_pyqt.py` with the file you copied above and received:

[image]

So looks like it's working!

charlesbaynham, 2020-06-18:
I also tried another conda artiq installation to see if that commit to the hydra build has fixed things, but I'm still getting the same error.

sbourdeauducq, 2020-06-18:
> I then ran `python3 test_pyqt.py` with the file you copied above and received:

Thanks! I don't know why it fails here then...

> I also tried another conda artiq installation to see if that commit to the hydra build has fixed things, but I'm still getting the same error.

Are you on the stable channel (conda.m-labs/artiq and not /artiq-beta)?
Note that the conda cache mechanism is broken (moving to MSYS2 and Pacman would fix that) and you may also need to delete some files manually.

charlesbaynham, 2020-06-18:
> Are you on the stable channel (conda.m-labs/artiq and not /artiq-beta)?
> Note that the conda cache mechanism is broken (moving to MSYS2 and Pacman would fix that) and you may also need to delete some files manually.

I was on the artiq-beta channel: I'm trying again now without any of the `conda.m-labs.hk` channels:

```
if ADD_CHANNELS:
    run("conda config --prepend channels m-labs")
    # run("conda config --prepend channels https://conda.m-labs.hk/artiq-beta")
    run("conda config --append channels conda-forge")

# Creating the environment first with python 3.5 hits fewer bugs in conda's broken dependency solver.
run("conda create -y -n {CONDA_ENV_NAME} python=3.5".format(CONDA_ENV_NAME=CONDA_ENV_NAME))
for package in CONDA_PACKAGES:
    # Do not activate the environment yet - otherwise "conda install" may not find the SSL module anymore on Windows.
    # Installing into the environment from the outside works around this conda bug.
    # run("conda install -y -n {CONDA_ENV_NAME} -c https://conda.m-labs.hk/artiq-beta {package}"
    #     .format(CONDA_ENV_NAME=CONDA_ENV_NAME, package=package))

   	run("conda install -y -n {CONDA_ENV_NAME} -c m-labs {package}"
        .format(CONDA_ENV_NAME=CONDA_ENV_NAME, package=package))
```

sbourdeauducq, 2020-06-18:
You do need the /artiq channel.

sbourdeauducq, 2020-06-18:
> Thanks! I don't know why it fails here then...

Do you know what MSYS version you have installed?

sbourdeauducq, 2020-06-18:
And what Windows version? It's Win10 here.

charlesbaynham, 2020-06-18:
> 
> 
> You do need the /artiq channel.

Yep, adding that the installation now works.

charlesbaynham, 2020-06-18:
> 
> 
> And what Windows version? It's Win10 here.

Windows 10 here too:
```
c:\artiq>python -c "import sys;print(sys.getwindowsversion())"
sys.getwindowsversion(major=10, minor=0, build=18362, platform=2, service_pack='')
```

charlesbaynham, 2020-06-18:
> 
> 
> > Thanks! I don't know why it fails here then...
> 
> Do you know what MSYS version you have installed?

MSYS doesn't really have a specific version since it's just a collection of packages. Here's a dump of all of mine:
[msys_versions.txt](https://github.com/m-labs/artiq/files/4797960/msys_versions.txt)

sbourdeauducq, 2020-06-20:
PyQt 5.12 that you have installed is over a year old. Can you confirm if the latest version works for you? I'm trying to downgrade here and see what happens.

sbourdeauducq, 2020-06-20:
Still fails...

sbourdeauducq, 2020-06-20:
Ok I found the problem. Creating a QApplication object breaks the Python REPL and commands you enter there are no longer transmitted to Python, which gives the impression that it has crashed. It also works here when running the Qt application from a file and not copy-pasting code into the REPL.

sbourdeauducq, 2020-06-20:
ARTIQ browser running in MSYS2 without conda

[image]

charlesbaynham, 2020-06-21:
> Ok I found the problem. Creating a QApplication object breaks the Python REPL and commands you enter there are no longer transmitted to Python, which gives the impression that it has crashed. It also works here when running the Qt application from a file and not copy-pasting code into the REPL.

How annoying... Congrats on getting it going!

If you're intending to move over to MSYS2 entirely does that mean you'd be dropping virtual environments?

sbourdeauducq, 2020-06-21:
Yes, Pacman does not support environments.

charlesbaynham, 2020-06-21:
Hmmm that would be a shame to lose. Personally I'd only use MSYS for artiq so it would be fine, but being able to isolate your dependencies is a feature that I think many would miss.