m-labs/artiq #1471: Conda -> MSYS2 on Windows

sbourdeauducq, 2020-06-20:
Currently, Windows support in ARTIQ relies on Conda and Visual Studio. These two software packages are very problematic:
* Conda is slow. It takes a ridiculous amount of time to install a few packages.
* Conda is buggy. In particular the dependency solver is so broken that Conda should not deserve the name of "package manager".
* The Conda codebase is full of hacks and difficult to understand and modify without breaking something.
* Conda is very poorly architected/designed. Here is an example: https://github.com/conda/conda-build/issues/695
* The Visual Studio installer is slow (over an hour installation time on a fast system) and Visual Studio is bloated (over 5GB for minimal install, 35GB for full install).
* The Visual Studio installer is buggy and unreliable. In particular, when trying to use the command line, the installation often silently fails if you give it a combination of options it dislikes.
* Visual Studio installations (like [Windows 10](http://itvision.altervista.org/why-windows-10-sucks.html)) are non-reproducible. Everything has to be downloaded from Microsoft and changes all the time with no proper version control.
* Visual Studio is generally [a mess](https://hackernoon.com/a-c-hello-world-and-a-glass-of-wine-oh-my-263434c0b8ad).
* After days of trying, I still cannot get CMake (used to build LLVM-or1k) to find Visual Studio.

By comparison:
* MSYS2 comes with the Pacman package manager from Arch Linux, which is fast and robust, and where dependency resolution actually works.
* It is easy to compile software inside MSYS2, it is simply a matter of installing the GCC toolchain with Pacman and then things work pretty much out-of-the-box (including LLVM-or1k).  (NB: compiling Conda-packaged software with MSYS2 seems quite problematic due to libraries not being found)
* Much faster installation times for everything, and much less bloat.

The main drawbacks of MSYS2, as far as I can tell, is some packages are missing (e.g. quamash, pyqtgraph) and there is no environment support (but do those actually work reliably with Conda anyway?).

It seems to me that a saner strategy is to fix whatever MSYS2 issues there are instead of continuing to invest resources into Conda/Visual Studio. Lots of money has been poured already into Conda (Continuum Analytics, NumFocus) and VS (Microsoft) with such a shoddy final result, so those tools seem like a dead-end to me.

So what I would like to support as installation methods is:
1. Nix on Linux
2. MSYS2 on Windows
3. Maybe Conda on Linux, since Conda is actually tolerable when the builds are run inside Nix sandboxes (e.g. if Conda corrupts its database, as it often does, it won't break other builds), and there isn't the additional toxicity of Windows and especially Visual Studio.

dhslichter, 2020-06-21:
Oof.  As horrible as it may be in various ways, Conda is definitely a de-facto "standard" for a lot of people using Python for scientific work.  This is not saying that this is the best or "right" way to do it, but that I think a large share of physicists using Python are doing so through Conda.  It's kind of like Powerpoint, or Word.  They are not the best, and they have irritating features, but they are pretty much universal.  One has to be careful not to throw out the baby with the bathwater.  

I agree that there are lots of issues with Conda, and plenty with Visual Studio too.  However, before we decide to change over to something else, I would want to know exactly how things would work.  For example, having environments in Conda is useful for being able to install a new version of ARTIQ to test without having to remove/destroy the current working version.  Is this something that would be addressed?  I think this is a very standard occurrence.  How much time is going to be spent re-inventing various wheels, bringing in packages that are missing, dealing with all the yak-shaving that inevitably will occur with MSYS2 as well?  

NIST pays a maintenance contract that specifically calls out maintaining Conda packages for Windows.  The rationale behind this is that, despite that fact that Conda is frustrating and complex for the people packaging ARTIQ, it generally works fine on the user end, and it makes it much easier for users to get going and to operate.  Because it is a feature for users, but a problematic/time-consuming process for those doing the development and CI work, it makes sense for the users to pay the developers for their time and effort in this direction.  If the state of Conda has become dramatically worse (doesn't seem like it's worse, just more of the same?) then the terms of such a maintenance contract can be renegotiated.

dhslichter, 2020-06-21:
To be clear: I am open to the MSYS2 proposal for Windows.  I am just hesitant to throw all our effort into a new method of doing things until there is more information about what the roadblocks will be, and most importantly how it will impact ARTIQ users.  I think it will require some further thought and demonstrations to show that it would in fact do all the things we would like it to do, and more reliably than the current solution.  Again, for sad IT policy reasons, it's important to maintain a Windows version of ARTIQ.

sbourdeauducq, 2020-06-21:
A few bits of information:
* Pacman has a rollback mechanism: https://wiki.archlinux.org/index.php/Arch_Linux_Archive. Note that conda environments (where things are put into different folders instead of replaced) are implemented with a rather gross binary patching hack, which is - for once - defensible since there isn't really a portable way to do it otherwise. But if package management is reliable (which it is with Pacman and not with Conda), there should be less of a need for environments.
* I have used Pacman with Arch Linux for years and do not remember any Pacman bug.
* The MSYS2 project has archives of packages that date back from at least 2015.
* Setting up a mirror, if desired, is very easy with rsync: https://www.msys2.org/wiki/Mirrors/ (good luck trying to mirror Anaconda.org on the other hand - and no, cas-mirror won't work).
* I tried installing a MSYS version from a year ago and everything worked correctly, it found and downloaded the older packages.

MSYS also has straightforward benefits for users, such as speed and reliability.

sbourdeauducq, 2020-06-21:
A bit hacky but this works:
https://gist.github.com/bparker06/260209ff3260c9b78b6ea2769ed13a69
followed with https://wiki.archlinux.org/index.php/offline_installation_of_packages

sbourdeauducq, 2020-06-21:
ARTIQ browser running in MSYS without Conda:
[image]
(IIRC the "uncaught exception" error is a Python 3.8 problem)

dtcallcock, 2020-06-21:
A few questions:

- Could a Conda installation (for data analysis etc.) live happily on the same machine? 
- If Conda on Linux is acceptable, is using that and WLS2 a solution?  Not sure WLS2 would meet NIST security requirements though as it's technically a VM.

> The main drawbacks of MSYS2, as far as I can tell, is some packages are missing (e.g. quamash, pyqtgraph)

How hard is it to add them?

sbourdeauducq, 2020-06-21:
>     * Could a Conda installation (for data analysis etc.) live happily on the same machine?

Yes.

>     * If Conda on Linux is acceptable, is using that and WLS2 a solution?  Not sure WLS2 would meet NIST security requirements though as it's technically a VM.

With WSL if you want a GUI you need to install a X server on the native Windows side, which is cumbersome, slow, buggy, and with poor usability. AFAIK this has not been addressed in WSL2.

> > The main drawbacks of MSYS2, as far as I can tell, is some packages are missing (e.g. quamash, pyqtgraph)
> 
> How hard is it to add them?

Those two are straightforward.

dtcallcock, 2020-06-21:
Could you just run through, in lay terms, what the following would look like in this future MSYS2 utopia:

- Installation of artiq
- Upgrade of artiq
- Clean re-install if everything ends up horribly broken

> With WSL if you want a GUI you need to install a X server on the native Windows side

Or leave the GUI on windows and just run the Master in WSL?  Sounds hacky though.

sbourdeauducq, 2020-06-22:
>     * Installation of artiq

We could probably make a modified MSYS installer that includes the ARTIQ packages. Then it's a traditional Windows GUI installer; it would also be self-contained and not require an internet connection.

If you already have MSYS, add our package source and then ``pacman -S artiq``. 

>     * Upgrade of artiq

``pacman -Syu`` in a MSYS shell.

>     * Clean re-install if everything ends up horribly broken

MSYS can be uninstalled from the Windows control panel.

It also seems you can make and restore backups of the MSYS state: https://www.msys2.org/wiki/MSYS2-reinstallation/ (see "Reversing the procedure")

sbourdeauducq, 2020-06-22:
> MSYS also has straightforward benefits for users, such as speed and reliability.

Regarding reliability: for example, pacman does not break with useless error messages such as the one below, which are very common with conda.

```
(artiq) wfvm@DESKTOP-5UAUJMN C:\Users\wfvm>conda install artiq
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: |
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

(artiq) wfvm@DESKTOP-5UAUJMN C:\Users\wfvm>
```

sbourdeauducq, 2020-06-22:
Here is also one of many examples of nonsense from the Conda dependency solver. This kind of scenario is also likely to occur on user machines.

```
>conda create -n artiq artiq python=3.8
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: /
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package python conflicts for:
artiq -> h5py -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0|>=3.5,<3.6.0a0|3.4.*']
python=3.8

(artiq) wfvm@DESKTOP-5UAUJMN C:\Users\wfvm>
```

dhslichter, 2020-06-22:
I understand that there are many issues with Conda.  I am not wedded to Conda as the solution.  My concerns are basically along the lines of what David has said:

1. There needs to be a simple, user-friendly (including non-*nix-user-friendly) method to install ARTIQ on Windows.  A standalone installer is a fine solution if it works reliably.
2. There needs to be a way to reliably roll back and upgrade ARTIQ versions on a given computer without damaging functionality, overwriting/deleting the repository, etc.
3. It must be possible to have Anaconda (or some other Python distribution) on the same computer (e.g. for data analysis) as ARTIQ, without either one interfering with the other (including when installing/de-installing either).  

> conda's free-for-all reliance on anaconda.org.

Is this not fixed by just keeping an appropriate set of packages in the m-labs channel so that they are not randomly broken by changes elsewhere?  And when installing ARTIQ, one puts the m-labs channel at highest priority?  I may be misunderstanding something here.  But if MSYS is going to require M-Labs providing additional missing packages already...

sbourdeauducq, 2020-06-23:
> Is this not fixed by just keeping an appropriate set of packages in the m-labs channel so that they are not randomly broken by changes elsewhere? And when installing ARTIQ, one puts the m-labs channel at highest priority?

No, for example, there would be a problem if the user wants to install something that depends partially on packages found in the m-labs channel and partially on others.

> I may be misunderstanding something here. But if MSYS is going to require M-Labs providing additional missing packages already...

We have already been providing additional missing packages for Conda, and even more so in the past (pyqtgraph, levenshtein, pygit2, etc.)
And it looks like packaging and maintaining something for MSYS2 would be 1000x less trouble than for Conda anyway.

dtcallcock, 2020-06-23:
Thanks for answering the dumb questions @sbourdeauducq 

> The main drawbacks of MSYS2, as far as I can tell, is some packages are missing (e.g. quamash, pyqtgraph) 

- Is there a list of the packages included somewhere?  Google just gives me PacMan games written in Python.
- If they are included, does this mean they are updated and maintained, or just that someone added some random version one time?
- If someone needs some other package for their ARTIQ code that isn't included, how do they go about using it (assuming average grad student level skillz)?

> We could probably make a modified MSYS installer that includes the ARTIQ packages.

What would a proof of principle that all this works as advertised take?  Is it reasonable to give it a go and see if people like it, or would it require new contracts and/or significant development time before we'd have the whole picture?

sbourdeauducq, 2020-06-23:
> * Is there a list of the packages included somewhere?  Google just gives me PacMan games written in Python.

https://packages.msys2.org/search

> * If they are included, does this mean they are updated and maintained, or just that someone added some random version one time?

They are updated and maintained.

> * If someone needs some other package for their ARTIQ code that isn't included, how do they go about using it (assuming average grad student level skillz)?

For dirty installations, pip is available.
For making a Pacman package, consider that the student won't be dealing with a constant stream of toolchain bugs as they would with Conda, and instead will use a much more solid and consistent packaging system.

dhslichter, 2020-06-24:
@sbourdeauducq thanks for your answers.  I am willing to give MSYS2 a try as the new method for installation on Windows.  I suggest we do a trial run (public beta?) and have people test it to make sure it meets their needs.  If it works and we decide to officially shift away from conda, then we can update the maintenance contract language accordingly.  

Do you have a sense of the breadth of the Windows ARTIQ user base, and/or a good method of contacting groups?  Making sure that current Windows ARTIQ users have a chance to weigh in, and to participate in a beta test, will be crucial.

dtcallcock, 2021-04-05:
I just spotted this [Mamba](https://github.com/mamba-org/mamba) business: https://medium.com/@QuantStack/open-software-packaging-for-science-61cecee7fc23

It seems to address some of the issues with Conda mentioned in the top post (tagline: "making Conda fast again").  No idea if it's suitable for us or if you've already taken a look @sbourdeauducq.

sbourdeauducq, 2021-04-05:
Looks good! I just think they're giving conda too much credit - the conda dependency solver is anything but "very efficient", and it's not even correct - but at least they're using the halfway-decent RPM solver in Mamba now.
I would still like a way to get rid of Visual Studio which is really one nasty piece of junk; and ideally we could cross-compile everything on Linux using mingw (MSYS2 also does not allow that, sadly)...

sbourdeauducq, 2022-01-24:
Dashboard, master and applets working... (The errors in the log are due to the yet incomplete NAC3 port of ARTIQ and are not specific to Windows)
[image]
NAC3 itself is cross-compiled with mingw from Linux; no more VS at all. Installation is fairly messy at the moment but it looks fixable.

sbourdeauducq, 2022-04-05:
ARTIQ with NAC3 can now be installed automatically with MSYS2 (just add the https://lab.m-labs.hk/msys2 repos and then ``pacman -S mingw-w64-x86_64-artiq``). From a cursory test, major ARTIQ features are OK.

One major advantage of MSYS2 is it is entirely free from Visual Studio trashware, and software can even be compiled and tested in Wine without roadblocks. This makes development a lot quicker and much less frustrating.

closed 2022-04-05 by sbourdeauducq