m-labs/artiq #1473: Visual Studio llvmlite-artiq build breakage

sbourdeauducq, 2020-06-22:
Using the old binary package to work around #1472, still no dice...
```
%BUILD_PREFIX%\Library\include\llvm/Support/Chrono.h(37): warning C4141: 'inline': used more than once [%SRC_DIR%\ffi\build\llvmlite.vcxproj] 
%BUILD_PREFIX%\Library\include\llvm/Support/Chrono.h(44): warning C4141: 'inline': used more than once [%SRC_DIR%\ffi\build\llvmlite.vcxproj] 
  Generating Code... 
LINK : fatal error LNK1181: cannot open input file 'C:\Program Files (x86)\Microsoft Visual Studio 14.0\DIA SDK\lib\amd64\diaguids.lib' [%SRC_DIR%\ffi\build\llvmlite.vcxp
roj]
Traceback (most recent call last): 
Trying generator 'Visual Studio 15 2017 Win64' 
  File "C:\Users\wfvm\Anaconda3\conda-bld\llvmlite-artiq_1592663949138\work\ffi\build.py", line 145, in <module>
    main()
  File "C:\Users\wfvm\Anaconda3\conda-bld\llvmlite-artiq_1592663949138\work\ffi\build.py", line 133, in main
    main_win32()
  File "C:\Users\wfvm\Anaconda3\conda-bld\llvmlite-artiq_1592663949138\work\ffi\build.py", line 80, in main_win32
    subprocess.check_call(['cmake', '--build', build_dir, '--config', config])
  File "C:\Users\wfvm\Anaconda3\conda-bld\llvmlite-artiq_1592663949138\_h_env\lib\subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', 'C:\\Users\\wfvm\\Anaconda3\\conda-bld\\llvmlite-artiq_1592663949138\\work\\ffi\\build', '--config', 'Release
']' returned non-zero exit status 1.
error: command 'C:\\Users\\wfvm\\Anaconda3\\conda-bld\\llvmlite-artiq_1592663949138\\_h_env\\python.exe' failed with exit status 1
```

sbourdeauducq, 2020-06-22:
The correct location seems to be ``C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\DIA SDK``

dnadlinger, 2020-06-22:
Are you launching this from withing VsDevCmd.bat or outside?

dnadlinger, 2020-06-22:
(or vcvars32.bat, for that matter)

sbourdeauducq, 2020-06-23:
``conda build`` loads such scripts, in addition to way too many VS setup commands...

sbourdeauducq, 2020-06-23:
We have automated and reproducible builds now. Full log: https://nixbld.m-labs.hk/build/69329/nixlog/4

sbourdeauducq, 2020-06-23:
Could be yet another conda bug?
```
(base) wfvm@DESKTOP-514UDU6 %SRC_DIR%>call "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" amd64
The system cannot find the path specified.
```

sbourdeauducq, 2020-06-23:
Note that the same error ``vcvarsall.bat`` error occurs in the llvm-or1k build, where it does not seem to cause such problems. https://nixbld.m-labs.hk/build/69317/nixlog/1

sbourdeauducq, 2020-06-23:
Using the freshly compiled LLVM package fixed this (yeah makes a lot of sense).

closed 2020-06-23 by sbourdeauducq