m-labs/artiq #1475: How to check DHCP status?

cyqian97, 2020-06-24:
# Question
How to check DHCP status?

## Category: networks

## Description
I was trying to connect the Kasli board to a Ethernet port on the wall and control it from remote. The IT department has assigned an ip address to the mac address of the board, but when I connected it I was still not able to ping its ip. The artiq_dashboard was not working also. I was thinking that this might be caused by the DHCP setting but I don't know how to check it. I will be very appreciated if you can give me some advice on my problem!

dnadlinger, 2020-06-24:
DHCP is not supported on the core device (Kasli).

dnadlinger, 2020-06-24:
(Use something like `artiq_mkfs storage.img -s mac <…> -s ip <…>` and `artiq_flash … -f storage.img storage start` to configure the IP address; there is probably documentation somewhere.)

sbourdeauducq, 2020-06-25:
DHCP support is https://github.com/m-labs/artiq/issues/1085

closed 2020-06-25 by sbourdeauducq

cyqian97, 2020-06-25:
> 
> 
> DHCP is not supported on the core device (Kasli).

Thank you very much for your quick reply! I guess then I have to connect Kasli to my PC.