m-labs/artiq #1478: 'With' statement with multiple items has incorrect behavior

lriesebos, 2020-06-30:
# Bug Report

## One-Line Summary

When using a `with` statement with multiple items, the behavior is incorrect.

## Issue Details

### Steps to Reproduce

```python
from artiq.experiment import *

class MyContext:
    def __init__(self):
        self.count = 0

    @portable
    def __enter__(self):
        print('enter')
        self.count += 1

    @portable
    def __exit__(self, exc_type, exc_val, exc_tb):
        print('exit')
        self.count -= 1

class MinimalContextExperiment(EnvExperiment):

    def build(self):
        self.setattr_device('core')
        self.context_a = MyContext()
        self.context_b = MyContext()

    @kernel
    def run(self):
        with self.context_a:
            with self.context_b:
                print('in both context, multiple with statements')
        print(self.context_a.count)
        print(self.context_b.count)

        with self.context_a, self.context_b:
            print('in both context, single with statement')
        print(self.context_a.count)
        print(self.context_b.count)

```

### Expected Behavior

The first two nested `with` statements should be semantically equivalent to the last `with` statement with two items.
Therefore, the printed results should also be the same (two times a `0`).

### Actual (undesired) Behavior

Exit functions are called incorrectly:
```
INFO:worker(2472,minimal_context_experiment.py):print:enter
INFO:worker(2472,minimal_context_experiment.py):print:enter
INFO:worker(2472,minimal_context_experiment.py):print:in both context, multiple with statements
INFO:worker(2472,minimal_context_experiment.py):print:exit
INFO:worker(2472,minimal_context_experiment.py):print:exit
INFO:worker(2472,minimal_context_experiment.py):print:0
INFO:worker(2472,minimal_context_experiment.py):print:0
INFO:worker(2472,minimal_context_experiment.py):print:enter
INFO:worker(2472,minimal_context_experiment.py):print:enter
INFO:worker(2472,minimal_context_experiment.py):print:in both context, single with statement
INFO:worker(2472,minimal_context_experiment.py):print:exit
INFO:worker(2472,minimal_context_experiment.py):print:exit
INFO:worker(2472,minimal_context_experiment.py):print:1
INFO:worker(2472,minimal_context_experiment.py):print:-1
```

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 20.04 (using nix)
* ARTIQ version: ARTIQ v5.7114.db2d9002
* Version of the gateware and runtime loaded in the core device: gateware and software 5.7072.029c6919
* Hardware involved: KC705

sbourdeauducq, 2020-11-22:
Reverted due to #1547.
We should also have regression tests for all these things...

lriesebos, 2020-12-02:
I am not sure if a fix is planned, but an alternative could be to throw a compiler exception when using a `with` statement with multiple contexts, basically not supporting the syntax. That will at least prevent silent incorrect behaviour. Unfortunately I am not familiar enough with the ARTIQ compiler at this moment to propose a coded solution myself.

PR #1526 (2020-10-07): fixes with statement with multiple items
# ARTIQ Pull Request

## Description of Changes
Fixes incorrect behavior for with statements with multiple items.

It seems that the lambda expression is lazily evaluated, and the exit function that it binds to changes to the one of the last item.

### Related Issue
Closes #1478

## Type of Changes

|   | Type |
| ------------- | ------------- |
| ✓  | :bug: Bug fix  |

## Steps (Choose relevant, delete irrelevant before submitting)

### All Pull Requests

- [x] Use correct spelling and grammar.
- [ ] Close/update issues.

### Code Changes

- [ ] Run `f [... cut]
commit 88d346fa (2020-10-07): fixes with statement with multiple items

Closes #1478
commit e23ed154 (2020-10-07): fixes with statement with multiple items

Closes #1478