m-labs/artiq #1481: Fastino wide interface configuration

pathfinder49, 2020-07-04:
# ARTIQ Feature Request
Add configuration option for Fastino interface for use in `gateware.target`

## Problem this request addresses

Reading the source code, it looks like Fastino is hard coded to use the narrow interface. This arragnement forces all Fastino boards in an installtion to use the same interface width. 
https://github.com/m-labs/artiq/blob/f265976df62ec5bc358de68f8808ade96198ae15/artiq/gateware/eem.py#L621-L626

## Describe the solution you'd like
The wide interface options like `log2_width` should be arguments to `gateware.eem.Fastino.add_std()`

pathfinder49, 2020-07-14:
@jordens Is it correct that `log2_width=5` is the only change needed to use the wide interface with the current implementation?

jordens, 2020-07-14:
No. You also need to use a different API in kernels. See the documentation and the relevant [commit](https://github.com/m-labs/artiq/commit/e803830b3b3cc791fd98425ac30a4efb1c210485#diff-6c527b06ec83bd78f3a1e9b785d616d6)

pathfinder49, 2020-07-14:
Thanks, so `log2_width=5` to build the correct kasli gateware and the "group" api for writing to Fastino?

jordens, 2020-09-03:
This setting was added for you to evaluate, test, and characterize the impact on RTIO fabric and performance of wide vs narrow RTIO channels. Before making it a stable API with documentation, integration, tests etc. could you share the results. Also depending on the results it might be much better to just choose one value.

hartytp, 2020-09-21:
@jordens my preference would be to keep the wide RTIO interface even if it doesn't work with DMA for now.

commit c55f2222 (2020-09-22): fastino: documentation and eem pass-through

* Repeat information about matching log2_width a few times
  in the hope that people read it. #1518
* Pass through log2_width in kasli_generic json. close #1481
* Check DAC value range. #1518
closed 2020-09-22 by jordens