m-labs/artiq #1486: ARTIQ GUIs (Nix): add wrapQtAppsHook

drewrisinger, 2020-07-16:
# Bug Report

## One-Line Summary

ARTIQ GUI apps on Nix should be wrapped with ``wrapQtAppsHook``.

## Issue Details

ARTIQ GUI apps have the following errors when run from command-line on WSL (Ubuntu on Win 2004, using Nix).
I would report in https://git.m-labs.hk, but registering accounts are locked there. From Nix, you need to wrap the binaries with ``wrapQtAppsHook``: https://nixos.org/nixpkgs/manual/#sec-language-qt

NOTE: I tried this on Ubuntu 18.04 (native), and didn't have this same error message.

### Steps to Reproduce

1. Create artiq env similar to one in ARTIQ 5 Manual, called ``test-artiq-qt.nix``:
```
{ pkgs ? import <nixpkgs> { }
, artiq-full ? import <artiq-full> { inherit pkgs; }
}:

pkgs.mkShell { 
  buildInputs = [
    (pkgs.python3.withPackages(ps: [artiq-full.artiq]))
  ];
}
```
2. ``nix-channel --add https://nixbld.m-labs.hk/channel/custom/artiq/full/artiq-full && nix-channel --add https://nixos.org/channels/nixos-20.03 nixpkgs``
3. ``nix-shell test-artiq-qt.nix --run "artiq_dashboard"``

### Expected Behavior

Should fail similar to 
```
WARNING:dashboard:quamash.QEventLoop:Event callback failed: [WinError 1225] The remote computer refused the network connection
[... traceback ...]
```

### Actual (undesired) Behavior

Log messages (on WSL):
```bash
# From ARTIQ 5 Manual: 
$ nix-shell my_artiq_env.nix 
$ artiq_dashboard
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted
```

Message on Ubuntu 18.04 (Native):
```bash
(process:18199): Gtk-WARNING **: 16:35:13.153: Locale not supported by C library.
	Using the fallback 'C' locale.
Gtk-Message: 16:35:13.170: Failed to load module "canberra-gtk-module"
Gtk-Message: 16:35:13.170: Failed to load module "canberra-gtk-module"
Qt: Session management error: None of the authentication protocols specified are supported
```

### Resolution

Modify this file: https://git.m-labs.hk/M-Labs/nix-scripts/src/commit/cc00cfc1a8bf16b6a7e25b2c8a1187b45c8efb3b/artiq-fast/pkgs/artiq.nix

Should add something like:
```nix
{..., wrapQtAppsHook, ...}:

buildPythonPackage{
  nativeBuildInputs = [... wrapQtAppsHook ...];
  dontWrapQtApps = true;
  postFixup = ''
    # TODO: repeat for each ARTIQ GUI program
    wrapQtApp "$out/bin/artiq_dashboard"
  '';
}
```

### Your System

* Operating System: Windows (running Linux via WSL 2.0 running Ubuntu 20.04). Also tested on Ubuntu 18.04.
* ARTIQ version: ARTIQ v5.7118.40dacb72

sbourdeauducq, 2020-07-17:
Thanks.
Is Qt on WSL2 able to use the GUI without a X server?

sbourdeauducq, 2020-07-17:
https://git.m-labs.hk/M-Labs/nix-scripts/commit/e75d16f55aabe179677356d6c66ab83bfbe8a128

closed 2020-07-17 by sbourdeauducq

drewrisinger, 2020-07-17:
I'll get back to you on that. Still testing it. Mostly just using it to test/compile Nix packages locally.

drewrisinger, 2020-07-17:
Also forgot to mention, but this is the tracking issue on Nixpkgs for these sorts of errors: https://github.com/NixOS/nixpkgs/issues/65399

drewrisinger, 2020-07-28:
Ok, after a little bit of hackery/annoying workarounds: I have ARTIQ running on Nix on WSL2 (Windows Subsystem for Linux), with GUI dashboard shown via X11 on Windows host. So no @sbourdeauducq re w/o X server.

Basic steps (a few from memory, might have missed minor details):
1. Install WSL2. https://aka.ms/wsl2-install
1b. Install Ubuntu or other WSL2 distro. Make sure that this is on WSL2: ``wsl.exe --list --verbose``, should show version==2.
2. In WSL: Install Nix: https://nixos.org/download.html
3. In WSL: Install ARTIQ Nix channel & create ARTIQ environment shell/descriptor: https://m-labs.hk/artiq/manual/installing.html#installing-via-nix-linux
4. On Windows: Install an X server & setup firewall.
  a. Install vcXsrv: https://sourceforge.net/projects/vcxsrv/
  b. Setup display variables and firewall rules. Launch Vcxsrv (called _"XLaunch"_ in Windows start menu). Follow directions here: https://stackoverflow.com/a/61110604
  c. Test that directions worked by starting ``xterm`` in WSL (install ``xterm`` if needed via ``sudo apt install xterm``)
  d. If firewall rules aren't working (I tried for a while and couldn't get them working), then follow instructions here: https://github.com/cascadium/wsl-windows-toolbar-launcher#firewall-rules. If still not working, disable Windows Firewall for the WSL network adapter ONLY (usual warnings about disabling firewalls apply here). Follow powershell instructions at end of previous link section.
5. In WSL: Launch ARTIQ master & dashboard.
  a. Start 2x Nix shells with the ARTIQ environment from step 3.
  b. Shell 1: ``artiq_master`` (other arguments as necessary)
  c. Shell 2: ``artiq_dashboard``. ARTIQ dashboard should launch on Windows desktop in a few seconds.

sbourdeauducq, 2020-07-29:
@drewrisinger Thanks for the info!