m-labs/artiq #1487: Kasli: accessing UART on windows

jkellerPTB, 2020-07-20:
# Question
## Category: Documentation

## Description
How can I access the UART port of a Kasli from a windows machine? I tried installing a USB serial port driver for interface 2 of the `Quad-RS232 HS` device, but could not connect to the corresponding COM port (serial connection from a linux system worked, so I assume I'm looking at the correct device).

Side note: `artiq_flash` can't seem to cope with the presence of any `Quad-RS232 HS` devices with drivers other than WinUSB. As soon as I installed any other driver, I got below error message. It goes away if I replace all installed drivers for these devices with WinUSB. If installing a different driver is necessary for accessing UART, this will become a problem.

```
(artiq5release) C:\>artiq_flash -V ptbin
Design: top;UserID=FFFFFFFF;COMPRESS=TRUE;Version=2019.2
Part name: 7a100tfgg484
Date: 2020/04/26
Time: 08:18:17
Bitstream payload length: 0x2ae794
Open On-Chip Debugger 0.10.0-00013-gbb7bedad (2018-02-17-05:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 25000 kHz
Error: libusb_get_string_descriptor_ascii() failed with LIBUSB_ERROR_PIPE
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6011, description 'Quad RS232-HS', serial '*' at bus location '*'

Traceback (most recent call last):
  File "C:\Users\username\.conda\envs\artiq5release\Scripts\artiq_flash-script.py", line 9, in <module>
    sys.exit(main())
  File "C:\Users\username\.conda\envs\artiq5release\lib\site-packages\artiq\frontend\artiq_flash.py", line 460, in main
    programmer.run()
  File "C:\Users\username\.conda\envs\artiq5release\lib\site-packages\artiq\frontend\artiq_flash.py", line 204, in run
    self._client.run_command(cmdline)
  File "C:\Users\username\.conda\envs\artiq5release\lib\site-packages\artiq\remoting.py", line 57, in run_command
    subprocess.check_call([arg.format(tmp=self._tmp, **kws) for arg in cmd])
  File "C:\Users\username\.conda\envs\artiq5release\lib\subprocess.py", line 271, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['openocd', '-s', 'C:\\Users\\username\\.conda\\envs\\artiq5release\\Library\\share\\openocd\\scripts', '-c', 'source [find board/kasli.cfg]; target create xc7.spi0.proxy testee -chain-position xc7.tap; flash bank spi0 jtagspi 0 0 0 0 xc7.spi0.proxy 0x2; gdb_port disabled; tcl_port disabled; telnet_port disabled; init; xadc_report xc7.tap; pld load 0 {C:\\Users\\username\\.conda\\envs\\artiq5release\\share\\bscan-spi-bitstreams\\bscan_spi_xc7a100t.bit}; flash probe spi0; flash erase_sector spi0 0 42; flash write_bank spi0 {C:\\Users\\username\\AppData\\Local\\Temp\\artiq_43d4hc0a_top.bit} 0x0; flash verify_bank spi0 {C:\\Users\\username\\AppData\\Local\\Temp\\artiq_43d4hc0a_top.bit} 0x0; flash probe spi0; flash erase_sector spi0 64 65; flash write_bank spi0 {C:\\Users\\username\\.conda\\envs\\artiq5release\\lib\\site-packages\\artiq\\board-support\\kasli-ptbin\\bootloader.bin} 0x400000; flash verify_bank spi0 {C:\\Users\\username\\.conda\\envs\\artiq5release\\lib\\site-packages\\artiq\\board-support\\kasli-ptbin\\bootloader.bin} 0x400000; flash probe spi0; flash erase_sector spi0 69 78; flash write_bank spi0 {C:\\Users\\username\\.conda\\envs\\artiq5release\\lib\\site-packages\\artiq\\board-support\\kasli-ptbin\\runtime.fbi} 0x450000; flash verify_bank spi0 {C:\\Users\\username\\.conda\\envs\\artiq5release\\lib\\site-packages\\artiq\\board-support\\kasli-ptbin\\runtime.fbi} 0x450000; xc7_program xc7.tap; exit']' returned non-zero exit status 1
```

sbourdeauducq, 2020-07-20:
> How can I access the UART port of a Kasli from a windows machine? I tried installing a USB serial port driver for interface 2 of the `Quad-RS232 HS` device, but could not connect to the [...]

Did you try with PuTTY? It is surprisingly trouble-free software, as far as Windows applications go. What is the error message?

> Side note: artiq_flash can't seem to cope with the presence of any Quad-RS232 HS devices with drivers other than WinUSB. 

Did you install the Zadig thing?

jkellerPTB, 2020-07-20:
Here's what I did in more detail:
* I'm using Zadig to install the WinUSB driver for Interface 0 of the `Quad-RS232 HS` device. This works, i.e. I can then use `artiq_flash`.
* In order to connect to the serial interface, I tried both the FTDI driver and `USB Serial (CDC)` as offered by Zadig for `Quad-RS232 HS Interface 2`. With the latter, the device then shows up as `COM6` in the device manager (in both cases, `artiq_flash` then fails as described above until I use Zadig to replace the driver for Interface 2 with WinUSB).
* In the device manager, I configure COM6 as 115200-8n1.
* Trying to connect with PuTTY using these settings, I get the error `Unable to configure serial port`
* I also tried serial.tools.miniterm, which gives me the error `A device attached to the system is not functioning`

On linux, `python -m serial.tools.miniterm /dev/ttyUSB2 115200` works.

ljstephenson, 2020-09-18:
I saw similar behaviour with `artiq_flash` trying to flash a kasli in W10: the issue is in the OpenOCD script located in `<conda location>\envs\<env name>\Library\share\openocd\scripts\board\kasli.cfg`, the device description `Quad RS232-HS` doesn't always match what Windows has decided to call it on that given day.

By changing the line

```
ftdi_device_desc "Quad RS232-HS"
```

to

```
ftdi_device_desc "USB <-> Serial Converter"
```

in my case. I found this by looking at Device Manager properties, Details tab, "Bus reported device description". Note this is not the "Device description", which was "USB <-> Serial Converter (Interface 0)"... Logical, I know.

In summary:

- Use Zadig to replace driver for _only_ Interface 0 with WinUSB
- `artiq_flash` fails with `Error: unable to open ftdi device with vid 0403, pid 6011, description 'Quad RS232-HS', serial '*' at bus location '*'`
- Edit `openocd\scripts\board\kasli.cfg` to reflect Windows whimsy
- `artiq_flash` works
- PuTTY also works on Interface 2 (whichever COM port that has been assigned)

hartytp, 2020-09-19:
> Edit openocd\scripts\board\kasli.cfg to reflect Windows whimsy

Is this definitely a windows/driver thing? I thought that description was something that's flashed into the FT232 and can be changed with e.g. FT PROG or the Kasli I2C tool. IIRC I've also had the issue on Linux and it can be fixed either by replacing the `Quad RS232-HS` with `*` (as you do) or by changing the flashed description to match the openocd script's expectation (which is better since you don't have to maintain a patched version of openocd).

sbourdeauducq, 2020-09-19:
> IIRC I've also had the issue on Linux

Yes, this happens also with Digilent JTAG probes and the scripts provided upstream openocd. I guess Digilent flash the FTDI EEPROMs inconsistently.

> and it can be fixed either by replacing the `Quad RS232-HS` with `*` (as you do)

Or remove the ``ftdi_device_desc`` line entirely.

ljstephenson, 2020-09-26:
@hartytp you're completely right about the option to change product description in FT PROG. Not sure how I missed it given that it's right above where I reprogrammed the serial number 🤦

hartytp, 2020-09-26:
@gkasprow can we ask TS/CTI to set the FTDI settings appropriately before shipping? It would be great if they set a unique serial number as well.

hartytp, 2020-09-26:
otherwise we should add a note about this in the artiq manual (@ljstephenson fancy spinning up a short doc PR?)

gkasprow, 2020-09-26:
I think this can be done. All boards are tested using an automated test bench so that should be simple

sbourdeauducq, 2020-09-27:
This would not solve it for the existing fleet of boards that have empty EEPROMs.
Where is that "USB <-> Serial Converter" string coming from? Windows driver or EEPROM? If EEPROM, who put it there?

hartytp, 2020-09-27:
AFAICT it's programmed into the EEPROM before Kasli shipped.

If you want a solution for existing boards then either change the openocd script or tell users to reprogram them (using the Kasli I2C tool or FTD PROG).

ljstephenson, 2020-09-28:
Confirmed that reprogramming EEPROM using FT PROG with correct description made everything work happily for flashing on W10. @hartytp I'll make a doc PR once I've cleared a couple of things off my to-do list, unless the solution is to delete the `ftdi_device_desc` line entirely as @sbourdeauducq suggests - presumably VID and PID are sufficient information, and these shouldn't change.

Re: unique serial numbers I'm agnostic, but one thing to consider is that they will make W10 enumerate each new device as a different set of 4 COM ports - could be a bug or feature depending on your perspective.

sbourdeauducq, 2020-09-28:
@ljstephenson Where did you purchase the problematic board from, and did you or someone else flash the EEPROM after receiving it?

ljstephenson, 2020-09-28:
The board was from Technosystems. I did flash the EEPROM after receiving (precisely to give a unique serial number), but FT PROG populates fields with the currently stored value unless empty, in which case they remain unpopulated, i.e. it was already programmed with "USB <-> Serial Converter"

gkasprow, 2020-09-28:
I can confirm that CTI implemented EEPROM configuration in the production test suite.

sbourdeauducq, 2020-10-15:
Looks like the COM port issue is a Windows/FTDI bug, not ARTIQ.

closed 2020-10-15 by sbourdeauducq

hartytp, 2020-12-15:
> The board was from Technosystems. I did flash the EEPROM after receiving (precisely to give a unique serial number), but FT PROG populates fields with the currently stored value unless empty, in which case they remain unpopulated, i.e. it was already programmed with "USB <-> Serial Converter"

FWIW: plugging a new Kasli from TS into a linux PC it comes up with the `Quad RS232-HS` description. Plugging it into a windows machine and opening FT PROG, the default option is `USB <-> Serial Converter`. I haven't dug into this exhaustively but it feels like FT_PROG just ignores the current value and overwrites it with one that's not default.

@sbourdeauducq since this is the default value used by FT_PROG should we change the openocd rule to allow it? Otherwise, we should probably add this to the docs somewhere since it's bitten me plenty of times.

NB if one programs the EEPROMs using @jordens the expected value is always set so we could also advertise that (although it's linux only AFAICT). https://github.com/quartiq/kasli-i2c/blob/master/kasli-ft4232h.conf.in

jordens, 2020-12-15:
Yes. `kasli-i2c` and the openocd scripts are intentionally compatible with both unmodified and flashed EEPROMs as long as they keep pid/vid/description. There is enough opportunity in manufacturer and serial for customization.

hartytp, 2020-12-15:
so tl;dr this is FT_PROG being dumb. Still though the openocd rule we have is quite restrictive and does introduce a bit of a foot gun particularly since the `kasli-i2c` tool isn't advertised anywhere. Can we easily relax it? e.g. is it harmful to only test vid + pid and not the description?

sbourdeauducq, 2020-12-16:
> @sbourdeauducq since this is the default value used by FT_PROG should we change the openocd rule to allow it?

That's much harder than you think considering the need to rebuild and package openocd on windoze. It's not in nix-scripts/wfvm yet.