m-labs/artiq #1488: Typical delay times in synchronous RPC calls?

JKiethe, 2020-07-22:
# Question
## Category: RPC
## Description
Is it normal, that a RPC call from the core device, that just returns True, needs 55ms on average?
We tested it on two different setups multiple times and measured around 55ms every time.
Typical ping times between host and core device are below a millisecond.

```python
from artiq.experiment import *

class RPCTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    def rpc_func(self) -> TBool:
        return True

    @kernel
    def kernel_func(self):
        self.core.reset()

        t1 = self.core.get_rtio_counter_mu()

        self.rpc_func()

        t2 = self.core.get_rtio_counter_mu()

        print("duration of the RPC: ", self.core.mu_to_seconds(t2 - t1))

    def run(self):
        self.kernel_func()
```

```python
#output
duration of the RPC:  0.054898408
```

## Your System

* Operating System: Windows 10
* ARTIQ version: 5.7110.88391010
* Hardware involved: Kasli

sbourdeauducq, 2020-07-23:
Fast Linux PC with Gb LAN and two switches between PC and KC705:
```
> artiq_run test.py
duration of the RPC:  0.0024392560000000003
```

Windows is probably not the problem (though it otherwise remains a toy operating system that is only good for running Steam and games, and you should not be using it in a laboratory); what network connection to the device are you using?

JKiethe, 2020-07-23:
Thank you for the test.
> what network connection to the device are you using?
 One setup uses Gb LAN and a media converter to switch an optical fiber.
The other setup currently uses an Gb USB to Ethernet converter.

Given that you measure way smaller delays, we now have the suspicion, that it has to do with the internal network setup.
We will try a few things and get back to you.

JKiethe, 2020-08-19:
After we restarted the PCs network connection to the internal network (which is not were the Sinara crate is connected), we were able to observe similar duration times for the RPC as @sbourdeauducq of roughly 2.5ms.

We monitored it for a few weeks and the 55ms RPC delay times were not observed again. The delay times were always between 2 and 3 ms.
It might just have been a general issue of the internal network at the initial observation. Why this influenced a totally different ethernet connection, I do not know.

As this is not a problem with the Sinara hardware, I will close this issue.

closed 2020-08-19 by JKiethe