m-labs/artiq #1490: Fastest data transfer from host to kernel

JKiethe, 2020-07-22:
# Question

## Category: RPC

## Description

What is the fastest way to transfer data (several tens of thousand bits) from the host to the core device, where the encoding of the data is irrelevant?

From #1255 we know, that lists are unfavorable due to their serialization.

A faster alternative is transfering bytes, but here we run into the problem, 
that the kernel doesn't accept more than 256 bytes:
`Terminating with exception (ValueError: bytes must be in range(0, 256))`
 Is this limit intentional?
This is it not a Python/host problem, since Python can convert a list with more than 256 
elements to bytes.

The length limitation does not seem to be apply for strings however, even though it can transfer data with about the same speed as bytes (around 2MB/s), but without this 256-byte-limit (data were some 10000 characters).

Apart from the mentioned methods, is there another one we can try for data transfer?

sbourdeauducq, 2020-07-23:
> A faster alternative is transfering bytes, but here we run into the problem,
> that the kernel doesn't accept more than 256 bytes:
> `Terminating with exception (ValueError: bytes must be in range(0, 256))`

What is the exact code you ran to get this error? The unit tests we run involve transfer of byte arrays that are much longer than 256.

2MB/s is also the rate we're getting.

JKiethe, 2020-07-23:
We found the error in our code (see below).
We used the bytes conversion of Python incorrectly, as it cannot convert a list of integers to bytes if any integer is bigger than 256.

As this is not actually an ARTIQ issue, I will close it.
Thanks for the help and sorry for not properly checking the Python documentation before writing this issue.

If you are interested here is the code to reproduce the errror:

```Python
from artiq.experiment import *

class BytesTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
    def bytes_func(self) -> TBytes:
        a = [0]*300
        for i in range(0, 300):  
            a[i] = i # this throws "ValueError: bytes must be in range(0, 256)" if i>256
        return bytes(a)
    @kernel
    def kernel_func(self):
        self.bytes_func()
    def run(self):
        self.kernel_func()
```

closed 2020-07-23 by JKiethe