m-labs/artiq #1491: Caught exception object causes memory corruption

pca006132, 2020-07-23:
# Bug Report

## One-Line Summary

Exception objects caught by `except ... as obj:` is a dangling pointer and could cause memory corruption.

## Issue Details

This is a problem due to the following code, which assumes the exception object would not be caught by the user:

https://github.com/m-labs/artiq/blob/2a2f5c4d58c5146479e52359943c0c9b5bd15548/artiq/firmware/ksupport/eh_artiq.rs#L166-L169

When the exception is caught, the object is actually a dangling pointer pointing to somewhere in the stack and can be overwritten by other code, or corrupt other memory when mutated (possible, not tested).

If we copy the exception object before raising the exception, we could avoid memory corruption, but we would get memory leak as we cannot deallocate the object after caught. 

This issue seems unsolvable in the stack only allocation model for the kernel.
The best way maybe is to disallow the use of `except ... as ...:` clause, so the user won't have access to the exception object.

### Example

```python
from artiq.experiment import *

class TestException(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        try:
            raise ValueError
        except ValueError as e:
            try:
                raise IndexError
            except:
                pass
            try:
                raise e
            except ValueError as e2:
                print("ValueError")
            except IndexError as e3:
                print("Should not happen")
```

### Expected Behavior

Prints `ValueError`.

### Actual (undesired) Behavior

Prints `Should not happen`, as the memory for `e` is overwritten in the second `raise`.

pca006132, 2020-07-23:
Update: We don't have to capture the exception object.

```python
from artiq.experiment import *

class TestException(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        try:
            try:
                raise ValueError
            finally:
                try:
                    raise IndexError
                except IndexError:
                    pass
        except ValueError:
            print("ValueError")
        except IndexError:
            print("Should not happen")
```
Same behavior.

sbourdeauducq, 2022-01-14:
Adding to ARTIQ-7 milestone as we want to keep runtime compatibility with NAC3.

closed 2022-01-25 by sbourdeauducq