m-labs/artiq #1502: artiq_flash can't find bitstreams, openocd, when installed via conda on windows.

maij, 2020-08-06:
When running artiq_flash I get the following error message:
"Cannot find proxy bitstream bscan_spi_xc7a100t.bit"

I've downloaded zadig (there doesn't appear to be an actual installer at this time, it doesn't require installation to run.) and I've installed the WinUSB driver for Quad RS232-HS interface 0.

The artiq_flash help is very specific to a *nix based operating system, it doesn't apply to a Windows system, but it leads me to believe that I still needed to install openocd and place the right bitstream file in the path. 

I've discovered that all the bitsreams are installed in the root of my conda environment, so I think the artiq_flash path options needs to be updated with os.getenv('CONDA_PREFIX') because that's where the bitstreams are installed by default. Further, openocd can't be installed by conda on Windows using `conda install openocd` as is suggested by the latest [installation manual.](https://m-labs.hk/artiq/manual/installing.html#installing-openocd). It fails with the following error:
```
Collecting package metadata (current_repodata.json): done                             
Solving environment: failed with initial frozen solve. Retrying with flexible solve.  
Collecting package metadata (repodata.json): done                                     
Solving environment: failed with initial frozen solve. Retrying with flexible solve.  
                                                                                      
PackagesNotFoundError: The following packages are not available from current channels:
                                                                                      
  - openocd                                                                           
                                                                                      
Current channels:                                                                     
                                                                                      
  - https://conda.m-labs.hk/artiq/win-64                                              
  - https://conda.m-labs.hk/artiq/noarch                                              
  - https://repo.anaconda.com/pkgs/main/win-64                                        
  - https://repo.anaconda.com/pkgs/main/noarch                                        
  - https://repo.anaconda.com/pkgs/r/win-64                                           
  - https://repo.anaconda.com/pkgs/r/noarch                                           
  - https://repo.anaconda.com/pkgs/msys2/win-64                                       
  - https://repo.anaconda.com/pkgs/msys2/noarch                                       
  - https://conda.anaconda.org/conda-forge/win-64                                     
  - https://conda.anaconda.org/conda-forge/noarch                                     
                                                                                      
To search for alternate channels that may provide the conda package you're            
looking for, navigate to                                                              
                                                                                      
    https://anaconda.org                                                              
                                                                                      
and use the search bar at the top of the page.
```

This has been a big headache to debug, I excpect the ability to flash the latest gateware to our artiq system should be seamless, rather than fraught with complications.

---
By manually moving the bitstream files to where artiq_flash searches for them, I then get this new error:
```
# artiq_flash                                                                                                  
Design: top;UserID=FFFFFFFF;COMPRESS=TRUE;Version=2020.1                                                       
Part name: 7a100tfgg484                                                                                        
Date: 2020/08/05                                                                                               
Time: 12:42:45                                                                                                 
Bitstream payload length: 0x2bd680                                                                             
Traceback (most recent call last):                                                                             
  File "C:\ProgramData\Anaconda3\envs\artiq\Scripts\artiq_flash-script.py", line 9, in <module>                
    sys.exit(main())                                                                                           
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\site-packages\artiq\frontend\artiq_flash.py", line 460, in main
    programmer.run()                                                                                           
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\site-packages\artiq\frontend\artiq_flash.py", line 204, in run 
    self._client.run_command(cmdline)                                                                          
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\site-packages\artiq\remoting.py", line 57, in run_command      
    subprocess.check_call([arg.format(tmp=self._tmp, **kws) for arg in cmd])                                   
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\subprocess.py", line 266, in check_call                        
    retcode = call(*popenargs, **kwargs)                                                                       
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\subprocess.py", line 247, in call                              
    with Popen(*popenargs, **kwargs) as p:                                                                     
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\subprocess.py", line 676, in __init__                          
    restore_signals, start_new_session)                                                                        
  File "C:\ProgramData\Anaconda3\envs\artiq\lib\subprocess.py", line 957, in _execute_child                    
    startupinfo)                                                                                               
FileNotFoundError: [WinError 2] The system cannot find the file specified             
```

The file in question is `openocd`.

sbourdeauducq, 2020-08-06:
> This has been a big headache to debug, I excpect the ability to flash the latest gateware to our artiq system should be seamless, rather than fraught with complications.

First of all, let's get this straight:
* Conda and Microsoft Windows are trashy, shoddy software that you ought not to use.
* Your headache is nothing compared to what we have to endure to get anything to work on Conda/Windows.

> The artiq_flash help is very specific to a *nix based operating system, it doesn't apply to a Windows system, but it leads me to believe that I still needed to install openocd and place the right [...]

It used to work correctly on Windows, though regressions wouldn't be surprising (did I say Windows and Conda are trashy?). Generally, ARTIQ itself does not include things that are specific to a package manager.

The openocd package is here:
https://anaconda.org/m-labs/openocd
Try installing it.

maij, 2020-08-07:
Hi @sbourdeauducq, thanks for your quick response, it's much appreciated!

After installed openocd and the bscan-spi-bitstreams package from the link provided I was able to flash one of our artiq, and I can communicate with it via the LAN. Unfortunately none of the demo code is working (at least using the "device_db.py" supplied to UNSW, I've yet to try the examples using the master device_db.py), I am even unable to turn off and on the user leds.

Could you please confirm that the kasli-sydney board is the right choice for the systems bought by UNSW (SN: 128779/128780)? 

I am also encountering the issue #1487 when trying to debug via the FTDI port, and so I'm a bit stuck for what to do.

I'm happy to have this issue closed, the only remaining problem is that the [documentation](https://m-labs.hk/artiq/manual/installing.html#installing-openocd) states: `conda install openocd`, which fails but `conda install -c m-labs openocd` works. It's strange because my channels list seems correct:
```
$ conda config --get channels
--add channels 'conda-forge'   # lowest priority
--add channels 'defaults'
--add channels 'https://conda.m-labs.hk/artiq'   # highest priority
```
Also the documentation doesn't mention that the package bscan-spi-bitstreams also needs to be installed manually. Genuine question, would it be worth packaging these items in an `artiq_flash_utilities` or similar?

sbourdeauducq, 2020-08-07:
> Could you please confirm that the kasli-sydney board is the right choice for the systems bought by UNSW

It is not the right one. Please email me (sb@m-labs) a picture of the front of the system with all the cards, and I'll check which firmware and device database you should be using.

> I've yet to try the examples using the master device_db.py

This will not work. A device database is specific to a certain configuration of cards.

> I'm a bit stuck for what to do.

In my experience, a very common situation when trying to perform seemingly simple tasks like this but with Microsoft software...

> Also the documentation doesn't mention that the package bscan-spi-bitstreams also needs to be installed manually.

Normally conda would install it as a dependency of openocd, but considering the reliability of the conda dependency solver I was suggesting checking manually that it is installed.

closed 2020-08-28 by maij