m-labs/artiq #1503: or1k-linux-ld not found with conda install

stffnm, 2020-08-06:
# Bug Report

## or1k-linux-ld not found with conda install 

installing artiq via conda seems to install an incompatible version of binutils-or1k-linux

## Issue Details

After installation, experiments cannot be submitted via the artiq dashboard. Instead, an error appears when trying to submit experiments.

### Steps to Reproduce

1. Install artiq with conda create -n artiq artiq=5
2. start master and dashboard
3. submit experiment

### Expected Behavior

Experiment runs

### Actual (undesired) Behavior

ERROR:worker(15,LED_test.py):root:Terminating with exception (FileNotFoundError: [Errno 2] No such file or directory: 'or1k-linux-ld': 'or1k-linux-ld')
Traceback (most recent call last):
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/master/worker_impl.py", line 290, in main
    exp_inst.run()
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/coredevice/core.py", line 129, in run
    self.compile(function, args, kwargs, set_result)
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/coredevice/core.py", line 112, in compile
    library = target.compile_and_link([module])
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/compiler/targets.py", line 191, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/compiler/targets.py", line 181, in link
    **{"obj{}".format(index): obj for index, obj in enumerate(objects)}) \
  File "/opt/conda/envs/test_artiq/lib/python3.7/site-packages/artiq/compiler/targets.py", line 32, in __enter__
    universal_newlines=True)
  File "/opt/conda/envs/test_artiq/lib/python3.7/subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "/opt/conda/envs/test_artiq/lib/python3.7/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'or1k-linux-ld': 'or1k-linux-ld'

### Workaround

run 

conda install -c m-labs binutils-or1k-linux

The following packages will be UPDATED:

  binutils-or1k-lin~      artiq::binutils-or1k-linux-2.30-0 --> m-labs::binutils-or1k-linux-2.30-7

This fixes the problem.

### My System 

* Operating System: Ubuntu 20.04 LTS inside Docker
* ARTIQ version: ARTIQ v5.7122.929b04da
* conda list after initial install: 
[artiq_not_working_conda_list.txt](https://github.com/m-labs/artiq/files/5033433/artiq_not_working_conda_list.txt)

sbourdeauducq, 2020-08-06:
>     conda create -n artiq=5

You don't need ``=5``.
The conda dependency manager is so broken that we are working around it by proposing only one possible version to conda.

Is it correct that the binutils package gets installed but then ARTIQ doesn't find it?

Why are you using conda and not Nix as recommended for Linux?

stffnm, 2020-08-06:
`=5` is just for clarity, the error is the same without it.

`Is it correct that the binutils package gets installed but then ARTIQ doesn't find it?`
To me it looks like the version on the `artiq` channel is older than on `m-labs`

I'm using conda simply because this is what people (non-programmers in our group) are used to. I am considering switching to Nix though...

sbourdeauducq, 2020-08-06:
>  To me it looks like the version on the `artiq` channel is older than on `m-labs`

The question I was asking is: when you used the original instructions, did it install the binutils package (as it should) but then ARTIQ did not find binutils?

> I'm using conda simply because this is what people (non-programmers in our group) are used to. 

Sadly, Anaconda Inc are putting a lot of marketing effort into peddling their low-quality product to scientists, with success. A bit like Microsoft with Windows...

stffnm, 2020-08-06:
> The question I was asking is: when you used the original instructions, did it install the binutils package (as it should) but then ARTIQ did not find binutils?

I'm not sure how you want me to check this. The package is listed as installed in `conda list`, as you can see in the .txt file.
I don't know anything about this package, can you give me more detailed instructions please?

sbourdeauducq, 2020-08-06:
> The package is listed as installed in `conda list`, as you can see in the .txt file

So it was there already before you installed manually? That's all I need to know.

stffnm, 2020-08-06:
Yes

sbourdeauducq, 2020-08-06:
The binutils package itself seems to work...
```text
conda-shell-chrootenv:sb@vulcan:~$ conda create -n test binutils-or1k-linux
Collecting package metadata: done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.6.14
  latest version: 4.8.3

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /home/sb/.conda/envs/test

  added / updated specs:
    - binutils-or1k-linux

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    _libgcc_mutex-0.1          |             main           3 KB
    binutils-or1k-linux-2.30   |                0        47.4 MB  https://conda.m-labs.hk/artiq
    libgcc-ng-9.1.0            |       hdf63c60_0         8.1 MB
    libiconv-1.15              |       h63c8f33_5         2.0 MB
    ------------------------------------------------------------
                                           Total:        57.5 MB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  binutils-or1k-lin~ artiq/linux-64::binutils-or1k-linux-2.30-0
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libiconv           pkgs/main/linux-64::libiconv-1.15-h63c8f33_5

Proceed ([y]/n)? 

Downloading and Extracting Packages
_libgcc_mutex-0.1    | 3 KB      | ##################################################################################################################### | 100% 
binutils-or1k-linux- | 47.4 MB   | ##################################################################################################################### | 100% 
libgcc-ng-9.1.0      | 8.1 MB    | ##################################################################################################################### | 100% 
libiconv-1.15        | 2.0 MB    | ##################################################################################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use:
# > conda activate test
#
# To deactivate an active environment, use:
# > conda deactivate
#

conda-shell-chrootenv:sb@vulcan:~$ conda activate test

CommandNotFoundError: Your shell has not been properly configured to use 'conda activate'.
To initialize your shell, run

    $ conda init <SHELL_NAME>

Currently supported shells are:
  - bash
  - fish
  - tcsh
  - xonsh
  - zsh
  - powershell

See 'conda init --help' for more information and options.

IMPORTANT: You may need to close and restart your shell after running 'conda init'.

conda-shell-chrootenv:sb@vulcan:~$ conda init bash
no change     /home/sb/.conda/condabin/conda
no change     /home/sb/.conda/bin/conda
no change     /home/sb/.conda/bin/conda-env
no change     /home/sb/.conda/bin/activate
no change     /home/sb/.conda/bin/deactivate
no change     /home/sb/.conda/etc/profile.d/conda.sh
no change     /home/sb/.conda/etc/fish/conf.d/conda.fish
no change     /home/sb/.conda/shell/condabin/Conda.psm1
no change     /home/sb/.conda/shell/condabin/conda-hook.ps1
no change     /home/sb/.conda/lib/python3.7/site-packages/xonsh/conda.xsh
no change     /home/sb/.conda/etc/profile.d/conda.csh
modified      /home/sb/.bashrc

==> For changes to take effect, close and re-open your current shell. <==

conda-shell-chrootenv:sb@vulcan:~$ bash
(base) conda-shell-chrootenv:sb@vulcan:~$ conda activate test
(test) conda-shell-chrootenv:sb@vulcan:~$ or1k-linux-ld
or1k-linux-ld: no input files
```

sbourdeauducq, 2020-08-06:
Works on my machine™ with all of ``artiq_compile``, ``artiq_run``  and ``artiq_master``. Probably the usual conda non-reproducible nonsense. Just use Nix or your workaround.

```text
conda-shell-chrootenv:sb@vulcan:~$ conda create -n artiq artiq
Collecting package metadata: done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.6.14
  latest version: 4.8.3

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /home/sb/.conda/envs/artiq

  added / updated specs:
    - artiq

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    _libgcc_mutex-0.1          |             main           3 KB
    artiq-5.7122.929b04da      |                0         517 KB  https://conda.m-labs.hk/artiq
    binutils-or1k-linux-2.30   |                0        47.4 MB  https://conda.m-labs.hk/artiq
    blas-1.0                   |              mkl           6 KB
    ca-certificates-2020.6.24  |                0         133 KB
    cached-property-1.5.1      |             py_1          10 KB
    certifi-2020.6.20          |           py37_0         159 KB
    cffi-1.14.0                |   py37he30daa8_1         226 KB
    curl-7.71.1                |       hbc83047_1         151 KB
    dbus-1.13.16               |       hb2f20db_0         589 KB
    expat-2.2.9                |       he6710b0_2         196 KB
    fontconfig-2.13.0          |       h9420a91_0         291 KB
    freetype-2.10.2            |       h5ab3b9f_0         915 KB
    glib-2.65.0                |       h3eb4bd4_0         3.5 MB
    gst-plugins-base-1.14.0    |       hbbd80ab_1         6.3 MB
    gstreamer-1.14.0           |       hb31296c_0         3.7 MB
    h5py-2.10.0                |   py37hd6299e0_1         1.1 MB
    hdf5-1.10.6                |       hb1b8bf9_0         4.8 MB
    icu-58.2                   |       he6710b0_3        22.7 MB
    intel-openmp-2020.1        |              217         940 KB
    jpeg-9b                    |       h024ee3a_2         248 KB
    krb5-1.18.2                |       h173b8e3_0         1.5 MB
    ld_impl_linux-64-2.33.1    |       h53a641e_7         645 KB
    libcurl-7.71.1             |       h20c2e04_1         313 KB
    libedit-3.1.20191231       |       h14c3975_1         121 KB
    libffi-3.3                 |       he6710b0_2          54 KB
    libgcc-ng-9.1.0            |       hdf63c60_0         8.1 MB
    libgfortran-ng-7.3.0       |       hdf63c60_0         1.3 MB
    libgit2-1.0.0              |       h1e06722_0         725 KB  conda-forge
    libiconv-1.15              |       h63c8f33_5         2.0 MB
    libpng-1.6.37              |       hbc83047_0         364 KB
    libssh2-1.9.0              |       h1ba5d50_1         346 KB
    libstdcxx-ng-9.1.0         |       hdf63c60_0         4.0 MB
    libuuid-1.0.3              |       h1bed415_2          16 KB
    libxcb-1.14                |       h7b6447c_0         610 KB
    libxml2-2.9.10             |       he19cac6_1         1.3 MB
    llvmlite-artiq-0.23.0.dev  |                0        13.1 MB  https://conda.m-labs.hk/artiq
    mkl-2020.1                 |              217       200.5 MB
    mkl-service-2.3.0          |   py37he904b0f_0         208 KB
    mkl_fft-1.1.0              |   py37h23d657b_0         158 KB
    mkl_random-1.1.1           |   py37h0573a6f_0         375 KB
    ncurses-6.2                |       he6710b0_1         1.1 MB
    numpy-1.19.1               |   py37hbc911f0_0          20 KB
    numpy-base-1.19.1          |   py37hfa32c7d_0         5.2 MB
[... 124 lines cut ...]
(artiq) conda-shell-chrootenv:sb@vulcan:~/artiq/artiq/examples/kc705_nist_clock$ artiq_compile repository/blink_forever.py
(artiq) conda-shell-chrootenv:sb@vulcan:~/artiq/artiq/examples/kc705_nist_clock$ artiq_run repository/blink_forever.py
WARNING:artiq.coredevice.comm_kernel:Mismatch between gateware (6.7289.504f72a0.beta) and software (5.7122.929b04da) versions
^CTraceback (most recent call last):
  File "/home/sb/.conda/envs/artiq/bin/artiq_run", line 10, in <module>
    sys.exit(main())
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/frontend/artiq_run.py", line 225, in main
    return run(with_file=True)
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/frontend/artiq_run.py", line 204, in run
    exp_inst.run()
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/coredevice/core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/coredevice/comm_kernel.py", line 501, in serve
    self._read_header()
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/coredevice/comm_kernel.py", line 116, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "/home/sb/.conda/envs/artiq/lib/python3.7/site-packages/artiq/coredevice/comm_kernel.py", line 97, in read
    rn = self.socket.recv(min(8192, length - len(r)))
KeyboardInterrupt
(artiq) conda-shell-chrootenv:sb@vulcan:~/artiq/artiq/examples/kc705_nist_clock$ artiq_master 
ARTIQ master is now ready.
WARNING:worker(3064,blink_forever.py):artiq.coredevice.comm_kernel:Mismatch between gateware (6.7289.504f72a0.beta) and software (5.7122.929b04da) versions
```

closed 2020-08-06 by sbourdeauducq