m-labs/artiq #1505: coredevice: Integers raised to an integer power not handled correctly

dnadlinger, 2020-08-09:
On host Python 3.8.5/NumPy 1.19.0,
```
>>> np.int32(10)**np.int32(10)
1410065408
```
which indeed makes sense
```
>>> 10**10 % 2**31
1410065408
```

On the core device, however, the following prints `31` on latest master (cf19c951):

```py
from artiq.experiment import *
import numpy as np

class PowTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def pow(self, a, b):
        return a ** b

    def run(self):
        print(self.pow(np.int32(10), np.int32(10)))
```

In fact, the return value seems to be uninitialised memory, as the base is converted to floating point before calling `powi`, which LLVM evaluates statically here, before the result is `fptosi`-d back to i32, which overflows and hence results in a poison value.

sbourdeauducq, 2022-01-07:
To be fixed in NAC3 (https://git.m-labs.hk/M-Labs/nac3/pulls/160).

sbourdeauducq, 2022-01-08:
Fixed in NAC3:
```
from numpy import int32
from min_artiq import *

@nac3
class Demo:
    core: KernelInvariant[Core]

    def __init__(self):
        self.core = Core()

    @kernel
    def pow(self, a: int32, b: int32) -> int32:
        return a ** b

    @kernel
    def run(self):
        print_int32(self.pow(10, 10))

if __name__ == "__main__":
    Demo().run()
```

```
> ../../target/release/runkernel ./module.elf 
print_int32: 1410065408
```

closed 2022-01-08 by sbourdeauducq