m-labs/artiq #1507: Dual EEM Sampler without SU-Servo possible

spomjaksilp, 2020-08-10:
# Question
Dual EEM Sampler without SU-Servo possible?

## Category: devices

## Description
Is the dual EEM mode only usable as SU-Servo?
When specifying `ports: [8, 9]` in contrast to `ports: [8]` and using `artiq_ddb_template`, there is now difference in the resulting `device_db`.

In both cases we get:

```
device_db["spi_sampler0_adc"] = {
    "type": "local",
    "module": "artiq.coredevice.spi2",
    "class": "SPIMaster",
    "arguments": {"channel": 0x000029}
}
device_db["spi_sampler0_pgia"] = {
    "type": "local",
    "module": "artiq.coredevice.spi2",
    "class": "SPIMaster",
    "arguments": {"channel": 0x00002a}
}
device_db["ttl_sampler0_cnv"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x00002b},
}
device_db["sampler0"] = {
    "type": "local",
    "module": "artiq.coredevice.sampler",
    "class": "Sampler",
    "arguments": {
        "spi_adc_device": "spi_sampler0_adc",
        "spi_pgia_device": "spi_sampler0_pgia",
        "cnv_device": "ttl_sampler0_cnv"
    }
}
```

thx :)

sbourdeauducq, 2020-08-10:
Yes. For kernels the speed is dominated by the RTIO/CPU overhead, so using two-EEM mode will not help. There are potential other uses for dual-EEM (e.g. RTIO DMA-in) but they are currently unfunded.

closed 2020-08-10 by sbourdeauducq

hartytp, 2020-08-10:
@sbourdeauducq semi-related, but how easy would it be to use wide-RTIO channels for inputs as well to provide a faster interface like the one for Fastino?