m-labs/artiq #1518: documentation: Fastino

hartytp, 2020-09-03:
Fastino's wide interface does not allow setting an arbitary subset of channels. Instead it one must update a fixed number of channels, which all have to be adjacent.

These "channel groups" aren't explained in the core device driver. Please can we have a few words on what the groups are, their usage and limitations? (e.g. the fact that once can only update adjacent channels).

@pathfinder49 was there anything else on the Fastino documentation front on your list?

jordens, 2020-09-03:
This setting was added for you to evaluate, test, and characterize the impact on RTIO fabric and performance of wide vs narrow RTIO channels. Before making it a stable API with documentation, integration, tests etc. could you share the results. Also depending on the results it might be much better to just choose one value.

hartytp, 2020-09-03:
@pathfinder49 if you have any other data points on this, please do share them here...e.g. how many Fastinos could you pack onto a master with the normal RTIO interface?

pathfinder49, 2020-09-03:
I haven't tested the normal interface beyond 2 fastino boards.

pathfinder49, 2020-09-03:
> @pathfinder49 was there anything else on the Fastino documentation front on your list?

It should also be explicitly documented, which methods require what gateware configuration. I assume the Fastino gateware is the same for all interfaces? 

It is ambiguous what Fastino.update() and Fastino.set_hold() do. It is also unclear how these interact with the wide interface. 

The return value of Fastino.voltage_to_mu() should be masked to be consistent with other drivers. 

> This setting was added for you to evaluate, test, and characterize the impact on RTIO fabric and performance of wide vs narrow RTIO channels. Before making it a stable API with documentation, [...]

Ack

jordens, 2020-09-03:
> It should also be explicitly documented, which methods require what gateware configuration.

That's documented.

> I assume the Fastino gateware is the same for all interfaces?

Yes.

> It is ambiguous what Fastino.update() and Fastino.set_hold() do.

I copied the documentation over from the PHY.

> It is also unclear how these interact with the wide interface.

They don't. They are orthogonal.

> The return value of Fastino.voltage_to_mu() should be masked to be consistent with other drivers.

Done.

pathfinder49, 2020-09-03:
> > It should also be explicitly documented, which methods require what gateware configuration.
> 
> That's documented.

To me that would include the fact that log2width must be configured in gateware.

> > The return value of Fastino.voltage_to_mu() should be masked to be consistent with other drivers.
> 
> Done.

Checking the zotino driver, this should actually be checked for over-/under-flow.

jordens, 2020-09-03:
> > > It should also be explicitly documented, which methods require what gateware configuration.
> > 
> > That's documented.
> 
> To me that would include the fact that log2width must be configured in gateware.

It does and did.
https://github.com/m-labs/artiq/blob/e803830b3b3cc791fd98425ac30a4efb1c210485/artiq/coredevice/fastino.py#L18-L23

> Checking the zotino driver, this should actually be checked for over-/under-flow.

Well, what is it? Let's leave it as it is until you have done the testing you wanted to do (#1481) and then revisit to finalize it.

pathfinder49, 2020-09-17:
To me it makes most sense for the Fastino behavior to match that of Zotino. It also seems appropriate to point out the voltage range similarly to the Zotino documentation. 
https://github.com/m-labs/artiq/blob/29c940f4e378073399412f3cc58c5443280bc714/artiq/coredevice/ad53xx.py#L83-L106

> > > > It should also be explicitly documented, which methods require what gateware configuration.
> > > 
> > > 
> > > That's documented.
> > 
> > 
> > To me that would include the fact that log2width must be configured in gateware.
> 
> It does and did.

References like RTIO PHI are unfortunately quite opaque to new users. Many (most?) new users will have recently complete a master/bachelor in physics and lack the background to easily understand these references. Adding a sentence like "`log2_width` must match the value configured in gateware." would make this more accessible.

jordens, 2020-09-18:
There is no a priori reason for fastino to blindly follow zotino.

I don't see how replacing RTIO Phy by Gateware would help. You wouldn't know what Gateware is or where and how it is configured. And explaining either here would be the wrong place.

pathfinder49, 2020-09-18:
> There is no a priori reason for fastino to blindly follow zotino.

Fair though I think the reasoning from #1444 applies here.

> I don't see how replacing RTIO Phy by Gateware would help. You wouldn't know what Gateware is or where and how it is configured. And explaining either here would be the wrong place.

I do not mean to remove RTIO Phy from the documentation. It is a useful description to a familiar user. The explicit mention of gateware would be in addition (at the risk of repetition). I also would not explain either concept here. I would simply offer both so that users have a better chance of having encountered the relevant concept.

Opaque references like RTIO PHYs are one of the criticisms I repeatedly hear from new users. One of the problems is that people don't know where to start looking for these as they have no concept of what they might be. 

I think the concept of the coredevice having gateware would be included in most initial user introductions. Referring to this makes it clear that `log2_width` is not a parameter that can be changed in software. It also gives users a clear place to start looking, as there is a folder with an identical name near the top of the folder structure (`artiq.gateware`). The value that needs to be matched is in `artiq.gateware.eem.Fastino` (though this should be refactored to be configurable from `artiq.gateware.targets.<name>` #1481).

Admittedly, it's been some time since i first used artiq. (Maybe some newer users might want to chip in? @sotirova)

jordens, 2020-09-18:
> I do not mean to remove RTIO Phy from the documentation.

But your suggestion does just that.

Anyway, for these one-word changes, please just file a PR and we can have a look at what you and others really want to read.
Otherwise the discussion-to-benefit ratio is just too low.

> I would simply offer both so that users have a better chance of having encountered the relevant concept.

That's not simple at all. The text would then be confusing two different things. Gateware is much less specific than RTIO Phy. Both are found directly in the directory structure.

> Opaque references like RTIO PHYs

It's an acronym plus an abbreviation. Opaque means that there is no significant other semantic content in an identifier other than its identifying function. That's not the case for acronyms and abbreviations.

> criticisms I repeatedly hear from new users

Sure. There are lots of things to complain about. But the complaints don't imply anything about the quality of changes claiming to address them. You have to show that this change actually makes the situation better.

If you insist that changing the text from "RTIO Phy" to "gateware RTIO Phy" resolves your criticism, then we can do that. But I'm sure this wasn't worth either of your or my time.

hartytp, 2020-09-18:
> Opaque references like RTIO PHYs are one of the criticisms I repeatedly hear from new users. One of the problems is that people don't know where to start looking for these as they have no concept of what they might be.

@pathfinder49 I agree that this is the kind of thing that causes new users confusion. However, it's not clear to me that your solution is the right one. IMHO the documentation for Fastino should be written to be clear and concise on the assumption that users have a certain base of knowledge. If users don't have a good starting point for acquiring that baseline knowledge, making the documentation for a particular PHY verbose is not the appropriate solution; better would be e.g. a glossary in the manual.

hartytp, 2020-09-18:
> Adding a sentence like "log2_width must match the value configured in gateware." would make this more accessible.

I don't think I agree here. The coredevice drivers are expected to solely be instantiated by the device manager, with all arguments supplied from the device db. The device db is expected to have static parameters that match various hardware settings and are not in general (attenuation settings being a bit of an exception) configurable at runtime (i.e. it's for things like RTIO channel numbers, PLL factors, etc).

In the long run, the right way of dealing with this is probably improved tooling that generates as much of the device db as possible from the gateware build so that these kinds of parameters (and, RTIO channel numbers etc) match the gateware without the user having to think.

hartytp, 2020-09-18:
I think these are essentially the same points that @jordens made. But it feels like a bit of an x/y problem to me in that you're trying to fix a symptom of a large problem (users not having basic understanding of how artiq works/terminology) rather than directly solving the problem itself (by providing further high-level documentation)

hartytp, 2020-09-18:
I do, however, agree that bounds checking in the `to_mu` functions feels like it's generally the appropriate thing to do.

commit 56aa22ca (2020-09-03): fastino: document/cleanup

* added documentation on `update`/`hold` mechanism
* mask machine unit values
* cleanup coredevice driver

close #1518