m-labs/artiq #1519: artiq_sinara_tester: Enter keypress detection with while loops gets stuck

HarryMakes, 2020-09-04:
# Bug Report

## One-Line Summary

Given an ARTIQ experiment (e.g. the `SinaraTester` class in artiq/frontend/artiq_sinara_tester.py) containing a while loop checking keypress of "Enter", when the user run it on ARTIQ commit b2572003acfef50b32aa00cd17388f00fed5f87a or above, the loop seems to stop after a few iterations, while this does not happen before this commit.

## Issue Details

### Steps to Reproduce

1.  Git-clone ARTIQ as local repo, and include it in PYTHONPATH with `export`.
2.  Checkout commit b2572003acfef50b32aa00cd17388f00fed5f87a on the local repo.
3.  Prepare a standalone Kasli with or without any peripherals, generate its device_db.py.
3.  Do: `artiq_sinara_tester`
4.  The first LED (on SFP1) blinks for 5 or 6 times, then stops.
5.  Try hitting Enter on the running `artiq_sinara_tester`. It doesn't respond.
6.  Ctrl-C killing the tester and re-run will repeat steps 4-5.

### Expected Behavior

Instead of step 4, the LED should blink continuously. Once the user hit Enter, the second LEd (on SFP2) will blink instead.

### Your System (omit irrelevant parts)

* Operating System:
* ARTIQ version: commit b2572003acfef50b32aa00cd17388f00fed5f87a
* ARTIQ Log shows NO errors.

HarryMakes, 2020-09-04:
Here is a simpler experiment that also shows the inability to keep detecting Enter keypress: https://gist.github.com/HarryMakes/f619d195bbfcbc92cf0fc89fb92ea09d

In commits before b2572003acfef50b32aa00cd17388f00fed5f87a, e.g. 69f0699ebd8db2232a705f61a5655f99fba96edc, when running this experiment, the current LED will stay on until the user press Enter, which is when the next LED will be lit up and stay on until the user press Enter again and end the experiment.

pca006132, 2020-09-04:
The reason for this is due to missing type conversion which raise an exception, cannot concat bool to bytearray.
I'm checking if there is any other problems as well.

pca006132, 2020-09-04:
This patch should be sufficient.
```diff
diff --git a/artiq/coredevice/comm_kernel.py b/artiq/coredevice/comm_kernel.py
index babf035b..98211d59 100644
--- a/artiq/coredevice/comm_kernel.py
+++ b/artiq/coredevice/comm_kernel.py
@@ -316,7 +316,7 @@ class CommKernel:
         self._write(self.pack_float64(value))
 
     def _write_bool(self, value):
-        self._write(1 if value == True else 0)
+        self._write(b'\x01' if value else b'\x00')
 
     def _write_bytes(self, value):
         self._write_int32(len(value))
@@ -425,7 +425,7 @@ class CommKernel:
         elif tag == "b":
             check(isinstance(value, bool),
                   lambda: "bool")
-            self._write_int8(value)
+            self._write_bool(value)
         elif tag == "i":
             check(isinstance(value, (int, numpy.int32)) and
                   (-2**31 < value < 2**31-1),
```

I think we should generate an exhaustive list of RPC with various type for testing... This kind of type error is annoying. I wish there is compile time check :(

closed 2020-09-04 by pca006132

HarryMakes, 2020-09-04:
Test successful with the change. Thanks for the fix!

commit 6195b1d3 (2020-09-04): rpc: fixed _write_bool

Closes #1519