m-labs/artiq #1520: Class variables of type 'bytes' get corrupted when RPC's are involved

jkellerPTB, 2020-09-10:
# Bug Report

## One-Line Summary

Kernel class variables of type bytes get corrupted between calls to different kernel functions when RPC's are involved.

### Steps to Reproduce

```python
class MinExample(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.blist = bytes([2])

    @kernel
    def make_list(self):
        self.blist = bytes([7])
        print("make list", self.blist)  # necessary for error to occur
            
    @kernel
    def run_k(self, n):
        self.core.reset()
        self.make_list()
                
        for _ in range(n):
            self.rpc()
            
        print("num rpc:", n, self.blist)
        
    
    def rpc(self):
        pass
        
    def run(self):
        for n in range(6):
            self.run_k(n)
```

### Actual (undesired) Behavior
Value of self.blist changes with usage of RPC's. Variable turns out different depending on number of RPC's.
```
INFO:dashboard:artiq.dashboard.experiments:Submitted 'repo:MinExample', RID is 586667
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 0 b'\x07'
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 1 b'\xb0'
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 2 b'\x9c'
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 3 b'\x05'
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 4 b'\x03'
INFO:worker(586667,minex.py):print:make list b'\x07'
INFO:worker(586667,minex.py):print:num rpc: 5 b'\x03'
```

### Note
For a class variable of type list(int), we get
```
INFO:dashboard:artiq.dashboard.experiments:Submitted 'repo:MinExample2', RID is 586668
ERROR:worker(586668,minex_list.py):root:While compiling <repository>/minex_list.py
<repository>\minex_list.py:12:22-12:25: error: the assigned value does not outlive the assignment target
        self.ilist = [7]
        ~~~~~~~~~~   ^^^
<repository>\minex_list.py:12:9-12:19: note: the assignment target is still alive after this function returns
        self.ilist = [7]
        ^^^^^^^^^^
<repository>\minex_list.py:11:19-11:23: note: the assignment target is introduced here as a formal argument
    def make_list(self):
                  ^^^^
<repository>\minex_list.py:10:5: note: the assigned value is alive from this point...
    @kernel
    ^
<repository>\minex_list.py:13:39: note: ... to this point
        print("make list", self.ilist)  # necessary for error to occur
                                      ^
ERROR:master:artiq.master.scheduler:got worker exception in run stage, deleting RID 586668
```
Is this error message just missing for the type `bytes`, and accessing class variables across different kernel functions not meant to work in general? For basic types (int, string, float), everything behaves as expected.

### Your System (omit irrelevant parts)

ARTIQ version: ARTIQ v5.7111.5168b831

dnadlinger, 2020-09-11:
This is https://github.com/m-labs/artiq/issues/1497. `bytes` allocates a value on the stack, so assigning the result to a class attribute, where it outlives the function call, is supposed to be illegal. Because of #1497, this is not detected, however. The RPC only plays an indirect role here in that it overwrites the (released, but still in use) stack memory in a fashion not optimised out by the compiler.

closed 2020-09-11 by sbourdeauducq

jkellerPTB, 2020-10-01:
Thanks for clarifying.

Are the basic types affected by #1497 as well (and it was just "lucky" that their memory location wasn't written to by the time we accessed them again), or is assigning those to class attributes legal for some reason?

dnadlinger, 2020-10-02:
It's not assigning to the attributes that isn't allowed, it's the fact that the value that is assigned to it (the buffer pointer/length of the byte array) references stack memory (the actual buffer with the contents of the byte array).

Primitive types don't have any indirections to other memory, so assigning them is fine. (And, thus, so is copying each element of an array of primitive element type individually, which can be a workaround if the array length doesn't change.)