m-labs/artiq #1523: Compilation faliure for numpy array attributes introduced

pathfinder49, 2020-09-30:
# Bug Report

## One-Line Summary
As of 575be2aeca88af00f69fe447b7a2fb1d1bcee8d4, experiments using numpy arrays in the core device passed via attribute from the host nolonger compile.

## Issue Details
Any experiment accessing a numpy array attribute on the core does not compile. The exact error changes through the git history and also depends on the array dimensions. However, as of 575be2aeca88af00f69fe447b7a2fb1d1bcee8d4, compilation fails for 1D and 2D arrays.

The following code demonstrates the issue
```Python3
from artiq.experiment import *
import numpy as np

class Exp(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.arr = np.zeros(1, dtype=np.int32)

    @kernel
    def run(self):
        print(self.arr)
```

### Expected Behavior

Numpy arrays passed as attributes from the host should be handeled by the compiler.

### Error Messages
The error message when run with 575be2aeca88af00f69fe447b7a2fb1d1bcee8d4 is 
```python3
(artiq-upstream) C:\Users\Marius\scratch\master>artiq_compile --class-name Exp exp.py
Traceback (most recent call last):
  File "C:\Users\Marius\Anaconda3\envs\artiq-upstream\Scripts\artiq_compile-script.py", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_compile')()
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\frontend\artiq_compile.py", line 64, in main
    attribute_writeback=False, print_as_rpc=False)
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\coredevice\core.py", line 112, in compile
    library = target.compile_and_link([module])
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\targets.py", line 199, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\targets.py", line 199, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\targets.py", line 152, in compile
    llmod = module.build_llvm_ir(self)
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\module.py", line 93, in build_llvm_ir
    attribute_writeback=self.attribute_writeback)
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 534, in process
    self.process_function(func)
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 695, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "c:\users\marius\scratch\upstream_artiq\artiq\artiq\compiler\transforms\llvm_ir_generator.py", line 1006, in process_Arith
    name=insn.name)
  File "C:\Users\Marius\Anaconda3\envs\artiq-upstream\lib\site-packages\llvmlite_artiq\ir\builder.py", line 24, in wrapped
    % (lhs.type, rhs.type))
ValueError: Operands must be the same type, got ({i32*, i32}, i32)
```
The error message changes several times. On the current master (as of 1c645d8857f27db72fa965b18d45b570aef99a71) the error for 1D and 2D arrays is:
```python3
AttributeError: 'TVar' object has no attribute 'name'
```

airwoodix, 2020-10-19:
We (obviously) observe the same on 90017da and it breaks a number of our experiments (our fault for running on beta). Is there a path towards solving this issue? I'm afraid I don't qualify here...

dnadlinger, 2020-10-19:
Looking into it. Arrays should just be directly emitted instead of synthesising AST literals (which is where this comes from).

dnadlinger, 2020-10-19:
(575be2a etc. weren't supposed to work in the first place, but the last commit in that merge should – and does, except for this, which, at a glance, is related to polymorphic ints.)

dnadlinger, 2020-10-19:
This is now fixed in master. It is, however, quite likely that there are still some regressions left concerning NumPy array support, as the in-tree tests are very limited. Previously, this wasn't much of an issue, since arrays were implemented as a thin veneer on top of lists in the compiler (which are pervasively used, and thus well-tested). The coming version, however, will provide proper multi-dimensional array support (including matrix operations, etc.), so the innards are now completely different. I don't have a ndarray-heavy code base to test this against myself (our experiments have been running master all along), so any further regression reports will be much appreciated.

PR #1537 (2020-10-19): compiler: Array quoting fixes
Fixes #1523; here to test Hydra integration.
closed 2020-10-19 by dnadlinger