m-labs/artiq #1524: Using unsigned ints on core device

ljstephenson, 2020-10-02:
# ARTIQ Feature Request

Unsigned ints aren't implemented on the core device - I seem to remember hearing a reason once for not doing it, but that has escaped me.

## Problem this request addresses

I am implementing an algorithm that uses unsigned integer math on the core device. Currently I'm having to use quite a few hacks to get around the differing behaviours of signed vs unsigned ints; I'm in the process of determining whether these hacks are a significant performance bottleneck, but they definitely introduce more code and hence more potential bugs in functions that I would expect to "just work" natively.

It also seems that the lack of unsigned integers has caused issues in the past (e.g. 3e84ec2b)

## Describe the solution you'd like

Implement unsigned ints on the core device

## Additional context

See example code for some of the workarounds, some of which are more onerous than others (i.e. comparison introduces extra branching). Also there is some unexpected behaviour with 64-bit casting; this may be a bug.

```python
from artiq.experiment import *
from numpy import int32, int64, uint32, uint64

class Ints(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    def run(self):
        # 0xffffffff = -1 in 32-bit signed
        x = int32(-1)
        print(hex(uint64(self.cast(x))))          # 0xffffffffffffffff = -1 in 64-bit signed
        print(hex(uint64(self.unsigned_cast(x)))) # 0xffffffff
        print(type(self.unsigned_cast(x)))        # <class 'numpy.int64'> as expected

        x = int32(-1)
        print(hex(uint64(int64(2**32 + x))))      # 0xffffffff as expected
        print(hex(uint64(self.cast_bug1(x))))     # 0xffffffffffffffff UNEXPECTED
        print(hex(uint64(self.cast_bug2(x))))     # 0xffffffffffffffff UNEXPECTED
        print(hex(uint64(self.cast_bug3(x))))     # 0xffffffffffffffff UNEXPECTED

        x = int32(-1)
        y = int32(31)
        print(self.bitshift(x, y))          # still -1, padded with sign bit as expected
        print(self.unsigned_bitshift(x, y)) # 1 as expected

        x = int32(-1)
        y = int32(1)
        print(self.less_than(x, y))          # true, as expected
        print(self.unsigned_less_than(x, y)) # false, as expected

    @kernel
    def cast(self, x:TInt32) -> TInt64:
        return int64(x)

    @kernel
    def unsigned_cast(self, x: TInt32) -> TInt64:
        return int64(x) & 0xffffffff

    @kernel
    def cast_bug1(self, x:TInt32) -> TInt64:
        return int64(2**32) + int64(x)

    @kernel
    def cast_bug2(self, x:TInt32) -> TInt64:
        return int64(2**32 + int64(x))

    @kernel
    def cast_bug3(self, x:TInt32) -> TInt64:
        return int64(int64(2**32) + int64(x))

    @kernel
    def bitshift(self, x:TInt32, bits:TInt32) -> TInt32:
        bits &= 31
        return x >> bits

    @kernel
    def unsigned_bitshift(self, x:TInt32, bits:TInt32) -> TInt32:
        bits &= 31
        mask = ~(~int32(0) << (32 - bits))
        return int32((x >> bits) & mask)

    @kernel
    def less_than(self, x:TInt32, y:TInt32) -> TBool:
        return x < y

    @kernel
    def unsigned_less_than(self, x:TInt32, y:TInt32) -> TBool:
        if not (x ^ y) & 0x80000000:
            # x and y have same sign, so comparison will be correct either way
            return x < y
        else:
            # different sign, if y < 0 then x < y in unsigned
            return y < 0
```

sbourdeauducq, 2020-12-02:
> I am implementing an algorithm that uses unsigned integer math on the core device.

> It also seems that the lack of unsigned integers has caused issues in the past (e.g. [3e84ec2](https://github.com/m-labs/artiq/commit/3e84ec2bf1d32e063dd22f59f65aef62a24c9911))

This is just because this code exploits the overflow. Is it also the case for your algorithm or are there additional reasons for wanting unsigned ints?

dhslichter, 2020-12-02:
We have a particular use case that involves exploiting overflow for unsigned ints, but in general it seems we should have the ability to use unsigned ints in kernels.  This reduces the chances of subtle errors in a variety of contexts (like the DDS register values above, but also for any similar tasks), and increases flexibility for users who may have a preference for a particular algorithm implementation.  Is there substantial penalty/complexity involved with implementing unsigned ints?

sbourdeauducq, 2020-12-03:
> This reduces the chances of subtle errors in a variety of contexts

> increases flexibility for users who may have a preference for a particular algorithm implementation

When all values are positive and there is no overflow, signed ints behave exactly the same as unsigned but with one bit less, no?

ljstephenson, 2020-12-10:
> When all values are positive ...

In my use case, overflows are normal and expected, so not having to write custom right shifts, comparisons, etc would be nice.

dhslichter, 2020-12-11:
> When all values are positive and there is no overflow, signed ints behave exactly the same as unsigned but with one bit less, no?

And for example with DDS FTWs, one needs all 32 bits (thus the issue with behavior there if one isn't careful).  

I guess the question is -- what is the downside to implementing both signed and unsigned ints?  Does it present any major issues or cause any performance degradations?

jordens, 2020-12-11:
DDS FTWs are twos complement (signed) integers. Treating them as unsigned is not helpful in the best of cases and fundamentally wrong in the worst.

dhslichter, 2020-12-11:
[image]
Not sure where it indicates that the FTW is signed in any of this.  If one seeks to program a frequency that is in the second Nyquist zone (something that we do), then using an FTW greater than 2^31 is appropriate.  The FTW is just added to the 32-bit unsigned phase accumulator each clock cycle of the DDS, and the overflow of the 32-bit unsigned addition is the phase wrapping.  Having an FTW greater than 2^31 just means you overflow on average more than once every two clock cycles.  This is accurate for generating a tone above fclock/2 (second Nyquist), of course it also produces the aliased image in the first Nyquist zone too on the output.  But from a math standpoint for programming the DDS, it's valid.  Of course, for a tone in the second Nyquist zone, one could program the FTW for the corresponding first Nyquist frequency and produce the same result.  However, it represents extra math and mental "don't screw this up, should I be adding or subtracting" load to do so.

dhslichter, 2020-12-11:
I return to my previous question: what is the downside to implementing both signed and unsigned ints? Does it present any major issues or cause any performance degradations?

sbourdeauducq, 2020-12-12:
There won't be performance degradations, it's just more code and a more complicated type system. There would be four integer types (32/64 signed/unsigned) instead of two.

sbourdeauducq, 2021-12-13:
Transferred to NAC3.

closed 2021-12-13 by sbourdeauducq

dimitsev, 2025-05-12:
The hack below works on Artiq 9 beta with the old compiler. (Copied from [my M-labs forum post](https://forum.m-labs.hk/d/950-unsigned-32-bit-integers-in-kernel-functions).)

### Unsigned 32-bit integers in kernel functions

The experiment below linearly maps a float between `0.0` and `1.0` to an unsigned 32-bit integer between `0` and `2**32-1`. The conversion happens on a Kasli SoC, completely without RPC calls.

Beware that you should run this experiment on a real hardware device. Running with `artiq/examples/no_hardware/device_db.py` changes the results.

```python
from artiq.experiment import EnvExperiment, kernel, rpc
from artiq.language.types import TInt32, TFloat
import numpy as np

@rpc
def print_binary(number, type_cast, nr_bits):
    bits = ""
    for i in range(nr_bits):
        bits = str(i % 10) + bits
    print("bits    :", bits)
    print("binary  :", f"{type_cast(number):{int(nr_bits)}b}")

@rpc
def rpc_python_benchmark(amplitude):
    r"""Linearly maps amplitude ∈ [0.0, 1.0] to an unsigned 32-bit integer {0,1,..., 2**32-1}.
    Comparison function for `kernel_hack()` with python types."""
    if amplitude < 0.0 or amplitude > 1.0:
        raise ValueError("Invalid AD9910 fractional amplitude!")
    if amplitude > 0.9999999998: # increasing the last digit from 8 to 9 fails
        amplitude = 0.9999999998
    x = round(amplitude * (1 << 32))
    print("decimal :", x)
    print_binary(x, np.uint32, 32)

@rpc
def rpc_numpy_benchmark(amplitude: TFloat):
    r"""Linearly maps amplitude ∈ [0.0, 1.0] to an unsigned 32-bit integer {0,1,..., 2**32-1}.
    Comparison function for `kernel_hack()` with numpy types."""
    if amplitude < 0.0:
        raise ValueError("Invalid AD9910 fractional amplitude!")
    elif amplitude <= 0.4999999998: # increasing the last digit from 8 to 9 fails
        x = np.int32(round(amplitude * (1 << 32)))
    elif amplitude <= 0.9999999998: # increasing the last digit from 8 to 9 fails
        x = np.int32(-round((1-amplitude) * (1 << 32))) # the minus sign is a hack
    elif amplitude <= 1.0:
        x = np.int32(-round((1-0.9999999998) * (1 << 32)))
    else:
        raise ValueError("Invalid AD9910 fractional amplitude!")
    print("decimal :", x)
    print_binary(x, np.uint32, 32)

@kernel
def kernel_hack(amplitude: TFloat) -> TInt32:
    r"""Linearly maps amplitude ∈ [0.0, 1.0] to an unsigned 32-bit integer {0,1,..., 2**32-1}.
    Hacking is necessary because the ARTIQ compiler does *not* know unsigned integers."""
    if amplitude < 0.0:
        raise ValueError("Invalid AD9910 fractional amplitude!")
    elif amplitude <= 0.4999999998: # increasing the last digit from 8 to 9 fails
        # no clue why minus sign is inverted compared to `rpc_numpy_benchmark`
        return np.int32(-round(amplitude * 2 * (1 << 31))) # (1 << 32) produces gargabe
    elif amplitude <= 0.9999999998: # increasing the last digit from 8 to 9 fails
        # no clue why minus sign is inverted compared to `rpc_numpy_benchmark`
        return np.int32(round((1-amplitude) * 2 * (1 << 31))) # (1 << 32) produces gargabe
    elif amplitude <= 1.0:
        return np.int32(round((1-0.9999999998) * 2 * (1 << 31))) # (1 << 32) produces gargabe
    else:
        raise ValueError("Invalid AD9910 fractional amplitude!")
    return np.int32(0) # prevents compiler crash

class UnsignedIntegerTest(EnvExperiment):
    """
    Do must run this with *real* hardware in `device_db.py`.
    We have tested unsigned integer hacks with
    `artiq/examples/no_hardware/device_db.py` before,
    and the results were *different* from the real hardware.
    """
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        amplitude = 0.5
        print("----------- RPC PYTHON TYPES -----------")
        rpc_python_benchmark(amplitude)
        print("----------- RPC NUMPY TYPES -----------")
        rpc_numpy_benchmark(amplitude)
        print("----------- KERNEL HACK -----------")
        x = kernel_hack(amplitude)
        print("decimal :", x)
        print_binary(x, np.uint32, 32)
```

### Why did we do all this?

The AD9910's digital ramp generator accepts unsigned 32-bit integers in these registers:

- `Digital Ramp Limit Register—Address 0x0B`
- `Digital Ramp Step Size Register—Address 0x0C`

See pages 28, 29 and 59 of the [AD9910 datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/ad9910.pdf).

Unfortunately, `artiq.coredevice.AD9910.amplitude_to_asf` (see [artiq/coredevice/ad9910.py#L747](https://github.com/m-labs/artiq/blob/b4d88f009b384d910397de1b0e9b2c4b0a01f12c/artiq/coredevice/ad9910.py#L747)) does not handle and cannot be adapted to handle unsigned integers.