m-labs/artiq #1534: WRPLL on Kasli 2.0

sbourdeauducq, 2020-10-15:


hartytp, 2020-10-18:
Current status:
- everything is nominally implemented, just needs testing and improved diagnostics
- testing with a satellite Kasli 2.0 at M-Labs, master is clocked from the Si5324 on-board xtal
- we've verified that the DDMTD and Si549s on the satellites behave as expected. for example, we can set an offset on the satellites and then measure the expected frequency difference on the DDMTD
- the main<->helper DCXO tags look good (jitter of a few tags)
- the CDR <-> helper rms jitter is a few times worse, but we see occasional phase jumps of nearly 500ns! Not sure if that's the transceiver PLLs or the xtal, but we should definitely test with a better frequency reference soon
- [ ] the phase unwrapping in the ddmtd core is not currently robust enough to handle the large phase jumps in the recovered clock. that should be easy, but still to do
- [ ] waiting for Weida to produce a finalised main DCXO loop filter and simulation results that we can test the artiq implementation against
- in the mean time, I've tried locking the helper DCXO to the free-running main DCXO. Before enabling the lock, I detune the main DCXO by 250Hz from its nominal. I then let the loop run for 100 cycles (4kHz cycle rate) before sleeping for 1s and taking another 100 data points. This is what I see...

```

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2020 M-Labs Limited

Bootloader CRC passed
Gateware ident 6.unknown.beta;satellite
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000011111111111000000000
Module 0:
00000000000111111111111000000000
Read leveling: 17+-5 16+-6 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005611s]  INFO(satman): software ident 6.unknown.beta;satellite
[     0.011890s]  INFO(satman): gateware ident 6.unknown.beta;satellite
[     0.163573s]  INFO(board_artiq::wrpll): initializing WR PLL...
[     1.036086s]  INFO(satman): uplink is up, switching to recovered clock
[     1.041252s]  INFO(board_artiq::wrpll): switching to recovered clock
[     1.049122s]  INFO(board_artiq::wrpll): Untrimmed oscillator frequencies:
[     1.254548s]  INFO(board_artiq::wrpll): helper clock frequency: 124995kHz (2314009)
[     1.260837s]  INFO(board_artiq::wrpll): main clock frequency: 124998kHz (2314079)
[     1.268323s]  INFO(board_artiq::wrpll): CDR clock frequency: 125000kHz (2314111)
[     1.275713s]  INFO(board_artiq::wrpll): Trimming oscillator frequencies...
[     1.282591s]  INFO(board_artiq::wrpll): sys count err -13
[     1.287987s]  INFO(board_artiq::wrpll): main counts err 32
[     1.293472s]  INFO(board_artiq::wrpll): helper counts err 32
[     1.299152s]  INFO(board_artiq::wrpll): ADPLL offsets: helper=118803 main=118799
[     1.416739s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: 39.708458 Hz (0.31767735 ppm), jitter 1.9741378 ps (7.896551 tags)
[     1.429112s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: 62.92658 Hz (0.503428 ppm), jitter 11.491379 ps (45.965515 tags)
[     1.441040s]  INFO(board_artiq::wrpll): increasing main DCXO by 1ppm (125Hz):
[     1.557341s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: -206.81953 Hz (-1.6546068 ppm), jitter 28.525864 ps (114.103455 tags)
[     1.569988s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: 59.771427 Hz (0.47818598 ppm), jitter 15.051725 ps (60.2069 tags)
[     1.583921s]  INFO(board_artiq::wrpll): new ADPLL: 118803 135981
[     1.848692s]  INFO(board_artiq::wrpll): main tag 29913, helper_diff -1720, helper adpll 8
[     1.855493s]  INFO(board_artiq::wrpll): main tag 28190, helper_diff -1723, helper adpll 12
[     1.863763s]  INFO(board_artiq::wrpll): main tag 26480, helper_diff -1710, helper adpll 16
[     1.872033s]  INFO(board_artiq::wrpll): main tag 24770, helper_diff -1710, helper adpll 19
[     1.880303s]  INFO(board_artiq::wrpll): main tag 23044, helper_diff -1726, helper adpll 23
[     1.888574s]  INFO(board_artiq::wrpll): main tag 21331, helper_diff -1713, helper adpll 26
[     1.896844s]  INFO(board_artiq::wrpll): main tag 19620, helper_diff -1711, helper adpll 30
[     1.905114s]  INFO(board_artiq::wrpll): main tag 17910, helper_diff -1710, helper adpll 33
[     1.913384s]  INFO(board_artiq::wrpll): main tag 16190, helper_diff -1720, helper adpll 37
[     1.921654s]  INFO(board_artiq::wrpll): main tag 14479, helper_diff -1711, helper adpll 40
[     1.929924s]  INFO(board_artiq::wrpll): main tag 12762, helper_diff -1717, helper adpll 44
[     1.938194s]  INFO(board_artiq::wrpll): main tag 11049, helper_diff -1713, helper adpll 48
[     1.946464s]  INFO(board_artiq::wrpll): main tag 9339, helper_diff -1710, helper adpll 51
[     1.954647s]  INFO(board_artiq::wrpll): main tag 7632, helper_diff -1707, helper adpll 55
[... 370 lines cut ...]
[     6.314767s]  INFO(board_artiq::wrpll): main tag 20313, helper_diff -56, helper adpll 14063
[     6.323124s]  INFO(board_artiq::wrpll): main tag 20247, helper_diff -66, helper adpll 14063
[     6.331481s]  INFO(board_artiq::wrpll): main tag 20211, helper_diff -36, helper adpll 14063
[     6.339838s]  INFO(board_artiq::wrpll): main tag 20154, helper_diff -57, helper adpll 14064
[     6.348195s]  INFO(board_artiq::wrpll): main tag 20109, helper_diff -45, helper adpll 14064
[... cut ...]
[     6.440123s]  INFO(board_artiq::wrpll): main tag 19545, helper_diff -52, helper adpll 14066
[     6.448480s]  INFO(board_artiq::wrpll): main tag 19492, helper_diff -53, helper adpll 14066
[     6.506841s]  INFO(board_artiq::wrpll): error 1 0
[     6.512090s]  INFO(board_artiq::wrpll): new ADPLL: 133438 135981
[     6.516717s]  INFO(board_artiq::wrpll): clock transition completed
[     6.791689s]  INFO(satman): TSC loaded from uplink
[     6.895659s]  INFO(satman): rank: 1
[     6.897775s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; }

```

hartytp, 2020-10-18:
NB the non-zero helper tags don't go away if we leave the loop for 10s to settle. Not sure if that's an issue with the loop not quite keeping up with the drift in the main DCXO? Anyway, it does lock down to a pretty low jitter so seems to basically work...

hartytp, 2020-10-18:
I *think* the residual error is just because the helper PLL isn't designed to be overly tight. In the data set above, you can see that the helper ADPLL (tunes the DCXO frequency with 1LSB =0.1ppb = 12.5mHz for 125MHz f_rtio) are moving in the right direction but the tags (phase difference between the two DCXOs with 1 tag = 50fs) are staying pretty constant. That suggests that what we're seeing is finite phase error due to finite low-frequency gain in the loop filter. The residual phase error there is ~12.5ps. That's potentially quite large since it implies that if the DCXO drift changed (e.g. due to environmental changes) we could get a phase shift by that much. Again, I assume this is just the helper PLL being deliberately a bit weak...let's see what happens with the main DCXO...

hartytp, 2020-10-19:
Robust phase unwrapping fixed.

Helper DCXO locked to the main DCXO with the filter input gain turned up a bit (bit shift increased from 22 to 28).

```

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2020 M-Labs Limited

Bootloader CRC passed
Gateware ident 6.unknown.beta;satellite
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000011111111111000000000
Module 0:
00000000000011111111111000000000
Read leveling: 17+-5 17+-5 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005612s]  INFO(satman): software ident 6.unknown.beta;satellite
[     0.011891s]  INFO(satman): gateware ident 6.unknown.beta;satellite
[     0.163573s]  INFO(board_artiq::wrpll): initializing WR PLL...
[     0.804580s]  INFO(satman): uplink is up, switching to recovered clock
[     0.809746s]  INFO(board_artiq::wrpll): switching to recovered clock
[     0.817608s]  INFO(board_artiq::wrpll): Untrimmed oscillator frequencies:
[     1.023034s]  INFO(board_artiq::wrpll): helper clock frequency: 124995kHz (2314010)
[     1.029324s]  INFO(board_artiq::wrpll): main clock frequency: 124998kHz (2314079)
[     1.036811s]  INFO(board_artiq::wrpll): CDR clock frequency: 125000kHz (2314111)
[     1.044200s]  INFO(board_artiq::wrpll): Trimming oscillator frequencies...
[     1.051078s]  INFO(board_artiq::wrpll): sys count err -13
[     1.056475s]  INFO(board_artiq::wrpll): main counts err 32
[     1.061959s]  INFO(board_artiq::wrpll): helper counts err 31
[     1.067639s]  INFO(board_artiq::wrpll): ADPLL offsets: helper=115090 main=118799
[     1.184985s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: -36.677742 Hz (-0.2934309 ppm), jitter 8191.982 ps (32767.928 tags)
[     1.197248s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: -12.604587 Hz (-0.10083977 ppm), jitter 8191.932 ps (32767.729 tags)
[     1.209724s]  INFO(board_artiq::wrpll): increasing main DCXO by 1ppm (125Hz):
[     1.325872s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: -285.2731 Hz (-2.2822545 ppm), jitter 8191.862 ps (32767.447 tags)
[     1.338046s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: -13.889131 Hz (-0.11111643 ppm), jitter 8191.922 ps (32767.688 tags)
[     1.353192s]  INFO(board_artiq::wrpll): new ADPLL: 115090 118799
[     1.623465s]  INFO(board_artiq::wrpll): ref_tag 82527164, main tag 81223883, main_diff -1303281, helper_diff -291, helper adpll 225, main adpll -321693
[     1.635666s]  INFO(board_artiq::wrpll): ref_tag 82559909, main tag 81256362, main_diff -1303547, helper_diff -289, helper adpll 264, main adpll -384957
[     1.649245s]  INFO(board_artiq::wrpll): ref_tag 82592512, main tag 81288848, main_diff -1303664, helper_diff -282, helper adpll 303, main adpll -442471
[     1.662826s]  INFO(board_artiq::wrpll): ref_tag 82625225, main tag 81321344, main_diff -1303881, helper_diff -272, helper adpll 341, main adpll -493198
[     1.676406s]  INFO(board_artiq::wrpll): ref_tag 82657894, main tag 81353814, main_diff -1304080, helper_diff -298, helper adpll 377, main adpll -536704
[     1.689986s]  INFO(board_artiq::wrpll): ref_tag 82690617, main tag 81386324, main_diff -1304293, helper_diff -258, helper adpll 415, main adpll -573008
[     1.703566s]  INFO(board_artiq::wrpll): ref_tag 82723286, main tag 81418834, main_diff -1304452, helper_diff -258, helper adpll 452, main adpll -602456
[     1.717147s]  INFO(board_artiq::wrpll): ref_tag 82755934, main tag 81451340, main_diff -1304594, helper_diff -262, helper adpll 486, main adpll -625613
[     1.730727s]  INFO(board_artiq::wrpll): ref_tag 82788678, main tag 81483851, main_diff -1304827, helper_diff -257, helper adpll 521, main adpll -643176
[     1.744307s]  INFO(board_artiq::wrpll): ref_tag 82821401, main tag 81516370, main_diff -1305031, helper_diff -249, helper adpll 555, main adpll -655904
[     1.757888s]  INFO(board_artiq::wrpll): ref_tag 82854102, main tag 81548889, main_diff -1305213, helper_diff -249, helper adpll 589, main adpll -664567
[     1.771468s]  INFO(board_artiq::wrpll): ref_tag 82886761, main tag 81581411, main_diff -1305350, helper_diff -246, helper adpll 622, main adpll -669907
[     1.785048s]  INFO(board_artiq::wrpll): ref_tag 82919526, main tag 81613950, main_diff -1305576, helper_diff -229, helper adpll 655, main adpll -672611
[     1.798628s]  INFO(board_artiq::wrpll): ref_tag 82952259, main tag 81646492, main_diff -1305767, helper_diff -226, helper adpll 686, main adpll -673296
[     1.812209s]  INFO(board_artiq::wrpll): ref_tag 82984992, main tag 81679011, main_diff -1305981, helper_diff -249, helper adpll 716, main adpll -672500
[     1.825789s]  INFO(board_artiq::wrpll): ref_tag 83017725, main tag 81711573, main_diff -1306152, helper_diff -206, helper adpll 748, main adpll -670678
[     1.839369s]  INFO(board_artiq::wrpll): ref_tag 83050480, main tag 81744085, main_diff -1306395, helper_diff -256, helper adpll 778, main adpll -668205
[     1.852950s]  INFO(board_artiq::wrpll): ref_tag 83083223, main tag 81776668, main_diff -1306555, helper_diff -185, helper adpll 809, main adpll -665382
[     1.866530s]  INFO(board_artiq::wrpll): ref_tag 83115978, main tag 81809219, main_diff -1306759, helper_diff -217, helper adpll 838, main adpll -662441
[... cut ...]
[    28.433218s]  INFO(board_artiq::wrpll): ref_tag 1731940577, main tag 1730269841, main_diff -1670736, helper_diff -3, helper adpll 2622, main adpll -826338
[    28.530506s]  INFO(board_artiq::wrpll): error 0 0
[    28.535772s]  INFO(board_artiq::wrpll): new ADPLL: 118277 118799
[    28.540405s]  INFO(board_artiq::wrpll): clock transition completed
[    28.832389s]  INFO(satman): TSC loaded from uplink
[    28.933544s]  INFO(satman): rank: 1
[    28.935659s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; }
```

That locks with a mean absolute deviation of 7 tags (~1.75ps).

hartytp, 2020-10-19:
Also confirmed we can lock the helper to the reference (CDR) clock
```

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2020 M-Labs Limited

Bootloader CRC passed
Gateware ident 6.unknown.beta;satellite
Initializing SDRAM...
Read leveling scan:
Module 1:
00000000000011111111111000000000
Module 0:
00000000000111111111111000000000
Read leveling: 17+-5 16+-6 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000004s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005611s]  INFO(satman): software ident 6.unknown.beta;satellite
[     0.011890s]  INFO(satman): gateware ident 6.unknown.beta;satellite
[     0.163574s]  INFO(board_artiq::wrpll): initializing WR PLL...
[     0.878471s]  INFO(satman): uplink is up, switching to recovered clock
[     0.883637s]  INFO(board_artiq::wrpll): switching to recovered clock
[     0.891492s]  INFO(board_artiq::wrpll): Untrimmed oscillator frequencies:
[     1.096920s]  INFO(board_artiq::wrpll): helper clock frequency: 124995kHz (2314010)
[     1.103210s]  INFO(board_artiq::wrpll): main clock frequency: 124998kHz (2314079)
[     1.110696s]  INFO(board_artiq::wrpll): CDR clock frequency: 125000kHz (2314111)
[     1.118086s]  INFO(board_artiq::wrpll): Trimming oscillator frequencies...
[     1.124964s]  INFO(board_artiq::wrpll): sys count err -13
[     1.130360s]  INFO(board_artiq::wrpll): main counts err 32
[     1.135845s]  INFO(board_artiq::wrpll): helper counts err 31
[     1.141524s]  INFO(board_artiq::wrpll): ADPLL offsets: helper=115090 main=118799
[     1.258710s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: -37.235706 Hz (-0.29789475 ppm), jitter 2.939655 ps (11.75862 tags)
[     1.271011s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: -25.024517 Hz (-0.20020226 ppm), jitter 11.310346 ps (45.241383 tags)
[     1.283536s]  INFO(board_artiq::wrpll): increasing main DCXO by 1ppm (125Hz):
[     1.399659s]  INFO(board_artiq::wrpll): MAIN <-> helper frequency difference: -285.94745 Hz (-2.2876496 ppm), jitter 73.63793 ps (294.55173 tags)
[     1.411945s]  INFO(board_artiq::wrpll): REF <-> helper frequency difference: -26.272934 Hz (-0.21018988 ppm), jitter 14.672414 ps (58.689655 tags)
[     1.427150s]  INFO(board_artiq::wrpll): new ADPLL: 115090 118799
[     1.696198s]  INFO(board_artiq::wrpll): ref_tag 91586627, main tag 90454055, main_diff -1132572, helper_diff -258, helper adpll 188, main adpll -334575
[     1.708399s]  INFO(board_artiq::wrpll): ref_tag 91619221, main tag 90486521, main_diff -1132700, helper_diff -174, helper adpll 214, main adpll -384547
[     1.721979s]  INFO(board_artiq::wrpll): ref_tag 91651816, main tag 90518993, main_diff -1132823, helper_diff -173, helper adpll 243, main adpll -428616
[     1.735559s]  INFO(board_artiq::wrpll): ref_tag 91684368, main tag 90551467, main_diff -1132901, helper_diff -216, helper adpll 266, main adpll -466406
[     1.749139s]  INFO(board_artiq::wrpll): ref_tag 91716911, main tag 90583933, main_diff -1132978, helper_diff -225, helper adpll 292, main adpll -497935
[     1.762720s]  INFO(board_artiq::wrpll): ref_tag 91749525, main tag 90616420, main_diff -1133105, helper_diff -154, helper adpll 321, main adpll -523502
[     1.776300s]  INFO(board_artiq::wrpll): ref_tag 91782068, main tag 90648918, main_diff -1133150, helper_diff -225, helper adpll 346, main adpll -543600
[     1.789880s]  INFO(board_artiq::wrpll): ref_tag 91814576, main tag 90681396, main_diff -1133180, helper_diff -260, helper adpll 371, main adpll -558834
[     1.803460s]  INFO(board_artiq::wrpll): ref_tag 91847191, main tag 90713885, main_diff -1133306, helper_diff -153, helper adpll 404, main adpll -569863
[     1.817041s]  INFO(board_artiq::wrpll): ref_tag 91879884, main tag 90746371, main_diff -1133513, helper_diff -75, helper adpll 431, main adpll -577357
[     1.830534s]  INFO(board_artiq::wrpll): ref_tag 91912434, main tag 90778870, main_diff -1133564, helper_diff -218, helper adpll 446, main adpll -581964
[     1.844114s]  INFO(board_artiq::wrpll): ref_tag 91945017, main tag 90811377, main_diff -1133640, helper_diff -185, helper adpll 466, main adpll -584281
[     1.857695s]  INFO(board_artiq::wrpll): ref_tag 91977645, main tag 90843867, main_diff -1133778, helper_diff -140, helper adpll 492, main adpll -584844
[     1.871275s]  INFO(board_artiq::wrpll): ref_tag 92010208, main tag 90876357, main_diff -1133851, helper_diff -205, helper adpll 514, main adpll -584119
[     1.884855s]  INFO(board_artiq::wrpll): ref_tag 92042865, main tag 90908888, main_diff -1133977, helper_diff -111, helper adpll 537, main adpll -582501
[     1.898435s]  INFO(board_artiq::wrpll): ref_tag 92075460, main tag 90941386, main_diff -1134074, helper_diff -173, helper adpll 558, main adpll -580317
[     1.912016s]  INFO(board_artiq::wrpll): ref_tag 92107885, main tag 90973913, main_diff -1133972, helper_diff -343, helper adpll 577, main adpll -577828
[     1.925596s]  INFO(board_artiq::wrpll): ref_tag 92140734, main tag 91006414, main_diff -1134320, helper_diff 81, helper adpll 611, main adpll -575233
[     1.939003s]  INFO(board_artiq::wrpll): ref_tag 92173267, main tag 91038942, main_diff -1134325, helper_diff -235, helper adpll 628, main adpll -572687
[... cut ...]
[    28.966925s]  INFO(satman): TSC loaded from uplink
[    29.068070s]  INFO(satman): rank: 1
[    29.070185s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; }
```

That locks with a mean abs error of 40 tags (10ps), presumably due to the higher noise floor of the transceiver PLL and/or reference oscillator used here.

Take aways: even during initial lock acquisition transients we don't see any glitches; the lock tracks in nicely and holds lock nicely. Next step is to get the main DCXO locked.

hartytp, 2020-10-19:
Some thoughts about "phase unwrapping". This is really a misnomer since we now use a 64-bit DDMTD counter so there are strictly no phase wrapping issues (which there were with the N-bit counter). The issue is really now just cycle skip corrections.

These skips happen because the DDMTD "collector" produces one tag difference output per pair of input tags, one from each DDMTD input. If the oscillator frequencies are not locked, one oscillator can lap the other, leading to a cycle of that oscillator being lost (a second tag comes in while we're still waiting for a tag from the other oscillator). When that happens, without a cycle slip correction, we get a large jump in the tag value due to the missed tag.  This gives the loop filters a huge kick and can easily end up with the PLL being unable to lock (these large kicks come in faster than the loop filter can correct for them).

We correct for this by looking for large discontinuities in the tag values and then adding/subtracting on the DDMTD counter period (2^15 tags) as required to remove the discontinuity. If the oscillators are close to locked then the amount we missed during the skip will be close to 2^15 so this will correct for the tag we missed. If we're far from lock (so that the increase in tags each cycle is not 2^15) there will still be a discontinuity, but far smaller than if we hadn't applied a correction.

There are probably better ways of implementing this correction: rather than looking for discontinuities, it's probably easier to just watch for tags arriving while we're in a wait state; rather than applying a blanket correction of 2^N we could base the correction on the tag we missed; etc. However, these glitches only occur during lock acquisition, not once the PLL is locked, and the current implementation gets us to lock as fast as any other and with a decently low resource usage, so I'm not inclined to put any time into improvements. It would be interesting to compare with CERN's approach, but I didn't find their implementation particularly easy to follow when I last looked at it...