m-labs/artiq #1539: artiq.test.coredevice.test_embedding.ArrayQuotingTest fails on Zynq

sbourdeauducq, 2020-10-23:
```
======================================================================
ERROR: test_quoting (artiq.test.coredevice.test_embedding.ArrayQuotingTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/nix/store/dqfdq864hrn1v8xs3glcrgd7zqcim70h-python3.8-artiq-6.7477.d672d2fc.beta/lib/python3.8/site-packages/artiq/test/coredevice/test_embedding.py", line 462, in test_quoting
    self.create(_ArrayQuoting).run()
  File "/nix/store/dqfdq864hrn1v8xs3glcrgd7zqcim70h-python3.8-artiq-6.7477.d672d2fc.beta/lib/python3.8/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/nix/store/dqfdq864hrn1v8xs3glcrgd7zqcim70h-python3.8-artiq-6.7477.d672d2fc.beta/lib/python3.8/site-packages/artiq/coredevice/core.py", line 135, in run
    self.comm.load(kernel_library)
  File "/nix/store/dqfdq864hrn1v8xs3glcrgd7zqcim70h-python3.8-artiq-6.7477.d672d2fc.beta/lib/python3.8/site-packages/artiq/coredevice/comm_kernel.py", line 363, in load
    raise LoadError(self._read_string())
artiq.coredevice.comm_kernel.LoadError: core1 failed to process data

----------------------------------------------------------------------
```

Device log:
```
[  3093.860055s] ERROR(runtime::kernel::core1): failed to load shared library: symbol lookup error: abort
```

sbourdeauducq, 2020-10-23:
That's because this new test uses ``assert`` in the kernel, which the compiler implements with a detailed error logged and then a call to ``abort`` (see ``visit_Assert`` in ``artiq_ir_generator.py``). This is inconsistent with CPython behavior where a failed assert raises ``AssertionError``.

I'm not sure if this behavior is useful, maybe assert support should just be dropped from the compiler...

dnadlinger, 2020-10-23:
Ack, looking into it.

dnadlinger, 2020-11-04:
I've implemented asserts via `AssertionError`s, as I find them quite useful/natural to have. (Maybe my sense of style for Python is odd; who knows.) https://github.com/m-labs/artiq/pull/1541, still needs testing before merging.

PR #1541 (2020-11-09): compiler: Raise exception on failed assert()s rather than panic
This allows assert() to be used on Zynq, where abort() is not
currently implemented for kernels. Furthermore, this is arguably
the more natural implementation of assertions on all kernel targets
(i.e. where embedding into host Python is used), as this way, the
exception information actually makes it to the user.

Since this does not implement printing of the subexpressions, I
left the old print+abort implementation as default for the time
being.

GitHub: Fixes #1539.

---

Needs testing still. (Even if Zynq was the specific motiva [... cut]
commit 4f311e74 (2020-11-09): compiler: Raise exception on failed assert()s rather than panic

This allows assert() to be used on Zynq, where abort() is not
currently implemented for kernels. Furthermore, this is arguably
the more natural implementation of assertions on all kernel targets
(i.e. where embedding into host Python is used), as it matches host
Python behavior, and the exception information actually makes it to
the user rather than leading to a ConnectionClosed error.

Since this does not implement printing of the subexpressions, I
left the old print+abort implementation as default for the time
being.

The lit/i [... cut]
closed 2020-11-09 by dnadlinger