m-labs/artiq #1548: Request: Update pyqtgraph & nix scripts

drewrisinger, 2020-11-24:
# ARTIQ Feature Request

## Problem this request addresses

``pythonDeps.pyqtgraph`` is out of date (v0.10.0 vs v0.11.0) in the nix-scripts path. I can't submit an issue there, so it has to be here.

## Describe the solution you'd like

Apply the following patch to nix-scripts with ``git am``:
* converts to using pname > name convention
* updates pyqtgraph 0.10.0 -> 0.11.0

Tested the evaluation locally, currently trying to build llvm but my PC is too slow/old to build all of ``artiq-fast`` in a reasonable time.

## Additional context

<details>
    <summary>Full patch file:</summary>

```
From 02044868964e497195aaf1e7c6be1b3b87af4e8c Mon Sep 17 00:00:00 2001
From: Drew Risinger <drewrisinger@users.noreply.github.com>
Date: Tue, 24 Nov 2020 09:26:35 -0500
Subject: [PATCH 1/3] conda build: allow pname or name convention

---
 artiq-fast/conda/build.nix | 14 ++++++++++++--
 1 file changed, 12 insertions(+), 2 deletions(-)

diff --git a/artiq-fast/conda/build.nix b/artiq-fast/conda/build.nix
index c6b7017..9e3db36 100644
--- a/artiq-fast/conda/build.nix
+++ b/artiq-fast/conda/build.nix
@@ -2,12 +2,22 @@
 # recipe must be a string pointing to a path within the source.
 
 { pkgs }:
-{ name, src, recipe ? "fake-conda"}:
+{ name ? null
+, src
+, pname ? null
+, version ? null
+, recipe ? "fake-conda"
+}:
 
+# Check that either name is specified or both pname & version are specified.
+assert (name == null) -> pname != null && version != null;
+assert (name != null) -> pname == null && version == null;
 let
   condaBuilderEnv = import ./builder-env.nix { inherit pkgs; };
+  realName = if (name != null) then name else "${pname}-${version}";
 in pkgs.stdenv.mkDerivation {
-  inherit name src;
+  name = realName;
+  inherit src;
   buildCommand =
     ''
     HOME=`pwd`
-- 
2.25.1

From 576df13533aad445e6bcf2b52c81f1f64cf902af Mon Sep 17 00:00:00 2001
From: Drew Risinger <drewrisinger@users.noreply.github.com>
Date: Tue, 24 Nov 2020 09:03:11 -0500
Subject: [PATCH 2/3] treewide: name -> pname

Changes "name = '...-${version}" -> "pname = '...'" syntax everywhere.
This matches current best nixpkgs convention.
The nixpkgs standard derivation will convert this to name =
"${pname}-${version}" automatically.
[... 326 lines cut ...]
--- a/artiq-fast/pkgs/python-deps.nix
+++ b/artiq-fast/pkgs/python-deps.nix
@@ -41,13 +41,14 @@ rec {
 
   pyqtgraph-qt5 = python3Packages.buildPythonPackage rec {
     pname = "pyqtgraph_qt5";
-    version = "0.10.0";
+    version = "0.11.0";
     doCheck = false;
+    pythonImportsCheck = [ "pyqtgraph" ];
     src = fetchFromGitHub {
       owner = "pyqtgraph";
       repo = "pyqtgraph";
-      rev = "1426e334e1d20542400d77c72c132b04c6d17ddb";
-      sha256 = "1079haxyr316jf0wpirxdj0ry6j8mr16cqr0dyyrd5cnxwl7zssh";
+      rev = "pyqtgraph-${version}";
+      sha256 = "03fvpkqdn80ni51msvyivmghw41qk4vplwdqndkvzzzlppimdjbn";
     };
     propagatedBuildInputs = with python3Packages; [ scipy numpy pyqt5 pyopengl ];
   };
-- 
2.25.1

```
</details>

drewrisinger, 2020-11-24:
Of relevance: https://github.com/NixOS/nixpkgs/pull/104728 was just merged, so a separate ``pyqtgraph-qt5`` shouldn't be needed anymore.

sbourdeauducq, 2020-11-24:
Thanks.
I have bumped pyqtgraph. The other changes do need testing - do you want a SSH account for a fast machine at M-Labs?
Your nixpkgs PR will only be in a release in about 6 months.

drewrisinger, 2020-11-24:
Yes, you're right. I forgot which version of ``nixpkgs`` that the ARTIQ branches are tracking, but I could try to backport that PR.

I'm agnostic about the SSH account, can go either way. I could also probably test it on a faster machine that's free at UMD.

sbourdeauducq, 2020-11-24:
> test it on a faster machine that's free at UMD.

That's probably easier.

sbourdeauducq, 2020-11-24:
> I could try to backport that PR.

Does NixOS accept upgrades like this on release branches?

drewrisinger, 2020-11-24:
> > I could try to backport that PR.
> 
> Does NixOS accept upgrades like this on release branches?

Usually not, unless it's a good reason. However, pyqt4 hasn't been updated in ~5 years, so I think that might be a good enough reason.

drewrisinger, 2020-12-16:
Sorry I let this slide for a while. I had to replicate most of the Hydra ARTIQ build environment/source patches locally to get it to work, but I built all of ``artiq-fast`` & ``artiq-full/extras.nix`` last night. I didn't build any vivado/board packages.

<details>
<summary>The updated patches</summary>

```
From f61df0b1e29700286ff2a5a1a2ba4a7deec34b1d Mon Sep 17 00:00:00 2001
From: Drew Risinger <drewrisinger@users.noreply.github.com>
Date: Tue, 24 Nov 2020 09:26:35 -0500
Subject: [PATCH 1/2] conda build: allow pname or name convention

---
 artiq-fast/conda/build.nix | 14 ++++++++++++--
 1 file changed, 12 insertions(+), 2 deletions(-)

diff --git a/artiq-fast/conda/build.nix b/artiq-fast/conda/build.nix
index c6b7017..9e3db36 100644
--- a/artiq-fast/conda/build.nix
+++ b/artiq-fast/conda/build.nix
@@ -2,12 +2,22 @@
 # recipe must be a string pointing to a path within the source.
 
 { pkgs }:
-{ name, src, recipe ? "fake-conda"}:
+{ name ? null
+, src
+, pname ? null
+, version ? null
+, recipe ? "fake-conda"
+}:
 
+# Check that either name is specified or both pname & version are specified.
+assert (name == null) -> pname != null && version != null;
+assert (name != null) -> pname == null && version == null;
 let
   condaBuilderEnv = import ./builder-env.nix { inherit pkgs; };
+  realName = if (name != null) then name else "${pname}-${version}";
 in pkgs.stdenv.mkDerivation {
-  inherit name src;
+  name = realName;
+  inherit src;
   buildCommand =
     ''
     HOME=`pwd`
-- 
2.25.1

From 6d58818bc0cfa3854564ed938013e231fc103a3c Mon Sep 17 00:00:00 2001
From: Drew Risinger <drisingr@umd.edu>
Date: Mon, 14 Dec 2020 20:30:21 -0500
Subject: [PATCH 2/2] treewide: name -> pname where possible

Changes "name = '...-${version}" -> "pname = '...'" syntax everywhere
that is easy.
This matches current best nixpkgs convention.
The nixpkgs standard derivation will convert this to name =
"${pname}-${version}" automatically.
This also has the added benefit of adding the version to the filepath
for several packages that didn't have it before.

The only remaining files are ./artiq-full/extras.nix &
./artiq-fast/pkgs/python-deps.nix.
extras.nix had a
semi-confusing function that I didn't want to mess with.
---
 artiq-fast/default.nix               | 15 ++++++++-----
 artiq-fast/pkgs/artiq.nix            |  2 +-
 artiq-fast/pkgs/libartiq-support.nix |  2 +-
 artiq-fast/pkgs/llvm-or1k.nix        |  4 ++--
 artiq-fast/pkgs/llvmlite-artiq.nix   |  2 +-
[... 308 lines cut ...]
       mkdir -p $out $out/nix-support
       cp build/mirny.jed $out
-- 
2.25.1
```
</details>

Build log (again, building all of ``artiq-fast & artiq-full/extras.nix``). I can share the full derivation file (``./nix/artiqpkgs.nix``).
Build fails due to missing vivado in FHS Env.
```
$ nix build -j4 -f ./nix/artiqpkgs.nix
builder for '/nix/store/czycwhxv2fjnm1pghxalmqkpvy04f5x1-artiq-board-kasli-tester-5.7139.a694d13.drv' failed with exit code 1; last 10 log lines:
    File "/nix/store/r4m3skd312pxim2djn14kcxcn2nzqlpa-python3-3.7.6-env/lib/python3.7/site-packages/misoc/integration/soc_core.py", line 206, in build
      self.platform.build(self, *args, **kwargs)
    File "/nix/store/r4m3skd312pxim2djn14kcxcn2nzqlpa-python3-3.7.6-env/lib/python3.7/site-packages/migen/build/xilinx/platform.py", line 59, in build
      return self.toolchain.build(self, *args, **kwargs)
    File "/nix/store/r4m3skd312pxim2djn14kcxcn2nzqlpa-python3-3.7.6-env/lib/python3.7/site-packages/migen/build/xilinx/vivado.py", line 218, in build
      _run_vivado(build_name)
    File "/nix/store/r4m3skd312pxim2djn14kcxcn2nzqlpa-python3-3.7.6-env/lib/python3.7/site-packages/migen/build/xilinx/vivado.py", line 69, in _run_vivado
      raise OSError("Subprocess failed")
  OSError: Subprocess failed
  
[46 built (1 failed), 9 copied (5478.4 MiB), 1942.4 MiB DL]
error: build of '/nix/store/czycwhxv2fjnm1pghxalmqkpvy04f5x1-artiq-board-kasli-tester-5.7139.a694d13.drv', '/nix/store/yhra64nf02xgaydr6lhcv022ndrag34p-artiq-board-kc705-nist_clock-5.7139.a694d13.drv' failed
```

sbourdeauducq, 2020-12-17:
@drewrisinger Thanks, merged.

closed 2020-12-17 by sbourdeauducq

drewrisinger, 2020-12-17:
You're welcome. Thanks for the quick merge!